openscad/openscad #1394: When highlighting several lines of code, allow Tab to indent all of those lines

Pikamander2, 2015-07-22:
In Notepad++ and many other text editors, you can highlight multiple lines of code and then press Tab to indent all of the lines.

In the OpenSCAD editor, it will instead replace all of the highlighted lines with a single tab character.

You can use Control + I to get the same behavior, but it would be nice to have an option to enable "tab key multi-line indenting".

t-paul, 2015-07-22:
That is already supported, although I just noted there might be an issue with the preferences setting controlling that feature.

There is `Editor -> Indentation -> Tab key function` which should decide whether to insert a tab character or to do indentation. For me (on Linux) it always indents (including multiple lines when highlighted) regardless of the setting.

On which OS does that happen?

MichaelAtOz, 2015-07-22:
Oops on my +1, it works that way on Windows, with Preference Tab Key Funct. set to Insert Tab.

Pikamander2, 2015-07-22:
I tried both of the "Tab Key function" settings but it still overwrites all of the text instead of indenting it.

I'm on Windows 7. Any idea what the issue is?

Pikamander2, 2015-07-22:
I fixed it. The problem was that the Editor Type was set to simple editor instead of Qscintilla editor. After changing that setting and restarting the program, it now works.

There's no warning that the tab key function doesn't work with the simple editor, so that might be an issue of its own.

t-paul, 2015-07-22:
Ah, ok. The simple editor essentially has no settings. This is only there as fallback in case there were issues with the new editor. I guess the old one might go way in the next release or so.

I'll close this issue then and create a new one to check the Tab Key Function setting...

closed 2015-07-22 by t-paul