openscad/openscad #1398: Please add ability to globally specify center = true

foulowl, 2015-07-24:
Hello,

I use "center=true" for all my primitives. AFAIK there is no way to specify this for the entire file. Per file basis is fine, or wrapping a {} block is fine too. Just sick of typing that over and over. :)

Thanks!!

TakeItAndRun, 2015-07-24:
You dont have to type the "center=" all the time. Just use the commands
without explicit definitions:

cube([10,10,10],true);
cylinder(10,2,2,true);

Still have to type true, though.

t=true;

TakeItAndRun, 2015-07-24:
If one would make center a "global" variable, than it would be $center.
More typing for the rest of us. :)

FluxIX, 2015-07-24:
I'm not sure I like introducing any more globals. I understand the frustration having to type it repeatedly, but when looking at the code, a global is not immediately visible, and centering is a very influential decision.

My argument doesn't even address the consequences of including a file setting globals.

foulowl, 2015-07-24:
Well the global $center would simply default to false, as it is currently. No more typing is needed for anyone else, no existing code is broken, it simply adds the ability to do $center=true at the top of your file and be done with it. 

Make it only apply to a module if you want, make it a function, any idea you have I will support as long as it provides some sort of shorthand. :)

I just grepped my source tree and for the parts I have designed in the past month or so, it looks like I have typed "center=true" a total of 411 times, for a total of 4.5k of unneeded characters. Please, there really is a motivation here.

Thanks!

MichaelAtOz, 2015-07-25:
```
module ccub(size=1,center=true) cube(size=size,center=center);

ccub([10,20,30]);

module ccyl(h=1, center=true,r,d,r1,r2,d1,d2) cylinder(h=h,center=center,r=r,d=d,r1=r1,r2=r2,d1=d1,d2=d2);

translate([20,0,0])
ccyl(h=5,r1=5,r2=5,h=10);
translate([-20,0,0])
    ccyl(d=2,h=12);
```

FluxIX, 2015-07-26:
I have an idea but I don't know how much change it required to implement it.

Each scope construct can have any named parameters set (even if the scope construct doesn't use it, in which case the proper behavior is to ignore the parameter). Each scope construct would inherit the parameter values from the previous scope, so a higher-level scope construct (like a `module`) could set a parameter value (like `center`) and all of its children would inherit that parameter value (although the children could override the parameter value).

This approach is more targeted than using globals, and also supports other properties besides `center`. 

If this approach were to be supported, I would recommend specifying a special parameter `ignore_heritage` which would ignore all otherwise inherited parameters.

t-paul, 2015-07-26:
Following the example @MichaelAtOz outlined, there might be an option that does not need the introduction of additional special variables.

Right now it's not possible to overwrite module names and still access the built-in implementation. Full support for namespaces might take a while to implement.

But: Maybe it's possible to define a way to only support built-in support first. That way the example from above could actually use `cube()` instead of `ccub()`.

MichaelAtOz, 2015-07-27:
> But: Maybe it's possible to define a way to only support built-in support first. That way the example from above could actually use cube() instead of ccub().

I've wanted something like that. It needs a way to specifically call the built-in module/function tho.

Something like `module cube(size=1,center=true) @cube(size,center);`

nophead, 2015-07-27:
I don't think being able to change the semantics of built in functions is a good idea. It just obfuscates the code. At least if you are forced to change the name there is no confusion.

t-paul, 2015-07-27:
@nophead that feature is there, overwriting built-in names is already possible. Right now there's just no way to access the built-in name again, it will always call the newly created one recursively. So at least it would be possible to overwrite ` cube()` using `linear_extrude()` [like so](https://gist.github.com/GilesBathgate/8442338#file-builtin-scad-L127).

I agree that it's a feature that could be used to confuse/obfuscate code. But that applies to other things too (e.g. operator overloading supported in many languages). If used in a careful and responsible way, it can help to make code clearer. I'm just thinking that the direct access to built-in version will very likely come anyway once namespaces will be implemented, and we certainly want to have those.

TakeItAndRun, 2015-07-27:
Transformation can already be resigned with this trick: use multmatrix()

module mirror(v,keep=false){
// keep the original
    if(keep)children();
// normalize v, in case it isent already
        u=v/norm(v);
        a=u.x;
        b=u.y;
        c=u.z;
    multmatrix(m = [
                [1-2*a*a, -2*a*b, -2*a*c, 0],
                [-2*a*b,1-2*b*b, -2*b*c, 0],
                [-2*a*c, -2*b*c, 1-2*c*c, 0],
                [0, 0, 0,  1]
    ])
        children();
}

mirror([1,1,1],true)cube(1);

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

ggallant571, 2019-06-25:
I realize that this is an old issue but I have been waiting 4 years for someone to solve. Spent about 2 hours, added 2 lines of code, moved 1 line, and have "$center" implemented. Default is false so there should be no adverse effects. I am NOT an expert openscad user and my test case is limited.

DejfRCR, 2021-04-07:
Well,
I wonder how is it possible that this mess still prevail. I asked google for the parameter to enable global centering today only to find out that flawless solution was lined up ten years ago, then another one and yet another one. There are three ways of implementing such possibility without breaking any existing code and yet it was not implemented and refused with completely nonsensual argumentation like "I dont need it, so its not an issue". Or there is a code readability argument which is hard to understand as currently every object behave differently: sphere is origined in its center...why? Why is it not centered on the bottom or top like cylinder? Or like cube in corner of surrounding cube? How could this make readability easier to anyone but folks hard fixed on current mess for years? No, it does not, it is a very hard for newcomer to read current code as one has to imagine each objects origin separately - and I have university education in geometry, so my imagination of things is way above average.

So let me put it from another perspective. For me, as former system admin (for about 10 years I work in small textile manufacture doing real things in real world to much satisfaction), programming is a matter of self deffence, when existing solution fail to satisfy my needs. Thus downloading sources of openscad and implementing this very feature as global variable dont seem as anything impossible, but that would require me to use my scad for every project and not be able to share my work with others without sharing my supertinybit improved openscad. I will do that eventually, after two or three more projects it would seem like the only way of not loosing my creative attitude for the tool. I feel like on school exam every time i need to translate or like a code monkey with every redundant center=true. 
Doing it once and handling it over thru newer versions of openscad are two very different things. Thus I will be locked in my unupdatable and incompatible universe that will even loose ability to renders works of other with new features.  I suppose there are other folks who already done this thing, each in different way, so even those universes are incompatible with each other. 

Thus I don't propose this feature to be added, that is impossible to push thru no matter the zero impact on anything and clear usefulness for many. I propose a fork that would include this one simple feature implemented in one of the ways proposed by other already:
1] $center as a global boolean variable, false in default
2] center() as a function that wraps all parts that require it
These two are simple bools that would require very low effort to implement and bring most of us most of what we need

1] $center as a global bool (or even float) field setting per axis centering to allow cube to behave same as default cylinder
2] center(x,y,z) as a function to wrap with per axis centering
These two would require a bit more effort and frankly these would break ability to read the code easily when used with too inconsistent approach. But they would allow lots of magic like in the good old assembler days when computers did only what we asked them to.
I dont get the code obfuscation argument for these as I cant imagine a malware in openscad code. Please enlighten me in this subject why could code obfuscation harm anything but ability of others to decipher authors messy work. You can write messy code in almost any language, its a matter of style and habits, not particular function.

So, please let me know whether there are:
1) users seeking this
2) users willing to take part on the fork as it would require some work with every new release and I am far from promising to do it myself every time in reasonable time

nophead, 2021-04-07:
Why don't you just define your own version of cube that centres the same as a cylinder if you want one? A lot simpler than maintaining a fork. You can't reference the original cube module but you can make a cube() module that linear_extrudes a square.

t-paul, 2021-04-07:
So lets use this `cube()` which implements usage of `$center`:
**ccube.scad**
```
module cube(v) {
    c = is_undef($center) ? false : $center;
    p = is_num(v) ? [v, v, v] : v;
    linear_extrude(height = p.z, center = c) square([p.x, p.y], center = c);
}
```
Now assume we have a library distributed by someone, we'd like to use that in our project:
**lib.scad**
```
include <ccube.scad>

module handle() {
    translate([4, 4, 0]) children();
    color("green", 0.3) cube(8);
    color("green", 0.3) translate([0, 0, 28]) cube(8);
}
```
And the project is:
**main.scad**
```
include <ccube.scad>

use <lib.scad>

handle() cylinder(r = 2, h = 36);
translate([20, 0, 0]) handle()
    translate([-1, -1, 0]) cube([2, 2, 36]);

// now how to use $center to make it centered at origin
// but not impact the imported library?
cube(5);
```

nophead, 2021-04-07:
lib.scad would not see the version of cube in ccube.scad, so it would not be affected if you set $center in main.

t-paul, 2021-04-07:
`ccube` is just the illustration of implementing `$center` in core. In that case it would be visible.

DejfRCR, 2021-04-07:
Wow, so you suggest using new complicated code just to replace it? I guess you have written more code for one object that it would require the openscad modification for all of them. Using your solution would give questions about what does the cube realy do, it would need to offer center=false option for the rare cases that might employ it in this paradigm - thus many lines for every affected object. It would be much easier to run my code thru awk and put ,center=true everywhere.  Much much cleaner way would be making module ccube() wrapping cube() inside it, ccylinder... In other words instead of clean and simple $center=true; one should produce pages of code to make it more readable.

I just dont get it why you insist on having so ununited way of looking at objects. I'd be happy with origin in corner of surrounding cube of sphere and cylinder. I dont really care where the origin is, but I want to have it in the same predictable place with all possible objects (and center is just the best for imagination). Offering such possibility would not break anything while it would be clear what commands do. Suggesting new mess as a solution to old mess is no solution for me, sorry.

And silverlining is that if fork succedes and code requiring it will emerge, eventually openscad will have to merge it in and fork will be no longer needed. - I guess a year, two at most. Difference would be, that it would be implementation forced from outside, probably worse than the one you would oppose least.

t-paul, 2021-04-07:
I was asking how to make `$center` work without breaking existing libraries. It's just using the script to actually see what's happening by actually trying it out without implementing things in C++.

nophead, 2021-04-07:
Why would anybody use a fork that breaks all their existing code? I have been using OpenSCAD for 10 years and have no problem remembering where the primitives are centred.

jordanbrown0, 2021-04-08:
The rules are really pretty simple:
- Things that supply their own coordinates, supply their own coordinates.  (Note that linear_extrude supplies its own coordinates only in X and Y.)
- Round things are centered on the origin.  (Note that a cylinder is only round in X and Y.)
- Everything else is in +X/+Y/+Z.

If you want to be picky, text() has an origin based on font layout data.

I'd like to see a universal three-axis origin-control mechanism, but that's because it's convenient, not because I find the default behavior hard to remember.

DejfRCR, 2021-04-08:
> Why would anybody use a fork that breaks all their existing code? I have been using OpenSCAD for 10 years and have no problem remembering where the primitives are centred.

Nothing that breaks any existing code was never suggested in any of the numerous threads about this subject. Yeah, I have no problem remembering where primitives are centered - in the middle, problem is imagining where their origin is and what do I have to do to shape them where I want. That is why I want them all easily originated in middle.
This debate is about keeping historical paradigm that (hopefully)  has its good reasons and nobody is suggesting to disable it. At the same moment, Assign function is deprecated and its removal will inevitably break some code and nobody has problem with that - so in my view its a grotesque show of misunderstanding.

> The rules are really pretty simple:
> 
> * Things that supply their own coordinates, supply their own coordinates.
>   
>   * polygon, polyhedron, import, linear_extrude X/Y, rotate_extrude input X/Y.
> * Round things are centered on the origin.
>   
>   * circle, sphere, cylinder X/Y, rotate_extrude result X/Y
> * Everything else is in +X/+Y/+Z.
>   
>   * square, cube, cylinder Z, linear_extrude Z, text.
> 
> If you want to be picky, text() has an origin based on font layout data.

Yeah, it is simple to remember, but hard to imagine during the design - I have those things in my head, in 95% cases even without a paper sketch - thus I need to have all origins at same position to produce code in a good flow. Cylinder is not centered at origin, but it originates in top/bottom while sphere originates in middle - these two for example differ for no practical reason. Objects with absolute coordinates (defined by corners, not sizes) like polygon/hedron work in very different way, so dont mess these into the debate - it would be nice and clear if every cube was polyhedron and things like cylinders and spheres were easy to define using it, we would have a completely different predictable universe, not the one I would favor, but also acceptable.

*************************************************************
Please folks, accept that I am going to make my own build including one fix or another no matter what. I need the tool for my work and I have to have comfortable environment. So I will produce code, that will be unrenderable anywhere outside my universe. 
What I am seeking is whether there are other people interrested in designing, building and using such environment, not to read again all the things that made no sense to me in previous threads. 

I dont really care, I built heavily customizable frame builder for lego that I found nowhere else using basic code done by other people - now I can build human size robots (far from that now) or complicated gearboxes impossible to build using original parts. Next revision with new features will be centering all origins saving 20% of lines and fixing numerous little flaws that need arbitrary float constants everywhere to translate correctly at exact positions proly breaking easy scalability for different filaments. I am going to produce numerous objects for textile industry and both are for my own use or use of my company. I don't care how many people will be able to use my code, but maybe there is a feasible way to extend the great tool which OpenScad surely is for everyone.

nophead, 2021-04-08:
I don't understand how it saves any lines of code. If you want cube to have its centre in the middle you just set center = true when you call it. It adds some characters, but not any new lines and some code would get longer where I would need to center = false as I don't always model from the centre.

t-paul, 2021-04-08:
> Nothing that breaks any existing code was never suggested in any of the numerous threads about this subject.

Yes it was

`$center` is not possible without breaking code using libraries

I asked about this very specifically and you did choose to ignore that very clear example I gave.

ggallant571, 2021-04-09:
I have a preference for the $center scheme and have it in a local build. 
The primary issue I see is compatibility with imported code. One 
possibility is to set the $center to false when encountering a "use 
<xxx.scad>" statement.

t-paul, 2021-04-09:
And again *not possible*. There is no way a variable can have multiple values in one scope, that would be an even more fundamental change of OpenSCAD behavior, so you can not just "set it before `use<>`" and have it different after.

And even if that would be possible it would portentially break when using `children()` as this is passing things from a different scop.

DejfRCR, 2021-04-09:
Wow, so now we all agree on that having $center=false break nothing? So what is this nonsense about?

And why do you charge only against one on the suggested implementations? center() {centered objects} would break nothing, only require also uncenter() or center(false) {} to be usable with imported uncentered code or 

That was my last reply to these voices. My post was about who is interrested in colaborating on such build, not why dont you want to take part of it. This basic misunderstanding of this simple things is base of our dispute. 

1) We have proven, that inside implementation of such behavior would be a nightmare wasting so much cpu power to overcome simple change in sources that it is not the way, even for trials. This solution is not acceptable
2) we have finally concluded that addition of global $center=false would break exactly nothing. So those voices are just emotionally motivated and in further discussion ignored
3) we have no argument against center() or more powerful align(), so implementation of these would again break exactly nothing.
4) we already know that breakage of some existing code with planned future new release in no issue for developers as deprecation of assign shows clearly.

So, if you oppose  the idea of possibility of unification of origins of objects it is only matter of your emotional state. I can understand that. I began improving existing code without any centering for a while before I found that feature exist. Since that moment I have not placed a single object without it and saved half of translations in the code and greatly simplified what remained. Imagining that I would have been going without it for years, I'd be jealous on folks who found it rightaway. But I would never pose nonsensual opposition to such extension acting like all knowing pro.

********************************************

 ggallant571  could you post your patch please?

t-paul, 2021-04-09:
Ok, so there's obviously no useful discussion to be had. Sorry I have even tried.

t-paul, 2021-04-09:
Closing this now as it leads just to discussion in circles.

closed 2021-04-09 by t-paul

MichaelAtOz, 2021-04-09:
@DejfRGR
> My post was about who is interrested in colaborating on such build

Not me, not interested, and I didn't hear anyone else.

I find your posts verging on offensive.

> Please folks, accept that I am going to make my own build including one fix or another no matter what. I need the tool for my work and I have to have comfortable environment. So I will produce code, that will be unrenderable anywhere outside my universe.
What I am seeking is whether there are other people interrested in designing, building and using such environment, not to read again all the things that made no sense to me in previous threads.
>
> I dont really care

OK, you have your build. No need to berate the rest of us.

jordanbrown0, 2021-04-09:
Why isn't the right answer to this question to just wrap the five primitives where centering of this sort is relevant, and then use those wrappers?

```
module box(dims) {
    cube(dims, center=true);
}
module rect(dims) {
    square(dims, center=true);
}
module cyl(h, r, d, r1, r2, d1, d2) {
    cylinder(h=h, r=r, d=d, r1=r1, r2=r2, d1=d1, d2=d2, center=true);
}
module lin_extrude(height, convexity, twist, scale, slices) {
    linear_extrude(height=height, convexity=convexity, twist=twist, scale=scale, slices=slices)
        children();
}
module surf(file, convexity) {
    surface(file=file, convexity=convexity, center=true);
}
```

No muss, no fuss, no incompatibility, no C++ work.  The only thing lost from the original proposal is that these new names are not listed in the OpenSCAD documentation.

ggallant571, 2021-04-09:
I am interested but leave your ego at home. If anyone else is interested in exploring a method to implement a $center feature please contact me at:

ggallant571@yahoo.com

My preference would be to use this forum for polite constructive communication. If a solution is feasible then propose it as a generic feature of openscad. If emotions (plus/minus) are stronger than reality then form a splinter project.

t-paul, 2021-04-09:
So far I have not found a solution that will not break in the general case, there's always corner cases that would cause problems with backward compatibility. And while OpenSCAD is not the most used software, there's a fair amount of people using it for [quite big and impressive projects](https://www.openscad.org/gallery.html).

Thank you, @ggallant571. I'm happy to follow a polite constructive communication, maybe there's a way to solve that issue, I'm not claiming to know everything, it's simply an issue that came up multiple times and so far the discussions were not able to come to a solution that does not break backward compatibility. If you find a way that works, it would be nice to discuss that in a new ticket.

I'm just not ok with (other) people coming here claiming everything is a mess in what I (not being native english speaker) would percieve as pretty rude messages claiming different things all the time while ignoring questions about problems raised.

ggallant571, 2021-04-09:
I'm a software engineer and a casual mechanical designer whose mindset gets completely lost with graphical tools. Came across openscad by accident about 5 years ago and have been a advocate since.

My number 1 problem with openscad  is keeping a consistent view of the project. For projects with multiple independent components I find using offsets from a common center point easier to program than absolute values from a corner.

The stability of the code set has greatly impressed me and I think highly of the developers who could do such a through job. The downside to doing a good job is the mental reluctance to change anything that "might" induce problems.

I have a strong reluctance to claims made by emotion rather than facts. This entire discussion has been nothing but emotion. I'm in favor of a group investigating, implementing, testing, documenting and proposing a change to openscad. I have no desire to be part of the group that accepts or rejects the proposal.

DejfRCR, 2021-04-09:
Openscad is a great tool, sure. Most super impressive usages will not be using centristic approach as it cant help them - they result of serious analytics geometry and when everything is precalculated on paper, then nobody cares of the origins during the translation into code. Centristic paradigm is useful for hobby projects and projects simple enough to be kept in one mind. Like everything I could need from it for my needs both personal and professional.

Jordanbrown: try this: box([5,0]); and try to find it in code - warning/error is on line of the module, so not feasible. As I already noted, the wraparound is much more work than what was suggested by t-paul. The aim is not to make it harder to code but to make it easier. The aim is not at wasting cpu power in robust wraparounds that would present new bugs. The aim is to make the code more readable by dropping ,true on every line.

t-paul: so far I have not seen any code that would be broken by pure existence of $center=false, only claims of its existence. In ten years worth of threads.

MichaelAtOz: I find it very offensive when I suggest a solution that relieves this project of this kind of circular debates and most come only to say how stupid that idea is and repeat what was said many times over.

*******************************************************

Yeah at this moment I am having great fun, off topic markings are cool message that this project is not willing to allow a spinoff  to form inside. I am sorry for the time I wasted here, If I knew, I would not bother with the insane ten wave captcha.

I wish this project the best a will make my own build or find one enabling the feature as soon as I make my conclusion of the best way to do it - which is sadly not possible here.

jordanbrown0, 2021-04-09:
```
WARNING: Unable to convert cube(size=[5, 0], ...) parameter to a number or a vec3 of numbers in file , line 2 
TRACE: called by 'cube' in file OpenSCAD-2021.01, line 2 
TRACE: called by 'box' in file OpenSCAD-2021.01, line 5 
```

DejfRCR, 2021-04-09:
Yeah 3 lines to read instead of one. Thanks for proving me right and goodbye, I will not write here anymore.

jordanbrown0, 2021-04-09:
The cube() case generates two lines, not one.
```
WARNING: Unable to convert cube(size=[5, 0], ...) parameter to a number or a vec3 of numbers in file , line 1 
TRACE: called by 'cube' in file OpenSCAD-2021.01, line 1 
```
And the ratio only gets less unfavorable as your cube is buried deeper inside a non-trivial project.

jordanbrown0, 2021-04-10:
The CPU expense of a wrapper is negligible.  It's not unmeasurable, but it's trivial compared to the geometry processing cost is so dominant.  It generally requires effort to write an OpenSCAD program where the actual execution time is significant compared to the geometry time.

```
n = 100;

module box(size) {
    cube(size=size, center=true);
}

for (x=[1:n], y=[1:n]) translate([x,y,0]) cube();
```
generates 10,000 cubes.

On my system it previews in about 0.85 seconds with cube() and 1.0 seconds with box().  That's execution time plus whatever processing the previewer needs to do.  F6 render, on the other hand, took 21:41.033 one time and 19:48.181 another time for the cube variant and 20:42.603 and 18:47.301 for the box variant.  I'm not going to claim that the box variant is faster, but it does seem clear that any difference is in the noise for anything other than a specialty benchmark.

MichaelAtOz, 2021-04-10:
Most of that render time is the union(), with lazy-union Total rendering time: 0:00:00.523

DejfRCR, 2021-04-11:
I know I promised not to write again, but I have an ultimate argument against $center solution: it is no solution at all, because it only shifts the burden of messy code from one group to another. It would inevitably split the code universe into true and false parts while folks producing universal code would have to make explicit declarations on every line ending like centrists currently do.
Thus I am exploring the other two ways I noticed here.

jordanbrown0: I am very sorry, I forgot one thing: Enable trace and then read three lines instead of one.
 Enabling trace is just another negligible nothing. And the very reason why todays computers are as slow as 20 years ago for so many tasks is negligible minimums multiplication. Every developer creates his tiny 1.01 increase and current working environment suffer from 1.01^X with X in at least thousands and rising every day.  I remember the times when assembler was the only tool for so many tasks and virtually every cpu cycle was worth protecting, thus my blood boils whenever I read such argumentation.
The time you spent disproving the least important part of my message is just briliant :)