openscad/openscad #1399: Refactor interface to CSG engine

bobc, 2015-07-24:
This relates to the general objective "Implement prototypes that could be tested inside the OpenSCAD framework" of https://github.com/openscad/openscad/wiki/Project:-Survey-of-CSG-algorithms 

This is a fairly major change, so I open this as a "Request For Comments"

Goal: allow a backend CSG engine to be chosen at compile time, or none. The command line and GUI should still run as far as possible without a CSG selected.

Rationale: with a cleaner interface, it will be easier to implement prototypes. I have a prototype which uses the Carve CSG library, but it is rather hacked.

*Comment : there is already partial support for this with "ENABLE_CGAL" flag, but it has decayed*

Proposal :  
- Create a new interface layer called "CSGIF" (better names welcome!)
- Create new class "CSGIF_Polyhedron" to replace CGAL_Nef_polyhedron
- single include file "CSGIF.h" selects between no CSG, CGAL or other (eg. carve).
- Functions which are specific to a CSG engine should be collected together, this includes all the CSG ops, export functions specific to CSG engine, and OpenGL rendering.

There are some functions which aren't CGAL dependent which should be made generic, e.g. CGALCache. There are also some parts that "deep dive" into CGAL structures, these need to be genericised.

There are some functions in CGAL which may not be in other CSG engines, e.g. Minkowski, hull. These would need to be implemented in the CSGIF layer, I suggest flags such as ENABLE_HULL etc if the functions are not available.

Plan:
- I will create a csg_refactor branch in my repo, and make a PR for review. 

I am quite open to questions, comments. Anything from coding style to architecture. I am quite motivated to get this done right!

oysteinkrog, 2015-07-25:
I'm curious; what are the advantages of being able to use other CSG libraries (such as carve)?

bobc, 2015-07-25:
CGAL gives good results but is relatively slow. On a quick test I did it was 12 times slower than Carve. The Holy Grail is a backend that is fast AND gives good results, but first we need to evaluate them in practical use with OpenSCAD programs.

My end goal is actually to add support for multiple volumes with different color or material properties. To do that will require more CSG operations, so a faster backend is a must.

Faster rendering and support for color are two frequently requested features.

bobc, 2015-08-04:
I have been mulling over this PR, and I think that although it works, it is not a very elegant solution. 

I think a better solution would be access the CSG interface via more usual C++ class mechanisms. Ie have a base class CSGIF and derive classes CSGIF_CGAL, CSGIF_Carve etc from it. There are 4 basic parts to the interface:
- CSG ops for geometry evaluation
- helper utilities, and conversions to/from other types (e.g Polysets)
- export to file
- rendering to screen

The first 3 perhaps should be all part of the same interface. Rendering to screen may have some dependencies on Qt which might want to keep separate. The advantage is that keeping all the backend related stuff in one interface makes the boundaries of the interface clearer, and allows some functions like export from polyset to be in a common base class.

An additional advantage is that the CSG backend could be selected at runtime, by feature option for example.

kintel, 2015-09-06:
@bobc Having (finally) read your code, I agree with your last comment. Some additional challenges:
* Partial results: We'll likely end up with a mixture of backends (e.g. carve for CSG, CGAL for hull; CGAL's hulls are fast). We thus needs multiple concrete implementations, i.e. multiple active subtypes of the CSGIF classes active.
* Caching: We already mix different representations of partial results. We need a better way of storing and retrieving such objects and knowing how to select the correct type in case multiple variants are available for the same key
* Interoperability between object types; converting from PolySet to CGAL_polyhedron isn't always possible due to manifoldness issues, but converting to e.g. Carve polygons might still be possible. How do we deal with this asymmetry? Right now, this causes some complexity since we cannot simply choose a master storage (hence the need of caching multiple different data types).