openscad/openscad #1407: Recursive module crash

untaugh, 2015-08-06:
I'm using linux and the following code makes openscad crash for me:

rec();

module rec(a=1)
{
    rec([a,10,10]);    
}

I tried many of the older recursion issues, and they were all detected properly, this one is not.

MichaelAtOz, 2015-08-07:
Confirmed on 2015.03 & snapshots, but 2014.03 correctly detects recursion. (Windows)

kintel, 2015-08-31:
Confirmed on Mac too

kintel, 2015-09-01:
Initial investigation shows that the stack check fails due to a lot of stack being used between our checks.
One particular crash using the file above used ~370K stack here:
https://github.com/openscad/openscad/blob/master/src/expr.cc#L412

The culprit appears to be that the std::vector copy constructor for a deep vector of vectors is recursive.
Our vectors are vectors containing boost::variant values.

kintel, 2015-09-03:
Partial stack trace:
```
    frame #5438: 0x00000001000444bb OpenSCAD`std::vector<Value, std::allocator<Value> >::vector(this=<unavailable>, __x=<unavailable>) + 187 at aligned_storage.hpp:57
    frame #5439: 0x00000001000444bb OpenSCAD`std::vector<Value, std::allocator<Value> >::vector(this=<unavailable>, __x=<unavailable>) + 187 at aligned_storage.hpp:57
    frame #5440: 0x00000001000444bb OpenSCAD`std::vector<Value, std::allocator<Value> >::vector(this=<unavailable>, __x=<unavailable>) + 187 at aligned_storage.hpp:57
    frame #5441: 0x0000000100046a39 OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] boost::detail::aligned_storage::aligned_storage_imp<24ul, 8ul>::address() const + 665 at aligned_storage.hpp:57
    frame #5442: 0x0000000100046a26 OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_>::which() const + 7 at aligned_storage.hpp:123
    frame #5443: 0x0000000100046a1f OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] boost::detail::variant::copy_into::result_type boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_>::internal_apply_visitor<boost::detail::variant::copy_into>(visitor=0x000060800023cec8) const + 3 at variant.hpp:2344
    frame #5444: 0x0000000100046a1c OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_>::variant(boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_> const&) + 8 at variant.hpp:1743
    frame #5445: 0x0000000100046a14 OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_>::variant(boost::variant<boost::blank, bool, double, std::string, std::vector<Value, std::allocator<Value> >, Value::RangeType, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_> const&) at variant.hpp:1747
    frame #5446: 0x0000000100046a14 OpenSCAD`std::vector<Value, std::allocator<Value> >::_M_insert_aux(__gnu_cxx::__normal_iterator<Value*, std::vector<Value, std::allocator<Value> > >, Value const&) [inlined] Va [... cut ...]
    frame #5453: 0x000000010006a0ae OpenSCAD`Context::setVariables(this=0x00007fff5f46a7e0, args=0x0000600000164690, evalctx=0x00007fff5f46ac08) + 382 at context.cc:88
    frame #5454: 0x000000010006e9f5 OpenSCAD`ModuleContext::initializeModule(this=0x00007fff5f46a7e0, module=0x0000600000164680) + 53 at modcontext.cc:68
    frame #5455: 0x0000000100062ed7 OpenSCAD`Module::instantiate(this=0x0000600000164680, ctx=<unavailable>, inst=0x00006000001809c0, evalctx=<unavailable>) const + 599 at module.cc:193
```

closed 2015-09-14 by kintel

kintel, 2015-09-14:
@untaugh Could you give the master branch a spin?

commit 7f9e7f9c (2015-09-14): Merge pull request #1432 from openscad/issue1407

Fix #1407 by storing vector values as heap objects (ValuePtr) rather than stack objects(Value)
commit 847cc76f (2015-09-14): Refactored to let VectorValues use ValuePtr instead of Value, to avoid having to copy large vectors. Attempt to fix #1407
commit fafd3e85 (2015-10-16): Refactored to let VectorValues use ValuePtr instead of Value, to avoid having to copy large vectors. Attempt to fix #1407

Conflicts:
	src/func.cc