openscad/openscad #1409: Preview broken with Windows 10 - with difference()

urmilparikh, 2015-08-09:
I upgraded my OS to Windows 10 and now I find that preview (F5) is broken in some cases; and that OpenSCAD shows different views at different angles. Render (F6) works fine, as expected.

Seems an issue with difference().

With this example code:
````
difference() {
	translate([-3, -3, 0]) cube([6, 6, 1]);
	translate([0, 0, -0.1]) cylinder(d=3, h=1.2, $fn=12);
}
````
I get this with F6 (as expected):
[image]

... but these with F5, at different view angles and at many view angles I do not see anything in preview:
[image]
[image]

t-paul, 2015-08-09:
Please post the information from `Help -> Library Info`.

Also check for updated graphics drivers. This type of issues are mostly caused by misbehaving drivers.

MichaelAtOz, 2015-08-09:
Check whether Edit/Preferences/Advanced/Force-Goldfeather makes any difference.

urmilparikh, 2015-08-10:
@MichaelAtOz: Yes, that fixes the issue.

@t-paul, I'll still update graphics driver and report result without 'Force-Goldfeather' set.

urmilparikh, 2015-08-10:
With updated driver as well I see this issue, so Force Goldfeather is the solution.

Thank you!

TakeItAndRun, 2015-08-10:
Now, dont close the issue. It should work without Force Gotdfeather

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!

urmilparikh, 2015-08-11:
Re-opening as requested.

t-paul, 2015-08-18:
@urmilparikh could you please post the content of the `Help -> Library Info` window. This has detail information about the graphics driver and the OpenGL context OpenSCAD got. Hopefully that helps to find the root cause of the issue.

urmilparikh, 2015-08-18:
Sure, here it is:

````
OpenSCAD Version: 2015.03-1
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.93 GB RAM
Compiler: GCC "4.9.2" 64bit
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.42.1
Application Path: C:/Program Files/OpenSCAD
Documents Path: D:\docs
Resource Path: C:/Program Files/OpenSCAD
User Library Path: D:/docs/OpenSCAD/libraries
User Config Path: C:\Users\Urmil\AppData\Local/OpenSCAD
Backup Path: D:/docs/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
D:/docs/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.3.0 - Build 10.18.15.4248
GL Renderer: Intel(R) HD Graphics 4400
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
[... 160 lines cut ...]
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_NV_DX_interop
````

t-paul, 2015-08-18:
Thanks, is that already with the updated graphics driver? The Intel Download Center seems to provide version [15.40.4.64.4256](https://downloadcenter.intel.com/download/25150/Intel-Iris-Iris-Pro-and-HD-Graphics-Production-Driver-for-Windows-10-64-bit) which sounds quite a bit newer than the one reported in the OpenSCAD info (10.18.15.4248).

urmilparikh, 2015-08-18:
I confirmed that Windows still says my driver is up-to-date.
It is possible that Windows is not yet installing this new version.

Neon22, 2015-08-19:
I have installed Build 10.18.15.4256 which windows informs me is the latest for my machine.
It shows same problem in F9 preview.
The "Force Goldfeather" fix works for this build.

urish, 2015-08-24:
Same here:

```
OpenSCAD Version: 2015.03-1
System information: Windows 8 (v6.2.0.0) 4 CPUs 15.91 GB RAM
...
OpenGL Version: 4.3.0 - Build 10.18.15.4256
GL Renderer: Intel(R) HD Graphics 4400
GL Vendor: Intel
```

Works only with the "Force Goldfeather" workaround

kintel, 2015-08-24:
@urish Is this on Windows 10 as well?

urish, 2015-08-24:
Yup, windows 10.

p.s. I love openscad!

kintel, 2015-08-24:
Thx - the only theory I have thus far is that this Intel driver is the first one that supports Windows 10, and there might be some initial issues. There are multiple other issues with this driver in other software (found by googling), so it might be worth waiting for an update.

(dev note: A better way of finding such issues might be for us to write a stand-alone OpenCSG test app, as an attempt to reduce the problem to as few components as possible.)

urish, 2015-08-24:
Thanks!

Are there any drawbacks for using the workaround?

kintel, 2015-08-24:
I think the drawback would only be rendering speed.

ehaskins, 2015-08-27:
I'm seeing the same issue. Surface Pro 3, with latest driver from Windows Update. 
```
OpenSCAD Version: 2015.03-1
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.91 GB RAM
...
OpenGL Version: 4.3.0 - Build 10.18.15.4256
GL Renderer: Intel(R) HD Graphics 4400
```
Full lib info output: https://gist.github.com/ehaskins/59108f38955d5b35e624

rossbagley, 2015-09-30:
I just manually installed the 15.40.4.64.4256 driver bundle and it reinstalled the HD Graphics Driver version 10.18.15.4248 (same version as I and everyone else using fully updated Windows 10 has).

And the "Force Goldfeather" worked for me as well. Thank <favorite deity>: waiting for rerender was making me code more carefully, but also driving me to use openscad less and less.

trayracing, 2015-10-01:
Doesn't occur in OpenSCAD 2015.03, but does in 2015.03-1 and later (including 2014.06.14).
I'm on Win10/Intel 4400/driver 10.18.15.4256. 
Slightly simpler repro case:
  difference()
  {
   cube(10, center=true);
   cube([5,5,20], center=true);
  }

trayracing, 2015-10-01:
In the two cube difference example, the model disappears when viewed from above or below. It appears without hole from the side. From a certain angle underneath, dependin the viewport size, I can see the either the cube or the hole, but not both. Screenshots attached, including a render of the intender scene, which looks correct.  Also, intersection() is not drawn correctly - looks like the first cube from above/below, but second cube from the side. union() looks correct. Goldfeather enable all look correct.'

[image]
[image]
[image]

keithjjones, 2016-03-19:
:+1:  Same issue.  Force goldfeather fixes it for me too. 

'''
OpenSCAD Version: 2015.03-2
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.92 GB RAM
Compiler: GCC "5.2.0" 64bit
Boost version: 1_57
Eigen version: 3.2.5
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\Keith\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/Keith/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Keith\AppData\Local/OpenSCAD
Backup Path: C:/Users/Keith/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/Keith/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.4.0 - Build 20.19.15.4331
GL Renderer: Intel(R) HD Graphics 5300
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
[... 161 lines cut ...]
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_NV_DX_interop
'''

pmjs1115, 2016-04-03:
As of April 3, 2016 this problem still exists with Windows 10, newly installed and updated in February. The fix also works.

michael13162, 2016-04-28:
April 28, 2016, and I'm having this problem on Windows 10 as well.

kintel, 2016-04-28:
@michael13162 Does the _Force Goldfeather_ option work? Are you using the latest drivers (Intel drivers are currently at [*.4380](https://downloadcenter.intel.com/download/25818/Intel-Graphics-Driver-for-Windows-7-8-1-10-15-40-?product=84188)) ?

(We're still short one Windows dev with an Intel GPU. I guess the former is harder to find than the latter)

michael13162, 2016-04-28:
Yes, _Force Goldfeather_ works, but I have no idea what it means.

Neon22, 2016-04-28:
nobody knows - and lots of people in lots of apps having similar diffculty with openGL drivers for intel chipset. not something we can fix in another way IMHO. we wait for new drivers from intel, or the heat death of the universe - whichever comes first.....

trayracing, 2016-04-29:
It broke between 2015.03 and 2105.03-1. That's not a big range of changes to look at.

kintel, 2016-04-30:
@trayracing There's nothing done between 2015.03 and 2015.03-1 that should affect this. Perhaps some small 3rd party library change is contributing to this issue. We're using MXE to build dependencies; perhaps something changed there.
If anyone has the time to compare Help->Library Info between the three versions 2015.03, 2015.03-1 and 2015.03-2, that would be helpful.

Also, if someone could point to the cheapest possible laptop which demonstrates this issue, that could also be helpful in terms of matching the necessary hardware with a developer.

@donbright @t-paul any idea of who may be interested in helping out if we could acquire the hardware?

jbeale1, 2016-06-26:
June 26 2016: I just upgraded my Dell Inspiron 15 Win7 (Pentium 3558U) laptop to Win10 and found this rendering bug. Selecting the option "Force Goldfeather" fixes it.
```
OpenSCAD Version: 2015.03-2
System information: Windows 8 (v6.2.0.0) 2 CPUs 3.91 GB RAM
Compiler: GCC "5.2.0" 64bit
GLEW version: 1.12.0
OpenGL Version: 4.3.0 - Build 10.18.15.4274
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
```

kintel, 2016-06-27:
@jbeale1 Could you give the latest drivers from Intel a spin? https://downloadcenter.intel.com/download/25977/Intel-Graphics-Driver-for-Windows-10-and-Windows-7-8-1-15-33-

jbeale1, 2016-06-27:
@kintel : This Dell laptop apparently needs Dell proprietary drivers, the generic Intel one warns that it has detected my laptop uses a customized driver and the standard Intel one may not work so I'm attempting to get the correct Dell driver but it's a bit of an adventure. Still trying...

jbeale1, 2016-06-28:
I tried to install the current Intel driver despite the warnings, and it refused to install saying "This computer does not meet the minimum requirements for this driver". Needs a newer CPU I guess (?)  I installed the current driver for my model laptop from Dell (Inspiron 3542
Service Tag: BMNY512), which is Intel HD version 20.19.15.4404 
Unfortunately, the problem remains, although I can still fix it with "Force Goldfeather".

trayracing, 2016-06-29:
jbeale, it's not uncommon. The intel installer appropriate for my HD 4400 on my HP refused to install, saying it hadn't been validated for my system. HP isn't inclined to update drivers, so I remain stuck at 10.18.15.4279 from a year ago. Any notebook vendors that don't play these games?

donbright, 2016-08-12:
@kintel it just so happens that my trusty old machine quit working recently and i had to buy a cheapo model from Staples and i got an HP 15-ac161nf which has an Intel HD  graphics thingy and it does in fact reproduce this bug on Windows 10(TM) 

@trayracing example box is invisible from the 'top' etc etc etc. It just disappears. Rotate to the side and it's fine. Hilarious.

the problem is i don't have much time... i shouldn't even be typing this. im supposed to be asleep. they dont like it when i start snoring in my cubicle. 

im also not supposed to be spending time describing what Goldfeather is. and that it's not a thing, it's a person. (Sorry alot of people know this already but someone above mentioned it so here goes)

It goes back to OpenCSG, check out this paper 

**OpenCSG: A Library for Image-Based CSG Rendering, by Florian Kirsch, Jürgen Döllner**

https://www.usenix.org/legacy/event/usenix05/tech/freenix/full_papers/kirsch/kirsch.pdf

Now as their paper describes, OpenCSG works by using various algorithms to fake 3d rendering using only 2d rendering tricks like stencil buffers and whatever. OpenCSG is designed to allow the user to choose which algorithm to use.  SCS is one, Goldfeather is another, 

Goldfeather is Jack Goldfeather... the lead author in a graphics paper that describes an algorithm. 
Here is the paper he wrote with Monar, Turk, and Fuchs about their ideas 

http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=28107&url=http%3A%2F%2Fieeexplore.ieee.org%2Fxpls%2Fabs_all.jsp%3Farnumber%3D28107

Goldfeather actually has a website! He is a professor in Minnesota. Check it out

https://apps.carleton.edu/profiles/jgoldfea/

------------

where i would start... with all this... is trying to get OpenCSG's example program (that comes with it's source code, which allows toggling different rendering algorithms) to build under MSYS2. Then see if it is reproducible problem there. narrow it down to whether it's OpenCSG or the way OpenSCAD uses OpenCSG.

urmilparikh, 2016-08-12:
Nice info, @donbright :)

kintel, 2016-08-12:
@donbright It would be really good to even ship a pre-compiled OpenCSG demo to help people confirm that this is reproducible.

trayracing, 2016-08-14:
@kintel here's the library diff:
```
$ diff *.txt
1c1
< OpenSCAD Version: 2015.03
---
> OpenSCAD Version: 2015.03-1
3c3
< Compiler, build date: GCC "4.9.2" 64bit, Mar 10 2015
---
> Compiler: GCC "5.2.0" 64bit
5c5
< Eigen version: 3.2.2
---
> Eigen version: 3.2.5
7,8c7,8
< OpenCSG version: OpenCSG 1.3.2
< Qt version: 5.4.0
---
> OpenCSG version: OpenCSG 1.4.0
> Qt version: 5.5.1
11,12c11,12
< GLib version: 2.42.1
< Application Path: C:/Users/moo/Programs/openscad-2015.03
---
> GLib version: 2.44.1
> Application Path: C:/Users/moo/Programs/OpenSCAD-2015.03-1-x86-64/openscad-2015.03-1
14c14
< Resource Path: C:/Users/moo/Programs/openscad-2015.03
---
> Resource Path: C:/Users/moo/Programs/OpenSCAD-2015.03-1-x86-64/openscad-2015.03-1
21c21
< C:/Users/moo/Programs/openscad-2015.03\libraries
---
> C:/Users/moo/Programs/OpenSCAD-2015.03-1-x86-64/openscad-2015.03-1\libraries
25c25
< GLEW version: 1.11.0
---
> GLEW version: 1.12.0

```

kintel, 2016-08-17:
@trayracing @donbright I believe the most likely is the OpenCSG update to 1.4.0. OpenCSG did update some core rendering code in 1.4.0 (a new fragment shader for ID color management), which could potentially trip up a shaky driver.

If we could isolate OpenCSG as the problem, I believe this could be possible to debug, especially if we could send an example app to the OpenCSG author :)

halxinate, 2016-08-18:
Sad. IG5500 with latest drivers and win10 - same issue. In addition, with the Goldfeather On, I'm having a ghosty flickering (on the preview rotation) "residue" in the removed volume, plus the color transparency doesn't work for the first object of the assembly (ladder), only for the first one. If you reorder them - ladder() becomes transparent, and the desk() - not (possibly an unrelated issue though). 

`$fn = 80;

difference() {
    assembly();    
    holes();
}

module assembly() {
    ladder();
    desk();
}

module ladder() {
    translate([2,0,0]) leg();
    translate([-2,0,0]) leg();
}

module desk(){
    difference() {
        translate([-2.5,0.5,-2]) color("yellow", 0.5) cube(size=[5,0.5,4]);
        translate([-1.5,0.5,-3]) cube(size=[3,0.5,2]);
    }        
}
module holes() {
    translate([2,2,1.5]) bolthole();
    translate([-2,2,1.5]) bolthole();
    translate([2,2,0.5]) bolthole();
    translate([-2,2,0.5]) bolthole();
}

module leg() {
    color("green",0.5) cylinder(h=10, d=1);
}

module bolthole() {
    rotate([90,0,0]) cylinder(h=5, d=0.2);
}`

trayracing, 2016-08-18:
>If we could isolate OpenCSG as the problem...
Not that I know anything about OpenCSG, but I just tried the OpenCSG example in 1.4.0 and it looks broken in 1.4.0 compared to 1.3.3. (Built in VisualStudio 2015).
1.4 shows a sphere, then cube, then nothing in a loop. 1.3.3 shows a CSG model spinning.
The OpenCSG web site says "On graphics hardware from Intel, OpenCSG does not always work well."

kintel, 2016-08-18:
If you can reproduce the problem with a pure OpenCSG example, that would be a good time to report this as an OpenCSG bug. It would be good if someone capable of reproducing this could report it, to be able to quickly respond to any ideas from the authors.

MichaelAtOz, 2016-08-18:
> plus the color transparency doesn't work for the first object of the assembly (desk), only for the second one. If you reorder them - desk() becomes transparent, and the ladder() - not (possibly an unrelated issue though). 

@halxinate see whether https://github.com/openscad/openscad/issues/310 explains what you see.

halxinate, 2016-08-18:
Yep, looks like exactly that 3+ years old issue (even though in my case all of the parts are colored and with 0.5 alpha). And it's definitely a separate bug. Thank you for the tip!

trayracing, 2016-08-18:
Here's a quick clip of the OpenCSG 1.4 example on Intel HD4400 Driver 10.18.15.4279.zip
(https://github.com/openscad/openscad/files/426063/OpenCSG.1.4.example.Intel.HD4400.Driver.10.18.15.4279.zip)

trayracing, 2016-08-18:
OK, I sent a short description of the situation and a video clip to the OpenCSG e-mail address.

kintel, 2016-08-19:
@trayracing thx

trayracing, 2016-09-09:
Good news! Florian of Openscad fame responded and has been able to reproduce the issue inside the test app. While Florian believes it to be a driver bug encountered by changes in the last OpenCSG release, Florian developed a workaround. Portability testing now with the intent of releasing an update very shortly.

kintel, 2016-09-10:
@trayracing Cool! Please keep us updated!

trayracing, 2016-09-10:
I confirmed Florian's change works locally in the OpenCSG example on my Intel GPU.

t-paul, 2016-09-11:
Very nice!

So now we need to get that update into MXE and update our build environment, oh my...

floriankirsch, 2016-09-11:
I have just released OpenCSG 1.4.1 - the same version I sent @trayracing for testing. The issue turned, in fact, out to be a driver issue of Intel, that was triggered with 1.4.0. A fragment program was introduced there, but still the fixed function OpenGL pipeline was used to setup vertex/texture coordinates. Somehow the Intel driver did not like this - the texture coordinates were just not set. I have now replaced the fixed function vertex setup with a vertex program.

trayracing, 2016-09-11:
Awesome - thank you very much, Florian!

t-paul, 2016-09-11:
@floriankirsch could you have a look if the project file patches needed for MXE (currently at 1.4.0) would make sense to be included upstream?

https://github.com/mxe/mxe/blob/master/src/opencsg-1-fixes.patch

t-paul, 2016-09-11:
https://github.com/mxe/mxe/pull/1521

floriankirsch, 2016-09-11:
@t-paul I'll consider this patch for the next release.

t-paul, 2016-09-14:
MXE is now updated to OpenCSG 1.4.1.

kintel, 2016-09-16:
@t-paul nice! Does this mean that we can spin up a Windows build for @trayracing and some of the others here to test?

t-paul, 2016-09-18:
Windows Snapshots with OpenCSG 1.4.1 (based on latest MXE)
* OpenSCAD-2016.09.18 - 32-bit [Installer](http://files.openscad.org/snapshots/OpenSCAD-2016.09.18-x86-32-Installer.exe) / [ZIP](http://files.openscad.org/snapshots/OpenSCAD-2016.09.18-x86-32.zip)
* OpenSCAD-2016.09.18 - 64-bit [Installer](http://files.openscad.org/snapshots/OpenSCAD-2016.09.18-x86-64-Installer.exe) / [ZIP](http://files.openscad.org/snapshots/OpenSCAD-2016.09.18-x86-64.zip)

trayracing, 2016-09-18:
Yay! Defect example program passes here with "force goldfeather" off. 
  HD Graphics 4400 
  Driver 10.18.15.4279
  OpenSCAD Version: 2016.09.18 (git a5e4def)
  Compiler: GCC "4.9.4" 64bit

BTW. I'm running Win10 anniversary edition, which is incorrectly reported in the lib info window as WIn 8:
   System information: Windows 8 (v6.2.0.0) 4 CPUs 5.93 GB RAM

[21 more replies]

closed 2016-09-18 by t-paul