openscad/openscad #1412: Unit tests fail with Z fighting issues

chrysn, 2015-08-11:
Both with hardware and software rendering, I'm getting failed unit tests with the current 2015.03-1 builds on Debian. The tests opencsgtest_issue1165 and {opencsgtest,throwntogethertest}_issue1215 are failing constantly.

As they (at least the issue1215 tests) already show Z fighting, please clarify: Are they supposed to test for a particular behavior of the graphics card? My impression from issue 1165 and issue 1125 is that this is all about geometry calculation. Isn't that sufficiently tested in the tests that don't rely on OpenCSG?

For the time being, I've blacklisted those tests in the Debian builds (which for the upcoming release require the test suite to pass completely).

chrysn, 2015-08-11:
As github does not allow HTML uploads, I've attached an exemplary screenshot from the report.

[image]

donbright, 2015-08-14:
Is this running a software renderer or is there a hardware renderer used? Could you copy paste the renderer info (the top part of the html test?).

kintel, 2015-09-06:
This test is meant to test a non-manifold export issue (after F6). The preview case is likely to cause issues cross-platform due to coincident surfaces, so I guess we should disable the preview test.

closed 2026-02-07 by github-actions[bot]

energypantry, 2026-02-07:
Hi! I can take this on with a fast turnaround. Do you accept USDC/USDT (any EVM chain) for payment? If yes, I can start immediately.