openscad/openscad #1413: Unit tests fail in software renderer only

chrysn, 2015-08-11:
There are some tests that are failing for me using the software renderer (libgl1-mesa-swx11 10.6.3-1), but not when using a hardware renderer.

Affected tests are throwntogethertest_issue1089, openscad-cameyeortho_camera-tests and openscad-cameyeortho-viewall_camera-tests.

issue1089 already looks like Z fighting at the expected image. For the camera tests, my best guess is that a frustum is so large that the the projected solid is so slim (low Z range) that even there, Z fighting occurs.

How would this best be resolved?

For the time being, I've blacklisted those tests in the Debian builds.

chrysn, 2015-08-11:
As github does not allow HTML uploads, attached are screenshots of the test report.

[image]
[image]

donbright, 2015-08-14:
Your post of the cameye ortho test (the ball with three subtracted cylinders) reminds me a lot of the failures I used to get on a machine with a PPC64 central processing unit running a newer version of Fedora which was using a different software rendering backend based on Gallium. This was a few years ago.

I used to just ignore that failure on that machine, I never did figure out the problem.

kintel, 2015-09-06:
I agree that this looks like a z buffer resolution issue, probably caused by a combination of a shallow z buffer and bad clipping planes on our side.