openscad/openscad #1414: Display CSG and CSG Export should not require full Preview or Render operation

clothbot, 2015-08-11:
If I load up one of my Rather Computationally Intensive Examples - https://github.com/clothbot/ClothBotCreations/blob/master/examples/quantize.scad - in OpenSCAD UI, I currently have to launch a Preview (F5) or Render (F6) operation before I can display or export the unfolded CSG.

I am able to hit the little stop/cancel icon very early in the process and export the resulting CSG.  This suggests to me that the full Preview/Render is not really necessary.

Please decouple CSG display/export from requiring a Preview or Render operation.  I think command-line OpenSCAD CSG export already does this, so hopefully this is relatively trivial to implement.

Thanks!

kintel, 2015-09-06:
One way of doing this would be to add a "Compile" menu item, which simply compiles the code, making export CSG available, as well as the Design->Display* functions.

MichaelAtOz, 2015-09-06:
...and not do a compile with F5/F6 unless editor has changed. Currently I think F6 re-compiles even if a F5 has just done a compile, with recursion some compiles can be lengthy.

jordanbrown0, 2026-02-08:
OK, I have to take some of that back.  I experimented with more pathological cases (Menger Sponge, n=6) and got a CSG Products generation time of about 8s.  Still, that's not long, and only wanting to generate CSG or AST is a pretty corner case.

I'm still going to leave this closed, but if somebody wants to reopen it I won't fight them.