openscad/openscad #1415: Unit tests relying on font files should explicitly specify them

chrysn, 2015-08-11:
The font tesselation unit tests, which rely on a particular implementation of the Liberation Sans font, so ship the .ttf file in their testdata directory. They do not, however, specify to actually use that file; instead, they use the regular font path, and if the OpenSCAD shipped fonts are not installed (eg. on Debian, where it is policy to use the fonts installed by the font packages), the shipped .ttf files are unused.

t-paul, 2015-08-11:
This should be handled by https://github.com/openscad/openscad/blob/master/tests/test_cmdline_tool.py#L236 setting the font path to the test fonts.

chrysn, 2015-08-11:
Thanks, I'd have tried a different approach but I'll have a look at why this does not work for me.

chrysn, 2015-08-11:
Ah -- that probably broke since we're using the real ../openscad now and not ./openscad (the OPENSCAD_TESTING) in issue #1362.

I've created pull request #1416 for that; please verify that it works in the plain building as well, it solves the issue in my cases.

chrysn, 2015-08-11:
Hm, that still seems not right cut it in all cases. Please only take the time for testing this if it looks plausible to you already, otherwise the current pull request might be a waste of time.

kintel, 2015-09-06:
I guess the problem is that even when specifying font paths in the .scad file, these paths are simply just added to FontConfig as they are discovered, and any fonts already on the system with the same name will be preferred. I didn't analyze this properly, but I can see that the fonts bundled with the OpenSCAD repos are still preferred when running the test suite.

t-paul, 2019-11-22:
I believe that was fixed quite some time ago.

closed 2019-11-22 by t-paul