openscad/openscad #1425: Animate Filename Generation - Duplicate/Missing Filenames

DBramley, 2015-08-20:
Due to the 5 decimal place rounding used in the calculation of $t, filenames generated using the formula, round(Steps*$t), are not contiguous.

Taking two Steps values as examples:

1) Steps=996, expected output 996 frames.

1000 frames are saved but 4 of these frames are overwritten by subsequent frames with the same name. There are no obvious gaps in the output, but 4 frame files are in fact missing.

The required time increment per step is 1/996=0.001004... but this is effectively rounded to 5dp so it becomes 0.00100, which is smaller than the required time increment. When step 126 is reached, $t=0+(126-1)*0.00100=0.125 so the file name is 0.125*996=124.5 which becomes "frame00125". However at the next step, 127, $t=0+(127-1)*0.00100=0.126 so the file name is 0.126*996=125.496 which becomes "frame00125" again, overwritting the previous frame.

2)Steps=1005, expected output 1005 frames.

1000 frames are saved, but there are 5 gaps in the sequence of file names generated.  These are obvious on inspection of the output.

The required time increment per step is 1/1005=0.000995... but this is effectively rounded to 5dp so it becomes 0.00100, which is largerer than the required time increment. When step 100 is reached, $t=0+(100-1)*0.00100=0.099 so the file name is 0.099*1005=99.495 which becomes "frame00099". However at the next step, 101, $t=0+(101-1)*0.00100=0.1 so the file name is 0.1*1005=100.5 which becomes "frame00101", so file name "frame00100" is skipped.

Further details can be found here on the OpenSCAD Forum:
  http://forum.openscad.org/Bug-in-Animate-tp13528p13539.html

I believe, as doug.moen suggested, that this issue could be solved by holding the current step as an integer, Step,  $t is then calculated from this as Step/Steps, incrementing from 0/Steps to (Steps-1)/Steps. The filenames can simply be generated using Step.

kintel, 2015-09-01:
@DBramley If you have a dev setup, could you give #1431 a spin?
If not, tell us which OS you're on and we might be able to get a build going.

DBramley, 2015-09-03:
I do not have a development setup.  I am running under Windows 7 64bit. Very happy to test if a compiled build can be supplied.
Thanks.

t-paul, 2015-09-03:
Snapshot builds (2015.09.03): [ZIP](http://files.openscad.org/snapshots/OpenSCAD-2015.09.03-x86-64.zip) - [Installer](http://files.openscad.org/snapshots/OpenSCAD-2015.09.03-x86-64-Installer.exe)

DBramley, 2015-09-04:
Testing completed using  http://files.openscad.org/snapshots/OpenSCAD-2015.09.03-x86-64.zip running under Win 7 64 bit, as suggested by Torsten.

Animate now appears to produces the correct number of contiguous frames.
Thanks.

NOTES:
1) In MainWindow::updateTVal()
    txt.sprintf("%.5f", t >= 1.0 ? 0.0 : t);
The test for t>=1.0 should now be redundant since, t = anim_step / anim_numsteps,  and anim_step has a maximum value of anim_numsteps-1. Suggest simplify to:
    txt.sprintf("%.5f", t);

2) The value of $t used within OpenSCAD appears to be the display value, e_tval, which is rounded to 5 decimal places. This accuracy compromise appears to be unnecessary, suggest storing the calculated t value instead and using this for $t, in MainWindow::updateTVal()
    double t = 1.0 * this->anim_step / this->anim_numsteps;

3) User changes to e_tval do not appear to be range checked ( 0 <= e_tval <= (anim_numsteps-1)/anim_numsteps).  By changing e_tval while animate is active and e_dump is checked I have produced saved frames that have been evaluated with $t=6! This could also cause an issue in MainWindow::updatedAnimFps()
    this->anim_step = this->e_tval->text().toDouble() * this->anim_numsteps;

4) I believe that the Dump Pictures check box should be automatically unchecked once a full cycle of frames have been dumped, I can see no point in continuing to dump and overwrite identical frames already saved. This could be achieved in this manner:

DECLARE
int anim_dump_start_step;
boolean anim_dumping;

INITIALISE
this->anim_dump_start_step = 0;
this->anim_dumping = false;

MAIN BODY
if (e_dump->isChecked() && animate_timer->isActive()) {
    if (anim_dumping && anim_dump_start_step == anim_step){
        anim_dumping=false;
        e_dump->Uncheck(); // correct method?
    } else {
        if(!anim_dumping) {
            anim_dumping = true;
            anim_dump_start_step = anim_step;
        }
    glReadBuffer(GL_FRONT);
    QImage img = this->qglview->grabFrameBuffer();
    QString filename;
    filename.sprintf("frame%05d.png", this->anim_step);
    img.save(filename, "PNG");
    }
}

// *Also need to catch the user unchecking the e_dump box and reset anim_dumping to false.*

-ends-

kintel, 2015-09-04:
I've implemented your suggested changes - please give it a spin once the snapshot binaries make it online.

DBramley, 2015-09-04:
Thanks.  I will test and report once snapshot binaries available (@t-paul please can you send me a link when ready, thanks).

DBramley, 2015-09-24:
@kintel changes to animate tested using OpenSCAD 2015.09.20 (6144d45), built using MSYS2 under Win 7 64bit.

All testing produced contiguous file names from frame00000 to frame[Steps-1] as expected.
The new auto uncheck functionality for the Dump Pictures check box worked as expected.

While trying to break it I noted the following (sorry):
1) If Steps (anim_numsteps) is changed while animate is active and dump pictures is checked, dump pictures can fail to uncheck, I assume that anim_dump_start_step >= anim_numsteps occurs. Suggest setting anim_dumping=false in updatedAnimSteps(). This should cause the file dump cycle to restart and thus anim_dump_start_step to be updated.
2) $t within OpenSCAD appears to be limited to 5 dp precision, I assume that this is because it is set equal to e_tval not anim_tval.
3) the range clamp in updatedAnimTval() for Time permits anim_tval=1, the maximum value should be (anim_numsteps-1)/anim_numsteps (assuming that the user has entered a Steps value first).  This could lead to an extra frame being produced erroneously on the first animation cycle with $t=1, however it appears that updateTVal() is always called before the first frame is generated and this fixes the problem...?

nophead, 2015-09-24:
Shouldn't it be $t = step / (anim_numsteps -1) so it does go from 0 to 1 on
the last step? It seems odd to me that it would stop short, especial for an
animation with only a few steps.

DBramley, 2015-09-24:
If Steps is set to 10, OpenSCAD produces 10 frames: 0 to 9, $t goes from 0.0 to 0.9 in steps of 0.1 .

nophead, 2015-09-24:
But why does it stop at 0.9 Logically it should go 0 to 1 in steps of 0.11111111. Or set frames to 11 to get steps of 0.1.

nophead, 2015-09-24:
For example I have exploded diagrams. If I wanted to animate them imploding I would set exploded to 1 - $t. But it would always stop short of finishing on a fully assembled assembly.

kintel, 2015-09-24:
@DBramley 
1) Will fix
2) I think you're experiencing the limited precision of echo(), not the $t value. We use anim_tval internally
3) I think @nophead's solution is better as some animations would have an important last frame. I'll implement it accordingly

kintel, 2015-09-24:
@DBramley Did you build yourself? If so please give the current master a spin

DBramley, 2015-09-25:
@kintel Yes I did build it myself, and yes I will give the new version a spin.

Regarding whether $t should actually reach 1, I think the answer is - it depends on what you are doing:
1) If as per @nophead 's example the animation moves an object from a to b, then as suggested  $t going 0 to 1 makes life easier.
2) However if you wanted to spin an object the obvious code would be:
rotate([0,0,360*$t])
In this case you don't want $t to be both 0 and 1 because these correspond to the same position (0 deg==360 deg) and this would lead to a jerk in the animation as $t cycles.
3) In animation in general an n second clip does include the frame at time=0 but does not include the frame at time n, because if it did the clip would be longer than n seconds. E.g. for a 2 second animation at 12fps:
In animation controls set: fps=12 steps=24 (2*12)
In OpenSCAD code: time=2*$t;
here variable time goes from 0 to 1 11/12 in steps of 1/12 second as required, giving precisely 2 seconds of video.
4) Extending example 3 (this is real code that I currently use) if the OpenSCAD code uses a variable time to generate a long animation and you want to extract a section the following code is clear and works:
time=start_time+$t*(stop_time-start_time);
Further sections of video generated in this way can subsequently be concatenated without any frame loss or duplication. 

On balance I think $t should not reach 1.

nophead, 2015-09-25:
Perhaps there should be $frame and $frames as well as $t. Otherwise I don't see how both use cases can be catered for. For example in my case I would not be able to generate the last frame without knowing the number of frames in the animation.

DBramley, 2015-09-26:
Yes, I agree, in fact oddly I had already written some code using a DOS batch file and the OpenSCAD command line interface that provides the code in OpenSCAD with $frame & $frames (http://forum.openscad.org/Bug-in-Animate-tp13528p13533.html)! In this scenario the system no longer needs to provide $t, but an equivalent can be calculated by the OpenSCAD user code as either $frame/$frames or $frame/($frames-1) as required.

However this functionality would require an overhaul of the animation interface in OpenSCAD. There are a whole list of features that I would like to see added :-) :
1) allowing the resolution of the dumped png images to be set (this is possible from the command line interface, but animation is not accessible this way)
2) making $frames writeable, so that user code in OpenSCAD can set it - some of my current code calculates the required number of frames and then echos it to the console for the user to type into the Steps box... not ideal.
3) allowing animation to be controlled from the command line interface, e.g. option --frames=n and allowing user code to control the command line camera rotation and distance via $vpr & $vpd.

In the absence of an overhaul, the current animation interface appears to me to be set up primarily to produce endless loop animations, e.g.:
rotate([0,0,360 * $t]) cube(20,center=true);
Given this, it is most likely that the frames produced for $t=0 & $t=1 will be the same and not both wanted. Taking nophead's example but as an endless loop, the object starts exploded, implodes then explodes again, before repeating.

t-paul, 2015-09-27:
The `$vpX` variables are already writable. But the important thing to note is that this and the animation is currently strictly a GUI feature. It's probably possible to somehow fake it by manually setting `$t` via the command line interface, but that has nothing to do with the actual animation code.

DBramley, 2015-09-28:
@t-paul I think that the $vpX variables currently have no effect on the command line camera...?

@kintel before I test, what is your decision:
1) Revert $t to step/steps;
2) Keep $t as step/(steps-1); or
3) Revert $t to step/steps but deprecate it and introduce $frame & $frames (or $step & $steps, etc.) to replace it?
My preference order, best option 1st is: 3,1,2. Note 1 & 3 don't break existing code.
Thanks.

t-paul, 2015-09-28:
The `$vpX` variables are also GUI only. There is a `--camera` command line option to set the camera position and direction.

kintel, 2015-09-28:
@DBramley As a quickfix, I guess we need to implement 1) to keep backwards compatibility, and defer this to a separate issue discussing the new feature. This would allow us to merge this into the 2015.03 branch.

kintel, 2015-10-16:
@DBramley Finally got around to applying the backwards compatible quickfix. Does this work for you?

DBramley, 2015-10-22:
@kintel Sorry for delay, school holidays here.

Changes to animate tested using OpenSCAD 2015.10.18 (9a83b0b), built using MSYS2 under Win 7 64bit.

All previous issues identified (above) tested and no problems found, except precision of $t generated by Animate:

OpenSCAD test code:
$t=1/3;
t_3dp=floor($t * 1000)/1000;
t_diff=$t-t_3dp;
text(str("$t: ",$t));
translate([0,-20,0]) text(str("t_3dp: ",t_3dp));
translate([0,-40,0]) text(str("t_diff: ",t_diff));

Test 1, Preview (F5):
$t: 0.333333
t_3dp: 0.333
t_diff: 0.000333333

Test 2, Comment out line 1, "$t=1/3;", and animate with steps=3, output at step 1:
$t: 0.33333
t_3dp: 0.333
t_diff: 0.00033

I believe test 1 and test 2 should be equivalent and give the same results...?

kintel, 2015-10-22:
@DBramley Thanks for testing. The output precision is a different discussion and doesn't affect the animation internals.

closed 2015-10-22 by kintel