openscad/openscad #1426: Minkowski reduces facets in arc

dlaw, 2015-08-20:
When I take a Minkowski sum, arcs are not rendered with the correct number of facets. Each corner in this model should have hundreds of facets but here there are clearly only five.

[image]
```
$fn=1000;
minkowski() {
  circle(1);
  square(1);
}
```

I observed this issue in openscad 2015.03.01-3 from the Arch Linux repositories. The issue did not occur with whatever version of openscad I was using in late 2013.

Please let me know what other tests I can carry out or information I can provide to assist with tracking down this issue.

t-paul, 2015-08-20:
That's probably a duplicate of #999 and caused by the vertex cleanup mentioned there in the comments.

dlaw, 2015-08-20:
This is definitely a duplicate of that issue. Thanks.
(I searched before posting, but only under keyword "minkowski".)

I will close this issue and contribute to #999 instead.

closed 2015-08-20 by dlaw

t-paul, 2015-08-20:
Cool, thanks.