openscad/openscad #1427: Cant render on a Macbook with Yosemite

carribeiro, 2015-08-26:
I have a Macbook Pro running OS X 10.10.5. OpenSCAD runs fine except for rendering; I'm totally unable to render, it keeps running forever and doesn't finish. As a result I can't save STLs which is why I'm using OpenSCAD first place.

carribeiro, 2015-08-26:
Followup, the problem disappeared.

I tried to render several times, and it wouldn't work. I tried loading simple models, didn't render either. However, after I opened the preferences window the first (had not changed anything before), it worked, and keeps working.

Is there any chance that some preference isn't set up correctly and was set up just by opening the preference window?

kintel, 2015-08-28:
That's odd.
If there is such a preference issue, you should be able to reproduce by deleting  the file ```$HOME/Library/Preferences/org.openscad.OpenSCAD.plist```

If you manage to reproduce, we'd like to help figuring out what happens.

carribeiro, 2015-08-30:
It's really odd. I left the computer rendering for a few minutes and it
wouldn't move the progress bar, not a single bit. After it worked the first
time, it worked fine every time.

(by the way, I used OpenSCAD before on Windows and Ubuntu Linux, so I knew
what I was expecting in terms of processing time. But it was my first time
on a Mac).

kintel, 2015-09-06:
@carribeiro Could it be related to caching? We cache already calculated geometry. If you quit&restart OpenSCAD it would have to recalculate.

t-paul, 2019-11-22:
Closing as the original problem disappeared and no further reports.

closed 2019-11-22 by t-paul