openscad/openscad #1428: Feature: sphere approximation by tesselation of platonic solids

BagelOrb, 2015-08-27:
This feature request is two-fold:

1
----
Implement all/some of the platonic solids: https://en.wikipedia.org/wiki/Platonic_solid
Tetrahedron
Cube 
Octahedron
Dodecahedron
Icosahedron

e.g.
`dodecahedron(r=10);`

note that `cube(r=sqrt(3));` should be equal to `cube([2,2,2],center=true);`

2
----
Use these shapes to approximate spheres in a different way; tessellate the surface and move the vertices to the correct location.
- up to a certain depth, recursively do:
- cut up each face into triangles (cut up each triangle into 4 triangles)
- move the introduced vertices: resize the vector from the origin to the vertex so that it has the length of radius of the sphere

donbright, 2015-08-27:
Perhaps you might be interested in Kit Wallace's page? 

http://kitwallace.co.uk/3d/solid-index.xq

He implemented numerous math solids in OpenSCAD

BagelOrb, 2015-08-28:
Indeed, I am interested!
However, the generated code is quite involved!

I was searching for a native function.

The function `polyhedron(.)` can help me a lot.

t-paul, 2015-08-28:
For the platonic solids http://www.thingiverse.com/thing:11224 is another (simpler) option. With license set to PD there's also no issue with using the files. From a short look at the source, it seems to follow the definitions from the Wikipedia page quite closely.
I think implementing those solids in the OpenSCAD core is probably not needed, as it's easy to provide them via an user-space library.

About the sphere tesselation there was a (discussion on the mailing list)[http://forum.openscad.org/New-Algorithm-for-Spheres-td13306.html] not long ago. There's no final conclusion, but it might make sense to have something like `icosphere` in addition to the existing `sphere` as some other programs (Blender) do.

kintel, 2015-08-28:
I agree with @t-paul that this should be done in user-space, but ideally as a built-in user-space library, which is well-documented and maintained as part of OpenSCAD.

carycode, 2015-09-24:
I've been told that there are several different ways to subdivide a triangle and push the vertices out to touch the surface of the bounding sphere. In particular, if you divide a triangle into 4 identical smaller triangles and push out the 4 new vertices to the sphere, then divide each of those smaller triangles into 4 identical tiny triangles and push out to the sphere, many of the final vertices will be in slightly different positions than if you divided the initial triangle into 16 identical tiny triangles before pushing out the vertices. Some geodesic dome builders find it convenient to tweak the positions on the sphere even more -- nudging a vertex so one of its struts becomes exactly the same length as a strut elsewhere, so the entire dome can be built with fewer kinds of struts.

I agree with kintel that platonic solids and geodesic domes should be done as a well-documented user-space library that is included as part of the normal package when people download and install OpenSCAD.
(Is there some way we could improve the OpenSCAD kernel to make it even easier to use that library in my own parts, or to make it easier to implement such libraries? Improving the core parts of OpenSCAD in this way sounds a lot more useful to me than sticking in a few more arbitrary special case "primitives").

kintel, 2015-09-24:
@carycode In terms of standard libraries:
We already have MCAD. It's a loose collection or utilities, but @hyperair is trying to refactor and redesign some of that into a more coherent library: https://github.com/openscad/MCAD/tree/dev
An official OpenSCAD library needs careful design. We started collecting some ideas at https://github.com/openscad/scad-utils, but more time is needed to author a good API.

zougloub, 2022-05-02:
The current tessellation of `sphere()` uses a lot of triangles around the poles ; one that would be based on solid subdivision would take a tiny bit more generation time and slightly longer code, but would make further processing (eg. `minkowski()`) much faster, and I'm not even talking about aesthetics 🤡 . I'll try to convince a young friend to propose an implementation.