openscad/openscad #1429: src/openscad.cc:386: possible slow code ?

dcb314, 2015-08-28:
[src/openscad.cc:386]: (performance) Passing the result of c_str() to a function that takes std::string as argument no. 1 is slow and redundant.

Source code is

   handle_dep(filename.c_str());

kintel, 2015-09-06:
std::ifstream only takes std::string when using C++11. We still support building on C++98 compilers.

dcb314, 2015-09-06:
>std::ifstream only takes std::string when using C++11

I think this is irrelevant.

$ fgrep handle_dep `find ../BUILD/openscad-2015.03-1 -name \*.h -print`
../BUILD/openscad-2015.03-1/src/handle_dep.h:void handle_dep(const std::string &filename);
$ 

No mention of ifstream there, AFAIK.

kintel, 2015-09-06:
wops, was looking at the wrong line. You're right. will fix.

commit ed637eb6 (2015-09-06): Minor fix: No need to convert to char*. Fixes #1429
closed 2015-09-06 by kintel