openscad/openscad #1430: Segfault in OSX when using concat on more than one vector of vectors

robertmassaioli, 2015-08-30:
When created a vector of vectors and then tried to use open scad to contact them I got a segfault. You can [find the core dump here][1].

 [1]: https://drive.google.com/file/d/0B-HVUbRON_gcal9ITWZnWXFheFU/view?usp=sharing

robertmassaioli, 2015-08-30:
I am not sure where the problem is but it seems to be somewhere in this method: https://github.com/openscad/openscad/blob/master/src/polyset-utils.cc#L52

robertmassaioli, 2015-08-30:
An example to generate the problem would be:

    module ramp_cam(displacement = [ [0, 0], [1, 1] ], baseHeight = 1, peakHeight = 2, radius = 1, width = 0.5, segments = 36) {
       step = 1 / segments;
       baseAngle = 360 / segments;
       
       innerRadius = radius - width;
       outerRadius = radius + width;
    
       deltaHeight = peakHeight - baseHeight;
    
       inner_bottom = [ 
          for(i = [0 : segments]) let (
             currentAngle = i * baseAngle
             ) [innerRadius * cos(currentAngle), innerRadius * sin(currentAngle), 0]
       ];
    
       outer_bottom = [ 
          for(i = [0 : segments]) let (
             currentAngle = i * baseAngle
             ) [outerRadius * cos(currentAngle), outerRadius * sin(currentAngle), 0]
       ];
    
       inner_top = [ 
          for(i = [0 : segments]) let (
             currentAngle = i * baseAngle,
             currentHeight = baseHeight + deltaHeight * lookup(i * step, displacement)
             ) [innerRadius * cos(currentAngle), innerRadius * sin(currentAngle), currentHeight]
       ];
    
       outer_top = [ 
          for(i = [0 : segments]) let (
             currentAngle = i * baseAngle,
             currentHeight = baseHeight + deltaHeight * lookup(i * step, displacement)
             ) [outerRadius * cos(currentAngle), outerRadius * sin(currentAngle), currentHeight]
       ];
    
       echo("Made points");
       cam_points = concat(inner_bottom, inner_top, outer_top, outer_bottom);
       echo(cam_points);
       echo(len(cam_points));
       echo("Joined points");
    
       ppl = segments + 1;
       ibo = 0;
       ito = ppl;
       oto = ppl * 2;
       obo = ppl * 3;
    
       outside_faces = [
          for(i = [0 : segments]) let (
             index = i,
             nextIndex = (index + 1) % (segments + 1),
             bottom_right = obo + index,
             bottom_left = obo + nextIndex,
             top_left = oto + nextIndex,
             top_right = oto + index
             ) [ bottom_right, bottom_left, top_left, top_right ]
       ];
    
       bottom_faces = [
          for(i = [0 : segments]) let (
             index = i,
             nextIndex = (index + 1) % (segments + 1),
             bottom_right = obo + index,
             bottom_left = obo + nextIndex,
             inside_left = ibo + nextIndex,
             inside_right = ibo + index
             ) [ bottom_right, bottom_left, inside_left, inside_right]
       ];
    
       inside_faces = [
          for(i = [0 : segments]) let (
             index = i,
             nextIndex = (index + 1) % (segments + 1),
             inside_right = ibo + index,
             inside_left = ibo + nextIndex,
             top_left = ito + nextIndex,
             top_right = ito + index
             ) [ inside_right, inside_left, top_left, top_right]
       ];
    
       top_faces = [
          for(i = [0 : segments]) let (
             index = i,
             nextIndex = (index + 1) % (segments + 1),
             top_inner_right = ito + index,
             top_inner_left = ito + nextIndex,
             top_outer_left = oto + nextIndex,
             top_outer_right = oto + index
             ) [ top_inner_right, top_inner_left, top_outer_left, top_outer_right ]
       ];
    
       echo("Made faces");
       cam_faces = concat(outside_faces, bottom_faces, inside_faces, top_faces);
       echo(cam_faces);
       echo(len(cam_faces));
       echo("Joined faces");
    
       polyhedron(points = cam_points, faces = cam_faces);
    }
    
    //ramp_cam(segments = 100); // This one works, the next does not.
    ramp_cam(displacement = [ [0, 0], [0.2, 1], [1, 0] ], segments = 4);

Interestingly the problem only seems to arise when I put more data in. Aka, it works for the simple cases of the module that I have defined.

robertmassaioli, 2015-08-30:
Okay, I think that I have worked out the problem. I tried to triangulate a face that had no width. I wonder why that exception was thrown though.

t-paul, 2015-08-30:
I can reproduce that with both 2015.03 and 2015.03-1 on Linux, but not on master.

I think it's fixed by https://github.com/openscad/openscad/commit/e6989526ca1514a4b6e04354040c0292d0bbc8d1.

robertmassaioli, 2015-08-30:
That is good news; I'll get the fix in the next release then. In the meantime I have fixed it on my end by making sure that there are no faces with overlapping edges. Cheers!

kintel, 2015-08-31:
Closing as duplicate of #1325

closed 2015-08-31 by kintel