openscad/openscad #1433: Feature request: 3D mouse support

greysAcademicCode, 2015-09-06:
It would be great to see support for [a 3D mouse](http://www.3dconnexion.com/products/spacemouse.html) added.

Maybe some effort could be borrowed from the FreeCAD project for this.
[This seems to be the patch](https://github.com/FreeCAD/FreeCAD/commit/012e50c524967d952371267683fc096ccd1eb76e) that added 3D mouse support to FreeCAD.

t-paul, 2015-09-06:
The https://github.com/openscad/openscad/tree/inputdriver2 branch has the (incomplete) prototype based on some code from the mailing list. It's basically working on all supported platforms, but still needs lots of work to tune the behavior and provide a reasonable configuration GUI.

greysAcademicCode, 2015-09-06:
Great!!
I can't get that branch to build though:
```
make: *** No rule to make target 'org.openscad.OpenSCAD.xml', needed by 'openscad_adaptor.h'.  Stop.
```
Can you offer any advice?

t-paul, 2015-09-06:
Whops, that file was hidden by .gitignore. Please pull and try compiling again...

greysAcademicCode, 2015-09-07:
It works!
By default, the inputs are not mapped in an intuitive manner (for me), but I assume you know that.
Are the settings in the input tab of the preferences connected to anything?

t-paul, 2015-09-07:
I'm not yet sure about how to do the preferences, I guess there should be at least some meta data from the device to tell the number of axis/buttons and so on. Also we probably want to remember defaults for different devices. In the end, the UI will probably be a much more complex thing compared to the actual device event handling :-).

The current defaults work for my Space Mouse Wireless with the Joystick driver (I think without any jsdev axis remapping), it's quite likely those are different with other drivers:
* Translation: Axis1 / Axis 2 (inverted) / Axis 3 (inverted)
* Rotation: Aixis 4 / Axis 5 (inverted) / Axis 6 (inverted)

MichaelAtOz, 2015-09-07:
@greysAcademicCode some more info may be helpful.
t-paul said

> It includes some initial support for mapping events via configuration, but there is no GUI yet.
>
The default config should be added when running the new version the first time. Right now it's basically the axis number with +/- defining the
direction:
>
[HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\InputDriver\axis]
"translateX"="+1"
"translateY"="-2"
"translateZ"="-3"
"rotateX"="+4"
"rotateY"="-5"
"rotateZ"="-6"
>
[HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\InputDriver\button]
"1"="viewActionResetView"
"2"="viewActionViewAll"
>
The names for the button actions are currently just the ones listed in the UI description file, e.g.
https://github.com/openscad/openscad/blob/master/src/MainWindow.ui#L365

I'm not sure what the Linux equivalents are...

greysAcademicCode, 2015-09-11:
@t-paul

How can I help get this into master?

About the preferences, what you have right now looks fine to me from a UI stand point. It looks like it captures all the functionality I'd expect and I don't think it needs any re-design or anything. It just seems that the drop-down lists to configure the axes don't actually do anything (should they at this point?).

Would it be hard to add a tickbox option so that the control was relative to the current viewport instead of relative to the master XYZ axes?

Can you tell me where I can go to configure what the axes do (like in the windows example in the comment directly above this one)? I've looked in `~/.config/OpenSCAD/OpenSCAD.conf` but I don't see anything there.

FYI: I use a 3DConnexion wired Space Mouse Pro. I'm running Arch Linux and I use this open source driver: http://spacenav.sourceforge.net/ which runs some daemon.

t-paul, 2015-09-13:
Hmm, I guess I should create a pull request to track things...

The viewport relative logic is partially there but the whole calculation needs review across the different platform drivers.

One of the initial patches also had some logic to determine the dominant axis and restrict changes to that. I think that's probably a useful feature too.

The configuration UI should work in general, but maybe I broke the spacenavd driver with the latest changes. I'll have a look. Maybe try with spacenavd stopped, the device should show up in `/dev/input/js0` via the `joydev` kernel module then.

greysAcademicCode, 2015-09-14:
I'm not getting any `/dev/input/js0` with `joydev` loaded.
```
$ uname -a
Linux [redacted] 4.1.6-1-ARCH #1 SMP PREEMPT Mon Aug 17 08:52:28 CEST 2015 x86_64 GNU/Linux
$ sudo lsmod|grep joydev
joydev                 20480  0
```
and when I plug in the device I see:
```
Sep 14 17:17:27 [redacted] kernel: usb 1-1.1.1: new low-speed USB device number 9 using ehci-pci
Sep 14 17:17:27 [redacted] mtp-probe[1633]: checking bus 1, device 9: "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.1"
Sep 14 17:17:27 [redacted] mtp-probe[1633]: bus: 1, device: 9 was not an MTP device
Sep 14 17:17:27 [redacted] kernel: input: 3Dconnexion SpaceMouse Pro as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.0/0003:046D:C62B.0006/input/input19
Sep 14 17:17:27 [redacted] kernel: hid-generic 0003:046D:C62B.0006: input,hidraw0: USB HID v1.11 Multi-Axis Controller [3Dconnexion SpaceMouse Pro] on usb-0000:00:1a.0-1.1.1/input0
Sep 14 17:17:27 [redacted] colord[540]: /usr/lib/colord/colord-sane: error while loading shared libraries: libsane.so.1: cannot open shared object file: No such file or directory
Sep 14 17:17:27 [redacted] /usr/lib/gdm/gdm-x-session[868]: (II) config/udev: Adding input device 3Dconnexion SpaceMouse Pro (/dev/input/event0)
Sep 14 17:17:27 [redacted] /usr/lib/gdm/gdm-x-session[868]: (II) No input driver specified, ignoring this device.
Sep 14 17:17:27 [redacted] /usr/lib/gdm/gdm-x-session[868]: (II) This device may have been added with another device file.
```
Do you have any special udev rules or anything? Maybe js0 won't appear for all SpaceMice?

MichaelAtOz, 2015-09-15:
> FYI: I use a 3DConnexion wired Space Mouse Pro. 

That uses different USB VID/PID codes. See [this](http://www.3dconnexion.eu/nc/service/faq/show_faq/7ece50ed-0b39-b57e-d3b2-4afd9420604e.html?tx_sugarkbfaq_pi1[ddx_keyword]=HID)

As I understand it, the OpenSCAD implementation uses USB HID and looks for matching VID/PID, so they need to be hard coded. At the moment, only the two wireless codes are implemented.

t-paul, 2015-09-15:
The USB IDs are not used in the drivers used normally on Linux. Right now there are:
* HIDAPI - Used on MacOS and Windows - needs the [USB IDs](https://github.com/openscad/openscad/blob/inputdriver2/src/input/HidApiInputDriver.cc#L45) / it works on Linux too, but needs additional privileges, so it's not ideal for the user
* Joystick driver - uses the Linux joystick device (currently fixed `/dev/input/js0`)
* SpaceNav driver - using the spacenavd daemon
* DBus driver - Linux only / not for actual devices but for remote control

On Linux, the easiest way with the Space Mouse Wireless is interestingly to go though the Joystick driver which is normally enabled on most systems. Maybe the Space Mouse USB IDs are missing from the Linux driver. I'll try to find out...

greysAcademicCode, 2015-09-15:
In case it helps, here is my `lsusb` for my mouse:
`Bus 001 Device 004: ID 046d:c62b Logitech, Inc. 3Dconnexion Space Mouse Pro`

MichaelAtOz, 2016-01-30:
This also needs a calibration option. My 3D mouse has slight movement when in the neutral position with OpenSCAD.
The 3Dconnection software calibration fixes this for when using their driver.
I assume it just reads the current values and uses them as an offset to get an adjusted neutral. 
It would be good to store that in the registry - or wherever.

klin39, 2016-06-10:
I tried compiling the latest branch of inputdriver2 and openscad crashes whenever the wireless dongle is plugged in while I open a new file. If I don't plug in dongle, I can open the document, but the Space Nav doesn't work. Am I missing a step? I am on a Mac OS X 10.11 El Capitan, running the latest 3DxWare.

t-paul, 2016-06-11:
It can't use the proprietary driver on MacOS, so maybe it fails to open the low level HID device driver due to the other driver already claiming the device. Could be some silly bug though, I don't think it was ever run on OSX 11.10 until now.
What's the USB device ID of the dongle?

klin39, 2017-04-19:
The Product ID for the Wireless SpaceMouse (#3DX700043) during wireless mode is:
Product ID: 0xc62f
Vendor ID: 0x256f 

Is that the number that is relevant?

t-paul, 2017-04-19:
Yes, that's the relevant USB IDs, that one is ([at least now in the inputdriver3 branch](https://github.com/openscad/openscad/blob/e941cf9bb110f1219d352fdef8e16c54a3a5722b/src/input/HidApiInputDriver.cc#L58)) listed.

MichaelAtOz, 2017-09-05:
I'm closing this to get the (my) bounty back so I can retarget it better.
I'll reopen it.

MichaelAtOz, 2017-09-05:
Re0pening...

t-paul, 2018-10-13:
Closing as #2130 is merged to master.

closed 2018-10-13 by t-paul