openscad/openscad #1434: Simple model creates error in latest slic3r

jbrown123, 2015-09-06:
Environment:
* OpenSCAD version 2015.03-1 
* Slic3r v1.2.9
* Windows 7 Professional SP1 64 bit

I have a very simple model that produces an odd error in Slic3r.  The generated STL looks fine in STLview and in MeshLab (reports no problems).  However, slic3r hates it.

If I stick two spheres on the end of a cylinder, it fails.  If I hull the same two spheres, it works.

When I view this in slic3r it sees the failing model as being oddly hollow and broken in the 3D tab and non-existent in the preview or layers tabs.  If you load an STL with just the failing object, slic3r says "no layers found" or it just hangs "processing triangulated mesh".

The 3D tab looks like this:
[image]

The preview tab looks like this:
[image]

Here's the code:
```
$fn=64;

color("red")
translate([0, -20, 0])
fails();

color("green")
translate([0, 20, 0])
works();

module fails()
{
    translate([-32, 0, 0])
    rotate([0, 90, 0])
    cylinder(r=10, h=64);

    //hull()
    {
        translate([32, 0, 0])
        sphere(r=10);
        translate([-32, 0, 0])
        sphere(r=10);
    }
}

module works()
{
    /*
    translate([-32, 0, 0])
    rotate([0, 90, 0])
    cylinder(r=10, h=64);
    */
    
    hull()
    {
        translate([32, 0, 0])
        sphere(r=10);
        translate([-32, 0, 0])
        sphere(r=10);
    }
}
```

MichaelAtOz, 2015-09-06:
Two curved surfaces joining has always caused OpenSCAD problems.
The issue is shown below. If you further zoom in it is a hole in the mesh.
[image]

I think this needs to be fixed.

donbright, 2015-09-07:
I did not get quite the exact same bug... but i did get some weird stuff. 
Ubuntu 14.04 amd64 slic3r 

[image]
[image]

However when I try a workaround, altering the $fn to 70 or 50, it seems to work fine

[image]

However when I alter the Ffn to 32 i again get a weird bug

[image]

there might be a weird resonance with powers of two and floating point.

donbright, 2015-09-07:
Yeah uhm @kintel @t-paul if you are reading this.... do you remember when i thought i was being so clever and changed the grid size in grid.h to be a multiple of 2 and it made CGAL go a little bit faster?

well uhm... i think it also kind of could be blamed for this type of issue... the way that floating point 'imperfectly' approximates decimal fractions could actually have been introducing a little bit of chaotic random-ish effect in the floating point patterns generated by CGAL, and ... it might be that this randomness was helping to kind of smooth out the weird non-uniform 3 dimensional lattice generated by floating point mathematics... and that maybe when i forced the Grid to be spaced at  an exact power of two, some of that randomness was lost...

its kind of like how when people march on bridges, they break their regular walking cadence to avoid generating a resonance and breaking the bridge..  ... my ""brilliant idea"" to speed things up was the equivalent of having all the little numbers march in a perfect cadence but it winds up breaking the bridge in some situations. just a theory....

the reason i think i might be right is that i tested changing the grid back to 0.001 and 0.000001 and this bug disappears. 

in src/grid.h the original grid was 

    const double GRID_COARSE = 0.001;
    const double GRID_FINE   = 0.000001;

and my binary grid (to speedup cgal) was

    const double GRID_COARSE = 0.0009765625;
    const double GRID_FINE   = 0.00000095367431640625;

again, changing back to the original grid size seems to avoid this situation. (yes it could be argued it's a bug in Slic3r... because Meshlab seems to think there is no problem..... then again... it is Real Weird that slicer has this problem)

kintel, 2015-09-12:
@MichaelAtOz The "3 volumes" thing is actually normal. CGAL reports empty space as one volume + one volume for each of the objects. There shouldn't be a hole in the mesh though, but it might look that way due to a shading artifact because some of the faces are oriented ~90 degrees relative to the adjacent ones.

kintel, 2015-09-12:
@donbright Your analysis sounds correct. I don't think changing the grid it back is the best solution though, but I'm also not sure how to best fix this.
If we could reduce the test case to a small number of triangles, we could easier try to figure out why slic3r complains.

t-paul, 2015-09-13:
As reference, there's also the same issue in the Slic3r repo -> https://github.com/alexrj/Slic3r/issues/3058

MichaelAtOz, 2015-09-14:
I got the STL from the slic3r post, loaded in Netfabb, and while the post says it is OK in Netfabb...

As I said "Two curved surfaces joining has always caused OpenSCAD problems.
...If you further zoom in it is a hole in the mesh." - 

I can't find the reference, but I noted this some time ago where every n multiples of $fn caused a problem - I discussed it here or on the forum, it was similar design, my [printable sausage](http://www.thingiverse.com/thing:149630) (so around Sept 2013).

Netfabb (studio basic 4.9) shows holes and many shells. 
[image]

Interestingly when I export it locally (OpenSCAD  2015.08.02) I get different number of shells & holes (15 & 21).

Also, prob obvious, but from my sausage experiment, changing it as below, is a workaround.
```
module fails()
{
    rotate([360/$fn/2,0,0]) // this bit
    translate([-32, 0, 0])
    rotate([0, 90, 0])
    cylinder(r=10, h=64);

    //hull()
    {
        translate([32, 0, 0])
        sphere(r=10);
        translate([-32, 0, 0])
        sphere(r=10);
    }
}
```

MichaelAtOz, 2015-09-14:
oops, it says OK in Meshlab...

donbright, 2015-09-15:
Oooh sorry I missed that... hm. I chopped away some of the sphere and cylinder and found some strange patterns that seem to be related to $fn. 

meshlab
[image]

internal debug svg stuff:
[image]

Maybe it's the old 'too tiny triangles' thing... when the Revolved Sphere (Latitude-Longitude) facets get in a certain rhythm with the cylinder facets then i guess it is making things that are degenerate. You can kind of see the patterns in Michael's screenshots reflected here. 

maybe it would be possible to chop out just the right part of the sphere/cylinder to isolate the problem with slic3r.