openscad/openscad #1435: Faces that are not on the same plane in polyhedron should be autosplit into triangle faces.

robertmassaioli, 2015-09-07:
To see a better description of this issue view this StackOverflow question: http://stackoverflow.com/questions/32428638/openscad-do-polyhedron-faces-have-to-be-on-the-same-plane

I think that the problem is that using polyhedron with quad faces that do not lie on a plane causes issues when you try and generate an STL file but not in the preview window.

donbright, 2015-09-07:
can you post your .scad code? OpenSCAD should triangulate some quad faces, as long as the four points are close to coplanar. (in fact most quad faces in most .scad files will need to be triangulated because in floating point based 3d spatial geometry most groups of 4 points are not coplanar). perhaps there is something else going on?

robertmassaioli, 2015-09-07:
That is interesting. In that case the code is here @donbright: https://bitbucket.org/snippets/robertmassaioli/5RBnp

donbright, 2015-09-07:
This might be an issue with the orientation of polygons in the polyhedrons generated. 

 see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#polyhedron_-_polygon_orientation_issues

i broke down your scad to two items and hit 'view / thrown together' and there are a lot of pink/purple faces indicating a potential ordering problem. 

[image]

donbright, 2015-09-07:
Yeah here is an example...

under outside_faces if i change 
```
     [ bottom_right, bottom_left, top_left, top_right ]
```
to  
```
      [ bottom_left, bottom_right, top_right, top_left ]
```

i go from this
[image]

 to this
[image]

robertmassaioli, 2015-09-07:
Ah dang! Okay thankyou. I'm going back to use that tool now and try and fix it up! I'll let you know what happens.

robertmassaioli, 2015-09-07:
Okay, it almost works perfectly now. Only one issue left:

[image]

    Compiling design (CSG Tree generation)...
    Rendering Polygon Mesh using CGAL...
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction
    ERROR: Alternate construction failed. CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction

I think that the error is the Alternate construction failed line that makes one whole polyhedron not show. So I'm looking into that now. :)

robertmassaioli, 2015-09-07:
And I worked out the last problem:

    ramp_cam(displacement = bouncyDisp, segments = 360, baseHeight = 1, peakHeight = 3, radius = 2.1, width = 2);

If you chance the radius to be 2 (instead of 2.1) then there is no hole in the middle and you get one edge that is overlapped many many times. So that causes the issue. Not it works perfectly:

[image]

Although it says that it is not a valid 2-manifold. So I still may need to fix something there.

kintel, 2015-09-07:
The underlying geometry library we use (CGAL) is very picky about manifoldness. Even if it reports not to be 2-manifold, it might be perfectly 3D printable, if that's your goal.

robertmassaioli, 2015-09-07:
I'll be honest, until just now I had no idea what a 2-manifold is. Then I [read a useful page on the topic][1]:

> A surface is a synonym for a 2-dimensional manifold. Complex 2-dimensional (real 4-dimensional) complex manifolds are also called surfaces. This article deals with real, compact, connected surfaces. Unless stated otherwise (Sections 2 and 3), surfaces without boundary are considered.

Oh, so a 2-manifold is just a surface that you can move over in two dimensions. Well that is much easier to undestand. It made me think that the problem was that I simply had too many objects in my scene that were not connected in a simple way. Thus I commented out all but one and presto:

[image]

Looks like I still have a minor warning in my code but I can deal with that. :) Thanks for all of the help!

 [1]: http://www.map.mpim-bonn.mpg.de/2-manifolds

closed 2015-09-07 by robertmassaioli

Neon22, 2015-09-08:
This looks tremendously useful. I hope you can consider contributing it for us to use in some way. Cheers...

robertmassaioli, 2015-09-08:
Yep @Neon22. I'm going to open source it once I am done. It would be great for people to create Cam's easily in OpenSCAD. 

Is there any list / feed / reddit / something that I should post it to when it's ready for a first review?

kintel, 2015-09-08:
The OpenSCAD mailing list is a good place to post. See here for the web front-end: http://forum.openscad.org

robertmassaioli, 2015-09-08:
Okay, I put the two functions in a repo, pushed it to BitBucket, Wrote a quick blog post on why you might want to 3D print Cam's and then posted it to the mailing list: http://forum.openscad.org/Making-Cam-s-using-OpenSCAD-td13747.html

Read and enjoy.

TakeItAndRun, 2015-09-09:
I would post your program on thingverse.
The reach is much greater there.

-- 
stempeldergeschichte@googlemail.com <karsten@rohrbach.de>

P.S. Falls meine E-Mail kürzer ausfällt als Dir angenehm ist:
Ich probiere gerade aus kurze Antworten statt gar keine Antworten zu
schreiben.
Wenn Du gerne mehr lesen möchtest, dann lass es mich bitte wissen.

P.S. In case my e-mail is shorter than you enjoy:
I am currently trying short replies instead of no replies at all.
Please let me know, if you like to read more.

Enjoy!