openscad/openscad #1437: Feature request: The ability to output stl or model/poly files from within a script

queglay, 2015-09-10:
This would be very useful for dynamically updating exports of multiple stl files for complex models, and being able to specify the filename for the component exported.

It would also allow export of many components used in a complex model to be imported into other software for animation or 3D printing.

My personal desire to see this would be so that I could create dynamic assembly animations within a framework I've created in Houdini that will be useful for open source hardware collaboration and construction.

t-paul, 2015-09-10:
Could you please explain the workflow with a specific example? Calling OpenSCAD from a different program to generate multiple parts from a single model description file is already possible.

There are some open issues/requests to declare the parts in a more concise way in the model and support for exporting those in one go from the GUI, but this would help mostly improving the direct user experience, not so much the use case of OpenSCAD as back-end driven by some other software.

clothbot, 2015-09-11:
Thinking out loud, adding an explicit file="filename.stl" argument, or 'cache=true' or 'cache=stl | dxf | svg' to render() to write out hierarchically-prefixed cache files could be interesting.

You would need the hierarchical-prefix mechanism to unique-ify the resulting files, but the final Render stage could significantly speed up.  It could also serve as a framework for (explicitly) threading the generation of the sub-geometry.

queglay, 2015-09-12:
At the moment, as far as I can see when we generate geometry form the command line you can only output one .stl, but it would be nice if these file names could be defined from within the .scad script itself, allowing multiple .stl files written out for different parts at once.  having these parts written out to different locations allows other software to update its components when the .scad file changes.

t-paul, 2015-09-12:
File names inside the scad file can cause quite some trouble when rendering the model on some other system and/or some other platform. Generating multiple parts using multiple calls to the openscad binary should be simple and even work around the currently still single threaded evaluation.

Multiple calls have indeed the issue that for bigger models, the render time increases a lot as everything needs to be evaluated again. Persistent caching could help with that, but I don't think a simple file name will help with that either. The cache must consider all module input variables, otherwise it would read invalid geometry if anything changes. Some time ago, someone brought up the idea to use memcached which might be a nice option.

doug-moen, 2015-12-10:
On this forum thread, I made the same suggestion that clothbot made: http://forum.openscad.org/How-to-check-if-a-file-exist-Why-not-export-from-code-td14982.html#a15034

Similar to clothbot, I think render() can be extended to do persistent caching, with an additional name argument that is used to form the file name of the cached STL file. As T-Paul mentions, we also need to store dependency information along with the STL file, so that we can determine if the STL file needs to be regenerated, the first time we reference it. The make/dependency feature (-m and -d command line options) already introduces the concept of a dependency file, so maybe we can leverage this code.

As T-Paul mentions, there is a problem if we allow full pathnames to be specified, because the syntax is OS dependent.

Another problem with full pathnames is security. If this feature enables a script writer to overwrite an arbitrary file anywhere in the file system, then it can be used to write malicious OpenSCAD scripts that perform some kind of attack on the host system. For example, you could upload a script to Thingiverse that attacks the thingiverse server executing openscad scripts, by ovewriting critical files. Or you could attack home users downloading OpenSCAD scripts from a public repository.

But we can implement persistent caching in a way that doesn't make OpenSCAD scripts OS-specific, and doesn't open up this security hole.

My suggestion is to add a 'name=string' argument to render(), where the name is restricted to be a valid OpenSCAD identifier between 1 and 32 characters long. So the name can contain only letters, digits and underscores*. This name is then used as the basis for constructing the filename of the STL file and the deps file used for persistent caching.

*To avoid a potential security hole on Windows systems, we should disallow the following names: CON, PRN, AUX, NUL, COM1, COM2, COM3, COM4, COM5, COM6, COM7, COM8, COM9, LPT1, LPT2, LPT3, LPT4, LPT5, LPT6, LPT7, LPT8, and LPT9 (case insensitive).

closed 2025-09-01 by jordanbrown0