openscad/openscad #1438: "Middle button pans in the xy pane" - not.

MichaelAtOz, 2015-09-14:
I was browsing the code and found the bit below.
On my Win7/63 middle button does not pan.
With Shift it does zoom.
I tried back to 2013.06 and a recent snapshot too.

QGLView.cc
void QGLView::mouseMoveEvent(QMouseEvent *event)
...
~line 237
```
 } else {
      // Right button pans in the xz plane
      // Middle button pans in the xy plane
      // Shift-right and Shift-middle zooms
      if ((QApplication::keyboardModifiers() & Qt::ShiftModifier) != 0) {
	      cam.zoom(-12.0 * dy);
      } else {

      double mx = +(dx) * 3.0 * cam.zoomValue() / QWidget::width();
      double mz = -(dy) * 3.0 * cam.zoomValue() / QWidget::height();

      double my = 0;
#if (QT_VERSION < QT_VERSION_CHECK(4, 7, 0))
      if (event->buttons() & Qt::MidButton) {
#else
      if (event->buttons() & Qt::MiddleButton) {
#endif
        my = mz;
        mz = 0;
        // actually lock the x-position
        // (turns out to be easier to use than xy panning)
        mx = 0;
      }
```

* reminder, check wiki when fixed - I don't think that functionality is mentioned.

kintel, 2015-09-14:
..so what does the middle button actually do? I don't have a middle button : /

MichaelAtOz, 2015-09-14:
It's pretty much application specific. Netfabb does pan with Middle, rotate with Right, select with Left for example (PITA).
Currently plain Middle does nothing in OpenSCAD, but the code appears to suggest otherwise.

I've got 5 buttons and two scroll dimensions :)
Wish I could assign them in OpenSCAD; I could easily write a AutoHotkey script tho...just lazy...

kintel, 2015-09-16:
I guess that code bitrotted. Doesn't make a lot of sense to clean this up I think - the whole mouse mapping stuff should be handled once we get a better OpenGL toolkit with proper camera support and mouse handling.

t-paul, 2015-09-16:
Just and additional note... there are quite some changes in the inputdriver branch (mainly shuffling code around to allow external entry points) and the topic fits with those changes too... maybe we can put it on the todo list there once I've created the pull request.

buzztiaan, 2016-04-06:
just a note, to say, i'd love if middleclick-pan would just do the same as mousewheel-scroll <3

t-paul, 2026-11-04:
I think the behavior still exists, but we also do have this now:

[image]