openscad/openscad #1439: Feature request: preserve individual shapes which overlaps

leikarne, 2015-09-16:
Consider the following openscad code:

circle(r=10);
circle(r=5);
circle(r=2.5);

The way openscad currently handles this, is by silently performing a union of these primitives.

It would be nice if openscad would preserve the individual circles, as I use them as cutting paths for a laser-cutter.

If this feature would be consider, I am not sure what the best way to implement it would be.
Perhaps:
individual() {
 circle(r=10);
 circle(r=5);
 circle(r=2.5);
}

In the meantime, I do the following:
module tube(r1,r2) {
difference() {
circle(r=r1);
circle(r=r2);
}
}
tube(10,5);
tube(5,2.5);
tube(2.5,0);

The drawback here, is that the material will be cut twice at the intersecting tubes - possibly ruining the material.

I hope you understand what I am asking.

Regards Leikarne

jordanbrown0, 2022-10-20:
What you really want is a way to represent a circle as a line, rather than as a filled figure.  #99 and #508 talk about this.

elnerd, 2023-02-07:
I am closing this issue as its duplicate of others with better descriptions.

closed 2023-02-07 by elnerd