openscad/openscad #1441: Feature Request: inverse transparency operator

mrvn, 2015-09-24:
When working with multiple objects it is very helpful to make objects that are in the way transparent and you can mark them as such. But sometimes you want to make everything but one object transparent. A "show me this with everything else transparent" operator.

t-paul, 2015-09-24:
See also #824.

mrvn, 2015-09-24:
Good point. Sometimes you want to see a subset of objects and tagging each separately would make one camera for each, not one camera for all of them. So maybe cameras could have an optional ID, e.g.

    °<1>obj1;
    obj2;
    °<1>obj3;
    °obj4;
    °<2>obj5;

This would have 4 cameras. The main view, one camera tagged "1", one anonymous camera and one camera tagged "2".

(syntax suspect to change :)