openscad/openscad #1442: Feature request: multiple/extra cameras

mrvn, 2015-09-24:
When building complex things out of parts I often need to see specific parts. But also check how it all fits together. So I add and remove ! a lot. It would be cool if there was a way to create extra cameras. So for example:

translate([5, 0, 0]) cube([10,10,10]);
°translate([-15, 0, 0]) cube([10,10,10]);

would have the main camera show both cubes and have a second camera showing just the second cube in a second (sub)window. Both could be individually zoomed and rotated or they could optionally be slaved to always show the same direction.

That way one would be able to see both the part and the whole in a single glance.

t-paul, 2015-09-24:
I think we can move that idea into #755 which is basically the same request. I like the idea to use a modifier, although that would restrict to 2 windows.

closed 2015-09-24 by t-paul

mrvn, 2015-09-26:
My intention, athought not so clear, was to have multiple (as many as you like) windows and a way to have modifiers for extra cameras. But you could have multiple windows for the same camera too. E.g. showing top, front, side and perspective view.

jordanbrown0, 2025-09-01:
Contrast with @6151:  this one requests views on different variations of the model (e.g. print vs assembled, complete vs cutaway, complete vs isolated to one subassembly.  That's what #755 requests too.  #6151 suggests multiple views of *the same* model.