openscad/openscad #1443: Recursion error message identifies wrong function

grantstevens, 2015-09-27:
// This example program aborts with the message:
//      ERROR: Recursion detected calling function 'cos' 
// The offending function is identified as "cos", which is unhelpful in finding the error,
// since the program from which this example is derived has many calls to cos().
// A more helpful error message would be:
//      ERROR: Recursion detected calling function 'my_recursive_function' 

x = my_recursive_function (0);

function my_recursive_function (a) =
(cos (a) > 1) // always false; this is a program bug in my_recursive_function().
    ? 0
    : concat (0, my_recursive_function (a + 0.01));

t-paul, 2015-09-27:
It's obvious the `my_recursive_function` would be more helpful in this specific case, but mentioning `cos` is not wrong, it's where the stack overflow detection triggered. Just highlighting the top level function will not work either as it's easy to construct a case where that's not the most helpful message either.
It might be possible to collect maybe 5 or 10 function names when unwinding from the stack overflow and showing those.

grantstevens, 2015-09-27:
Good feedback - thank you!

Yes, I agree the stack trace is very helpful in the tools I use that offer it. Is it a possibility for OpenSCAD?

MichaelAtOz, 2015-10-06:
#741

kintel, 2015-10-07:
Stack trace would be a new feature.
In the meantime, as an improvement, perhaps we could report the deepest user-defined function found, assuming we don't have bugs in any built-in recursive function.

grantstevens, 2015-10-10:
That sounds good to me - thank you!

monstercolorfun, 2015-10-19:
Hi all, i just had the same error, i am making many primitives inside a loop using cos for their placement in space. Cylinder works fine, and sphere gives me- Recursion detected calling function 'cos' 

it seems that the cos of placing the sphere and the cos used internally to make the sphere are giving the error.

monstercolorfun, 2015-10-19:
I solved it using fibonacci sphere from thingiverse, top bit of programming.

monstercolorfun, 2015-10-19:
Oh sorry no i was programming too fast, i had written two sphere functions, one containing teh other. error doesnt exist.

kintel, 2015-11-15:
On second though; we don't actually maintain a separate expression stack, so we'd have to build out that functionality -> a bit much work.

MichaelPFrey, 2018-11-10:
#2547 partially addresses this by a different approach

MichaelPFrey, 2018-12-29:
#2645 is a stack trace implementation

MichaelPFrey, 2019-01-30:
I think we can now close this issue. The error messages may still identify the wrong function, but with line number and call trace, there is now enough context to figure out what is going on.

closed 2019-01-30 by kintel

jamessynge, 2019-02-10:
I'm wondering if the recursion detection has been a bit too aggressive. I had working models which stopped working in the last week (I'm using the night builds), with reports about recursion being detected, but the reports don't identify modules which are actually recursive. I do have helper modules that are sometimes used in what might appear to be a recursive fashion; for example, mirrored (shown below) is sometimes called in a nested fashion, but it isn't an infinite loop:

```
module mirrored(v=undef) {
  children();
  mirror(v=v) children();
}

mirrored([1, 0, 0]) {
  mirrored([0, 1, 0]) {
    translate([10, 10, 0]) sphere(d=5);
  }
}
```

This example doesn't rigger the warning, but is the kind of thing that I use which does. I'm wondering if others have been seeing the same issue, maybe with an easy repro (mine hasn't been easy to figure out so far).

MichaelPFrey, 2019-02-10:
@jamessynge 
- please open a separate issue
- please post the specific error message - I can tell a few things from that
- please state which specific openscad versions you are comparing
- please state your operating system

Next thing: The message is actually kind of wrong. **The warning is triggered, once we reach 90% stack usage.** The typical reason to hit the stack sealing is recursion. But you can run out of stack for other reasons, which also triggers the recursion error. (Yeah, that is kind of wrong and not really helpful)

Next thing: Are you under windows or linux? Windows and Linux use stack slightly different, so an example that fails under Windows might run without issue on Linux (or fail also, but with a different warning).

So what you might "feel" like a too aggressive recursion detection is more likely higher stack usage for some reason/recent change. Again: That can depend on Windows/Linux/Mac specific changes.

MichaelPFrey, 2019-02-15:
@jamessynge 
You are right, I now have a test case and an idea, see #2787

jamessynge, 2019-02-24:
Michael, thanks very much for following up on this. Sorry I didn't see your reply in time to follow the process before you came up with the test case. I was motivated to check this because I again ran into the problem.