openscad/openscad #1444: Semi-colon causes silent union?

WarlockD, 2015-09-29:
NOTE:  I havn't tried other versions, but this is on the x64 2015.03-1 Windows Version.

I am not sure if this is part of an existing issue or not but I have been having a bugger of a time with this.

To sum up, I just wanted to make a cylinder with a triangle grove near the middle.  I have part of it done but the issue is, completely by accident btw, I found the entire shape of the object changes if there is a semicolon or not.

First, this is what I want:

 cylinder(h=20,d=25);
 translate([0,0,20]) // No semicolon
union() {
    cylinder(h=2,d1=25,d2=23);
    translate([0,0,2]);
    cylinder(h=2,d1=23,d2=25);
}

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 42
Geometry cache size in bytes: 154712
CGAL Polyhedrons in cache: 43
CGAL cache size in bytes: 6410824
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Simple:        yes
   Vertices:      148
   Halfedges:     588
   Edges:         294
   Halffacets:    296
   Facets:        148
   Volumes:         2
Rendering finished.
[image]

If I put the semi-colon next to translate....

 cylinder(h=20,d=25);
 translate([0,0,20]);  // semicolon
union() {
    cylinder(h=2,d1=25,d2=23);
    translate([0,0,2]);
    cylinder(h=2,d1=23,d2=25);
}
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 42
Geometry cache size in bytes: 154712
CGAL Polyhedrons in cache: 43
CGAL cache size in bytes: 6410824
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Simple:        yes
   Vertices:       60
   Halfedges:     180
   Edges:          90
   Halffacets:     64
   Facets:         32
   Volumes:         2
Rendering finished.
[image]

Saving it, opening and closing it doesn't seem to matter.

kintel, 2015-09-29:
This is not a bug:

If you put the semicolon directly after the translate, the translation will be applied to nothing (the semicolon ends the statement right there).
Your subsequent union is still being performed, it's just hidden inside your larger cylinder, i.e. not being translated.

I do agree that we could offer a warning when attempting to perform operations on nothing, as that may be a common error.

closed 2015-09-29 by kintel

TakeItAndRun, 2015-09-29:
This would be against the policy of OpenSCAD to fail silently on errors.  ;)