openscad/openscad #1445: dxf export... compatibility issues.

zanglerecke, 2015-09-30:
some entities are not right, or problematic as some other Programs are not capable of displaying it
klayout/ freecad/ ...
-> export as svg import in inkscape, export as dxf-R14 solves the problem, but thats little bit anoying

further suggesstions:
1. declare at the save button which version of dxf will be exported.
2.  only line entities -> polyline and text entities are realy simple done in dxf

t-paul, 2015-09-30:
Sorry, but "some issues" with "some other programs" is not very helpful for getting problems fixed. 

Can you please provide some more details, e.g. see the [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs.3F) for how to find the version information.

Do you have some (simple) example scripts that produce the issue?

More general improvements to the DXF import/export is on the wishlist: https://github.com/openscad/openscad/wiki/Project%3A-Improve-DXF-import-and-export

zanglerecke, 2015-10-02:
the issue is that it could not be opend, or importet. 
with "some issues" i was trying to say, im not able to find the source of the problem and have no clu where it can come from.

some other Programs: klayout, and freecad for example. 
in contrary to them Autocad 2015 is able to read it and: if i just open it, rezoom and save it, those other programs are able to open the file. 

my conclusion is that something is wrong in the dxf file.

the Version of my used Programs are:
OpenSCAD Version: 2015.03-1
System information: Windows 7 (v6.1.0.0) 2 CPUs 3.87 GB RAM
Compiler: GCC "4.9.2" 64bit
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.42.1
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\...\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/.../Documents/OpenSCAD/libraries
User Config Path: C:\Users...\AppData\Local/OpenSCAD
Backup Path: C:/Users/.../Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/.../Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

klayout:0.24.1 r2982

zanglerecke, 2015-10-02:
module PC(Nr_of_Rows, Grid_size, rad) {
    union(){
       for (y=[(-1)*(Nr_of_Rows-1):1:Nr_of_Rows-1]){  // y-LINE        
           rowlengh=(Nr_of_Rows*2)-abs(y);
       for (x=[((-1)*rowlengh/2)+1:1:(rowlengh/2)-1]){   // x-ROW
                translate([x * Grid_size,  y * sqrt(3)/2 * Grid_size,0])
                         circle(r = rad*Grid_size);         
}}}}

module hex(size){
    union(){for(r=[-60, 0, 60]) rotate([0,0,r]) square([size*tan(30), size],true);
}}

module hexring(in_size,wall){
difference(){
    in_size=in_size*2;
    union(){for (r=[-60,0,60]) rotate([0,0,r]) square([(in_size+wall)*tan(30), in_size+wall],true);}
    union(){for (r=[-60,0,60]) rotate([0,0,r]) square([in_size*tan(30), in_size],true);}
}}

module ring(rad,wall){
difference(){     
        circle(rad); 
        circle(rad-wall); 
}}

module spirale(){
    r=100;
    k=0.005346798;
    A=100;
p=1;
psi=137.5077;
dia=500;
     
union(){
    for( f= [16:1:377]){    
        translate([A*f*exp(k)*cos(psi*f),   A*f*exp(k)*sin(psi*f),   0]) {
        color("black"){circle(r*5);}
        }}
    }

translate([0,0,20]){
   color("red"){  
    difference(){     
        circle(r*dia); 
        circle(r*(dia*0.95)); 
}}}

}

$fn=64;
a=2.6;
max_device_size_to_corner=56.00000000001;
num=ceil(max_device_size_to_corner/a);
if (ceil(max_device_size_to_corner/a)==floor(max_device_size_to_corner/a)){
    echo("exact given Size, will be 1 Row smaller -> ERROR!");
   }

echo(num);
rad=0.283;
device_distance=400;
nr_of_devices_per_row=20;
rad_beg=0.2;
rad_end=0.4;
delta_rad=(rad_end-rad_beg)/(nr_of_devices_per_row);
for(step=[0:1:nr_of_devices_per_row-1]){translate([step*device_distance,0,0]){
rotate(90)union(){
        PC(num,a,rad_beg+(step*delta_rad));
}}}

TakeItAndRun, 2015-10-02:
Your program doesnt run.
There are a lot of * missing.
Like:

*(in_size+wall)*tan(30)

for (y=[(-1)*(Nr_of_Rows-1):1:Nr_of_Rows-1]){ // y-LINE*

t-paul, 2015-10-02:
The missing `*` were caused by the markdown formatting, fixed now.

kintel, 2015-10-02:
I could reproduce in KLayout.
I couldn't get the DXF importer to work in FreeCAD.

Does anyone know what versions of DXF are supported by these two tools?
@zanglerecke What other tools didn't manage to import the DXF?

t-paul, 2015-10-02:
Import works fine  with LibreCAD, although it's really slow due to the huge number of lines generated by the OpenSCAD export.

FreeCAD uses: https://github.com/yorikvanhavre/Draft-dxf-importer

kintel, 2015-10-02:
Btw., any DXF file should fail, e.g. this fails in KLayout too: ```circle();```

I tried to install the dxf importer in FreeCAD, but FreeCAD keeps trying to auto-install it and fails, telling me to manually install it, which I did. Perhaps a platform-specific issue..

Neon22, 2015-10-02:
Inkscape was able to read but shows as many individuallines with no structure.
On saving the files are very different. I have not analysed further.

DorianPIERRE, 2018-11-08:
It's the same for me ...
I export form openscad a dxf and when i open it in illustrator, it import it as text file but if i open an other sourced dxf file it's good. It seeam like a little mistake that unvalidate the openscad dxf fiels ...