openscad/openscad #1446: Menergschwamm segfaults at depth 4

mrvn, 2015-10-02:
// The following segfaults at depth 4

overlap = 0.123456789;

module MengerCut(full_size, depth, x, y, z, size) {
    translate([x + size, y + size, -overlap])
        cube([size, size, full_size + 2 * overlap]);
    translate([x + size, -overlap, z + size])
        cube([size, full_size + 2 * overlap, size]);
    translate([-overlap, y + size, z + size])
        cube([full_size + 2 * overlap, size, size]);
    if (depth > 1) {
        s = size / 3;
        d = depth - 1;
        for (w = [0 : 2]) {
            for (v = [0 : 2]) {
                for (u = [0 : 2]) {
                    if (  ((u != 1) && (v != 1))
                       || ((u != 1) && (w != 1))
                       || ((v != 1) && (w != 1))) {
                        MengerCut(full_size, d, x + u * size, y + v * size, z + w * size, s);
                    }
                }
            }
        }
    }
}

module Menger(full_size, depth) {
    difference() {
        cube([full_size, full_size, full_size]);
        MengerCut(full_size, depth, 0, 0, 0, full_size / 3);
    }
}

// depth 3 works fine
// Menger(27, 3);
// depth 4 sdegfaults
Menger(81, 4);

kintel, 2015-10-02:
This is a stack overflow occuring during normalization

donbright, 2015-10-03:
confirmed on ubuntu linux 

```
(gdb) r
Starting program: /home/don/src/openscad/binmaster/openscad y.scad -o y.png
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Compiling design (CSG Products normalization)...

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff2a4fc86 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff2a4fc86 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff2a527b0 in malloc () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff330fdad in operator new(unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00000000004a4ee9 in CSGTerm::createCSGTerm (
    type=type@entry=CSGTerm::TYPE_DIFFERENCE, left=..., right=...)
    at ../src/csgterm.cc:93
#4  0x00000000004aa43b in CSGTermNormalizer::match_and_replace (
    this=this@entry=0x7fffffffba90, term=...)
    at ../src/csgtermnormalizer.cc:132
#5  0x00000000004abb9b in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:79
#6  0x00000000004abc11 in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:87
#7  0x00000000004abc11 in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:87
#8  0x00000000004abc11 in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:87
#9  0x00000000004abc11 in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:87
#10 0x00000000004abc11 in CSGTermNormalizer::normalizePass (
    this=this@entry=0x7fffffffba90, term=...) at ../src/csgtermnormalizer.cc:87
#11 0x00000000004abc11 in CSGTermNormalizer::normalizePass ( . . . 
```

thehans, 2015-11-15:
Increasing the stack size with `ulimit -s 32768` lets me preview it without crashing on ubuntu.  I didn't try actual render though.

Here's another menger approach that uses nothing but union(implicit).  It can preview n=4 without hitting the default stack limit.  Render takes about 1hr and 8.3GB RAM though.

```
module mengerSponge(x,y,z, s, n) {
  translate([x,y,z]) {
    if (n>0) {
      i = n-1;
      s2 = s*2;
      scale(1/3) {
        mengerSponge( 0, 0, 0, s, i);
        mengerSponge( s, 0, 0, s, i);
        mengerSponge(s2, 0, 0, s, i);
        mengerSponge( 0, s, 0, s, i);
      //mengerSponge( s, s, 0, s, i);
        mengerSponge(s2, s, 0, s, i);
        mengerSponge( 0,s2, 0, s, i);
        mengerSponge( s,s2, 0, s, i);
        mengerSponge(s2,s2, 0, s, i);
        mengerSponge( 0, 0, s, s, i);
      //mengerSponge( s, 0, s, s, i);
        mengerSponge(s2, 0, s, s, i);
      //mengerSponge( 0, s, s, s, i);
      //mengerSponge( s, s, s, s, i);
      //mengerSponge(s2, s, s, s, i);
        mengerSponge( 0,s2, s, s, i);
      //mengerSponge( s,s2, s, s, i);
        mengerSponge(s2,s2, s, s, i);
        mengerSponge( 0, 0,s2, s, i);
        mengerSponge( s, 0,s2, s, i);
        mengerSponge(s2, 0,s2, s, i);
        mengerSponge( 0, s,s2, s, i);
      //mengerSponge( s, s,s2, s, i);
        mengerSponge(s2, s,s2, s, i);
        mengerSponge( 0,s2,s2, s, i);
        mengerSponge( s,s2,s2, s, i);
        mengerSponge(s2,s2,s2, s, i);
      }
    } else {
      cube(s);
    }
  }
}

mengerSponge(0,0,0,81,4);
```

Dont even think about n=5 though

kintel, 2015-11-15:
There's also https://github.com/openscad/openscad/blob/master/examples/Old/example024.scad

thehans, 2020-06-04:
Normalization stack issues were fixed by #2886 and #3122

closed 2020-06-04 by thehans