openscad/openscad #1448: crashes when polyhedron has a collapsed face

mrvn, 2015-10-05:
I've build a polyhedron shaped like a cube with one side smaller than the opposite side. The size is parametric depending on the size of the surrounding structure.

Now to do a print test I scaled down my object to the point where the smaller side of the polyhedron became a line instead of a rectangle. Effectively the whole thing became a 3 sided prism. I'm guessing that does bad things to calculating the normals and triangulation and openscad crashed.

TakeItAndRun, 2015-10-05:
Would you mind shareing your code?

mrvn, 2015-10-06:
w = 1; // crash with w = 0

    polyhedron(
        points=[[0,     0, 0],
                [1,     1, 0],
                [1 + w, 1, 0],
                [2 + w, 0, 0],
                [0,     0, 1],
                [1,     1, 1],
                [1 + w, 1, 1],
                [2 + w, 0, 1]],
        faces=[[0, 3, 2, 1],
               [0, 1, 5, 4],
               [1, 2, 6, 5],
               [2, 3, 7, 6],
               [3, 0, 4, 7],
               [4, 5, 6, 7]]);

kintel, 2015-10-06:
Duplicate of #1325 - fixed in master

closed 2015-10-06 by kintel