openscad/openscad #1449: CGAL errors rendering polygon mesh

mrvn, 2015-10-05:
I'm playing around with making my own threaded rods (screws) and nuts and I run into the following:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe != 0 File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 212

mrvn, 2015-10-05:
// Copyright 2015 Goswin von Brederlow <goswin-v-b@web.de>
// License: GPL

module Screw(
    length=50,                      // axial length of the threaded rod
    pitch=10,                       // axial distance from crest to crest
    minor_radius=5,                 // radial distance from center to groove
    major_radius=10,                // radial distance from center to crest
    profile=[[0/4, minor_radius],   // profile of the thread
             [1/4, minor_radius],
	     [2/4, major_radius],
	     [3/4, major_radius]],
    lead_in_start=270, 	  	    // degrees for the thread grow to start at bottom
    lead_in_end=630,   	  	    // degrees for the thread grow to end at bottom
    lead_out_start=720,	  	    // degrees (from end) for the thread grow to start at top
    lead_out_end=360,  	  	    // degrees (from end) for the thread grow to end at top
) {
    echo("Screw(", length=length, pitch=pitch, minor_radius=minor_radius,
    	 major_radius=major_radius, profile=profile,")");

    /* profile of thread
     *      |           |
     *      |           |
     * 1.00 |      1.00 |_____
     *       \               |
     *        \              |
     * 0.75    \             |
     *          |  0.66      |
     *          |           /
     * 0.50     |          /
     *         /          /
     *        /    0.33  /
     * 0.25  /          |
     *      |           |
     *      |           |
     * 0.00 |      0.00 |
     */

    // number of facets going around the screw
    facets = max(3,				// Use at least 3 faces
                 ($fn > 0) ? $fn		// If $fn is set then use it
                 : min(2 * PI * major_radius / $fs, 360 / $fa)); // use $fs or $fa
    echo(facets=facets);

    // helper functions
    function scale(r, a, b, z) = (z - a) / (b - a) * r + (1 - (z - a) / (b - a)) * minor_radius;
    function radius(i, r, z) =
        ((z < -pitch + pitch * lead_in_start / 360) || (z > length - pitch * lead_out_end / 360)) ?
	    minor_radius
	  : (z < -pitch + pitch * lead_in_end / 360) ?
	      scale(r, pitch * (lead_in_start / 360 - 1), pitch * (lead_in_end / 360 - 1), z)
	    : (z > length - pitch * lead_out_start / 360) ?
	        scale(r, length - pitch * lead_out_end / 360, length - pitch * lead_out_start / 360, z)
              : r;
    function point(i, r, z, base) = [radius(i, r, base) * cos(i * 360 / facets),
    	     	      	       radius(i, r, base) * sin(i * 360 / facets),
			       z];
    function range(n, i=0, step=1, vec=[]) =
        (n <= 0) ? vec : range(n - 1, i + step, step, concat(vec, [i]));
    function map(indexes, val, i=0, res=[]) =
        (i >= len(indexes)) ? res : map(indexes, val, i + 1, concat(res, [val[indexes[i]]]));

    // polygon for thread
    num_loops = floor(length / pitch + 2);
    num_thread_points = len(profile);
    slices = num_loops * num_thread_points;
    num_points = facets * slices + 1;

    echo(num_loops=num_loops, num_thread_points=num_thread_points, slices=slices);

[... 96 lines cut ...]
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
	translate([-major_radius - overlap, -major_radius - overlap, length])
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
    }
}

translate([40, 0, 0]) {
    difference() {
        Screw(length=length, pitch=pitch, minor_radius=minor_radius, major_radius=major_radius, profile=profile, lead_in_start=0, lead_in_end=0, lead_out_start=0, lead_out_end=0);
	translate([-major_radius - overlap, -major_radius - overlap, -pitch - overlap])
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
	translate([-major_radius - overlap, -major_radius - overlap, length])
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
    }
}

translate([65, 0, 0]) {
    difference() {
        Screw(length=length, pitch=pitch, minor_radius=minor_radius, major_radius=major_radius, profile=profile2, lead_in_start=0, lead_in_end=0, lead_out_start=0, lead_out_end=0);
	*translate([-major_radius - overlap, -major_radius - overlap, -pitch - overlap])
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
	*translate([-major_radius - overlap, -major_radius - overlap, length])
	    cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
    }
}

donbright, 2015-10-06:
not sure if this will help but it looks like some of your polyhedron faces might be oriented backwards (they are purple in ThrownTogether render)

[image]

i chopped down the last part of your code to get an easier look:
```    overlap = 0.123456789;
    length=10;
    pitch=10;
    minor_radius=5;
    major_radius = 10;
    mid_radius=8;
    profile = [[0/3, minor_radius],
               [1/3, minor_radius],
           [2/3, major_radius],
           [3/3, major_radius]];
    profile2 = [[0/6, minor_radius],
                [1/6, minor_radius],
            [2/6, major_radius],
            [3/6, major_radius],
            [4/6, mid_radius],
            [5/6, major_radius],
            [6/6, major_radius]];

    translate([20, 0, 0]) {
        difference() {
            Screw(length=length, pitch=pitch, minor_radius=minor_radius, major_radius=major_radius, profile=profile);
        translate([-major_radius - overlap, -major_radius - overlap, -pitch - overlap])
            cube([2 * major_radius + 2 * overlap, 2 * major_radius + 2 * overlap, pitch + overlap]);
        }
    }
```

mrvn, 2015-10-06:
That is indeed the cause for the errors. openscad should catch that though.

kintel, 2015-10-07:
There are many things we could catch. They all require some sort of mesh analysis. Once we have the mesh analysis in place, that could also be used to validate imported meshes and even sanity-check exported meshes. See e.g. #590 #348 #1042.
I'll close this, assuming that one of the above issues is good enough for now. If we need a separate issue for detecting winding order of user-defines polyhedrons, please open one so we can keep track of that.

closed 2015-10-07 by kintel

mrvn, 2015-10-08:
Maybe catch the error from cgal (not analysing the mesh) and add "This might be cause by inside out faces" to the output so people know to fix their mesh.

kintel, 2015-10-08:
@mrvn I'm not  a huge fan of "this might be.." error messages. We've done that in the past, with way too many false positive, causing even more confusion.
Color coding in preview mode is probably a cleaner quick strategy,