openscad/openscad #1451: does not set window title

mrvn, 2015-10-06:
When I open a file in openscad (from command line or menu) the Window title still reads "Untitled" instead of the filename. When multiple openscad windows are open this gets confusing.

kintel, 2015-10-06:
Which version of OpenSCAD is this and on which OS?

mrvn, 2015-10-06:
OpenSCAD version 2015.03-1 on Debian GNU/Linux amd64.

t-paul, 2015-10-06:
Works for me on Debian testing using the version from the Debian repository.

Please add some more details.
* What are the exact steps to show the issue?
* Which window manager / desktop environment is running?
* Where is the package coming from? If not the one from the Debian/testing repo, please post the `Help->Library` Info content.

mrvn, 2015-10-07:
- "openscad foo.scad" or selecting a file in the open dialog shows the issue.
- qvwm, no desktop environment
- % apt-cache policy openscad
openscad:
  Installed: 2015.03-1+dfsg-2+b2
  Candidate: 2015.03-1+dfsg-2+b2
  Version table:
 *** 2015.03-1+dfsg-2+b2 0
        500 http://ftp.de.debian.org/debian/ sid/main amd64 Packages
        100 /var/lib/dpkg/status

I guess it could be my window manager but it works in every other application. xterms even set a new title every time I cd into a dir. But I will try a different WM tonight.

jordanbrown0, 2025-03-09:
Seems to work correctly on Windows.  Perhaps somebody with a Linux rig handy can test, confirm, and close this.