openscad/openscad #1452: Feature: support for anaglyph 3D

mrvn, 2015-10-06:
In case anaglyph means nothing to you that is 2 pictures superimposed, one with a red tint and one with a blue tint. If you are wearing 3D (red/blue colored) glasses each eye sees only one image and you see 3D.

https://en.wikipedia.org/wiki/Anaglyph_3D

josefpavlik, 2015-11-04:
Interesting idea.
I've start to work on it, you can try it on https://github.com/josefpavlik/openscad , branch anaglyph.
For now it's only test, the anaglyph view is always on, now I'm working on enabling/disabling etc.

hzeller, 2016-07-09:
This looks awesome @josefpavlik !
I enjoyed quite a bit looking at my existing OpenSCAD models with this change.

It also patches against current master with only a few lines offset.

I'd like to encourage you to get it pull-requested into mainline!

shalnoff, 2017-06-22:
@josefpavlik thank you for great improvement! Unfortunately, as I see, it is still not included in the main branch. So I just would like to ask whether you solve conflict issues mentioned here https://github.com/openscad/openscad/pull/1500? 

Very desirable feature.

josefpavlik, 2017-07-25:
I've resolve the conflicts, now it's usable with latest version.
Please write some comment to discussion on https://github.com/openscad/openscad/pull/1500 to encourage the maintainers to include it to main version.