openscad/openscad #1453: Missing dependencies ubuntu 14.04.3 LTS

bigbear3001, 2015-10-11:
When building on my almost freshly installed ubuntu 14.04.3 not all needed dependencies where added by uni-get-dependencies.sh.

Missing dependencies that where not present on my system:
* qt4-qmake
* libqscintilla2-dev

bigbear3001, 2015-10-11:
i guess the error message for libqscintilla2-dev was just a warning message:
```
root@358f6fd07a66:/opt/openscad# qmake
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
WARNING: /opt/openscad/scintilla.pri:13: Unable to find file for inclusion qscintilla2
Project MESSAGE: Using local copy of qscintilla2.prf instead.
```

bigbear3001, 2015-10-11:
This can also be solved by installing qt5-default which sets qt5 as default for qmake.

Closing because i'm not sure if qt5-default should be added as a dependency.

closed 2015-10-11 by bigbear3001

donbright, 2015-10-13:
if it is this confusing, it is a bug. it is supposed to "just work". i will take a look this week.

donbright, 2015-11-14:
fixed by https://github.com/openscad/openscad/pull/1486 see also #1474