openscad/openscad #1454: unexpected result is unexpected

greysAcademicCode, 2015-10-11:
The following code takes a sphere then subtracts the same sphere rotated by 90 degrees about the x axis from it:
```
difference() {
 sphere(1);
 rotate([90, 0, 0]) sphere(1);
}
```
My expectation is that the spheres subtract cleanly from each other because they should have rotational symmetry about any axis passing through their center. Instead I get art:
[image]
I also get a warning: `WARNING: Object may not be a valid 2-manifold and may need repair!` which I though might be telling me that the objects did not subtract cleanly, but that warning comes and goes depending on the rotation angle I choose and I often get no warning when the spheres don't subtract cleanly.

The above is obviously a contrived example for illustrative purposes. Here's an example of one small spot in a big complicated model getting messed up by arc approximation. Unexpected stuff like this is really hard to track down: https://github.com/lautr3k/RepRap-iTopie/issues/7

I'm a new openSCAD user and maybe it's my fault for not knowing its quirks & limitations. I was aware that openSCAD approximates arcs, but I am not aware of all the ways this limitation can rear its ugly head in practice. These failure modes can be very unexpected and problematic.

Neon22, 2015-10-11:
5 fold symmetry (your top face is a pent) rotated 90 degrees will overlap poorly. Try an even number.

Neon22, 2015-10-11:
you set that with $fn=32 (say) in case you don;t know

jordanbrown0, 2025-09-01:
Unless we eventually gain the ability to process true circles and spheres (vs approximations built out of line segments), this can't happen.  Particular cases can work - if your $fn is a multiple of 4, you can rotate by 90° - but it can't work in general.

closed 2025-09-01 by jordanbrown0