openscad/openscad #1455: concave minkowski crash in CGAL

kintel, 2015-10-15:
The following code causes a crash in CGAL.
Note: It appears that this CGAL exception isn't catchable since the CGAL's Nef_polyhedron_3 destructor crashes during unwinding.

```
module object() {
     difference() {
          translate ([0, 0, 12]) cube(30, center=true);
          translate ([0, 0, 12]) cube ([20, 10, 10], center=true);
          translate ([0, 0, 3])  cylinder ( h=10, r1=8, r2=8, center=true);
     }
     
     difference() {
          translate ([0, 0, 0])  cube ([18,18,25], center=true);
          translate ([0, 0, 12]) cube ([20, 10, 10], center=true);
     }
}

minkowski() {
     object(); 
     cube(2);
}
```

```
CGAL error: assertion violation!
Expression : e_below != SHalfedge_handle()
File       : /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h
Line       : 657
Explanation: 
Refer to the bug-reporting instructions at http://www.cgal.org/bug_report.html
OpenSCAD(42174,0x7fff73440310) malloc: *** error for object 0x7fd610d3bc10: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
```

donbright, 2015-10-16:
simplified code to crash...
```
module object() {
     difference() {
          cube(5);
          cube(3);
          translate ([0, 0, 2]) cube (2);
     }

     difference() {
          cube (5);
          cube (3);
     }
}

minkowski() {
     object();
     cube();
}

```

it appears there is a stray vertex introduced somehow  in the union of the two main shapes.

This is a simple render of just the object() with wireframe CGAL mode turned on. 

```
   difference() {
          cube(5);
          cube(3);
          translate ([0, 0, 2]) cube (2);
     }

     difference() {
          cube (5);
          cube (3);
     }
```

[image]

The number of vertices should be 14 but it is listed as 15, and if you look closely there is a stray 'yellow' dot in the middle of open space. 

In fact that dot appears to be at the exact corner where the two 'negative cubes' of the first object meet each other.... 

[image]

[image]

If we look at the two objects by themselves... it appears that perhaps the similarity between the two is a key factor.

[image]

kintel, 2015-10-16:
Thanks for the analysis!
This CGAL assertion isn't even catchable (due to another issue in CGAL I think).
One interesting observation: Internally, we make a copy of a CGAL Nef_polyhedron_3 prior to performing  the minkowski sum. If I pass the original object instead of the copy, it doesn't crash. I'll submit that separately to CGAL and keep digging.

kintel, 2015-10-16:
See https://github.com/CGAL/cgal/issues/422

donbright, 2015-10-16:
hey that is pretty awesome that CGAL is on github!

donbright, 2015-10-16:
I also checked OpenSCAD 2013.06, in the original .scad code at top, it segfaults. 

With the simplified code to crash, it produces this error

CGAL error in CGAL_Nef_polyhedron's minkowski operator: CGAL ERROR: assertion violation!
Expr: N.is_valid(0,0)
File: /usr/include/CGAL/convex_decomposition_3.h
Line: 161

It also shows 15 vertices after union, as in the above example. 

I dumped the "sphere map" of the vertices too, it apparently shows the "stray vertex" at 2,2,3  is different than the others, but i'm not sure if it matches the "antenna" example in the CGAL documents.  http://doc.cgal.org/latest/Nef_3/index.html#Chapter_3D_Boolean_Operations_on_Nef_Polyhedra

```
<!-- vertex sphere map begin. vertex counter is 1
<!-- vertex coordinates: 0,0,5-->
Sphere Map - Statistics
|V| = 3
|E| = 6
|L| = 0
|F| = 2
|Fcs| = 2

 the sface cycles of sface 0 start with an
  shalfedge from 5,0,0 to 0,5,0
 the sface cycles of sface 1 start with an
  shalfedge from 0,5,0 to 5,0,0

  vertex sphere map end -->

 . . . ( skip listing vertexes 2 thru 13 ) . . . 

<!-- vertex sphere map begin. vertex counter is 14
<!-- vertex coordinates: 0,3,3-->
Sphere Map - Statistics
|V| = 3
|E| = 6
|L| = 0
|F| = 2
|Fcs| = 2

 the sface cycles of sface 0 start with an
  shalfedge from 0,-15625,0 to 0,0,-625
 the sface cycles of sface 1 start with an
  shalfedge from 0,0,-625 to 0,-15625,0

  vertex sphere map end -->

<!-- vertex sphere map begin. vertex counter is 15
<!-- vertex coordinates: 2,2,3-->
Sphere Map - Statistics
|V| = 0
|E| = 0
|L| = 2
|F| = 2
|Fcs| = 2

 the sface cycles of sface 0 start with an
  shalfloop lying in the plane 0/1 0/1 244140625/1 0/1
 the sface cycles of sface 1 start with an
  shalfloop lying in the plane 0/1 0/1 -244140625/1 0/1

  vertex sphere map end -->
<!-- sphere map end -->

note that
|V| =   number of svertices
|E| =  number of shalfedges: 
|L| = number of shalfloops:  
|F| = number of sfaces: 
|Fcs| = number of sface cycles: 
```

This would seem to indicate the sphere map of the Nef Polyhedron3 vertex at 2,2,3 has no svertices and no shalfedges.

kintel, 2015-10-16:
Interesting. Perhaps it would be possible to fix that CGAL issue by manually cleaning the Nef_3 polyhedron before attempting a convex decomposition (since we try to avoid using CGAL's minkowski anyway).

kaklik, 2018-04-10:
Hello, is there some progress on this bug? I have another very simple code which produces the CGAL crash in the case where rendering is executed by F6.

    difference (){
        minkowski()     
        {
            difference ()
            {
                cube([68, 969, 3], center = true);
                translate([26, 0, 0.5])
                    cube([4, 2*969, 2], center = true);
            }
            rotate([90,0,0])
                cylinder(r=1, h=1, center = true);
        }
        cylinder(r = 1, h = 2*969, center = true);
    }

The error console output follows: 
`Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251 `

kintel, 2018-04-10:
@kaklik Which version of CGAL are you using? (Help->Library Info)

kaklik, 2018-04-10:
OpenSCAD Version: 2018.04.10.nightly (git 9bd8e87)
System information: Linux 4.4.0-119-generic #143-Ubuntu SMP Mon Apr 2 16:08:24 UTC 2018 x86_64 Ubuntu 16.04.4 LTS 4 CPUs 7.69 GB RAM
Compiler: GCC "5.3.1 20160413" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.0

kintel, 2018-04-10:
@kaklik That system is a bit old. Could you try this with a system that has CGAL-4.8, as this issue should have been resolved there.

t-paul, 2018-04-10:
I got the same error on Debian/testing which has 4.11. Strangely it went way after rendering the parts separately. And I don't mean just the error hidden, it produced the (I believe) intended result without missing cuts.

t-paul, 2018-04-10:
Oh, it's actually like that:
1) run F6 without F5 first -> no error
2) run F5, then F6 -> error -> flush cache -> F6 -> no error

kaklik, 2018-04-10:
@t-paul  Yes, I could confirm this behavior on my system too.  Running into the error is annoying, because it needs the flushing caches, or restarting the OpenSCAD.  It is trivial on small designs, but problematic in the case of projects with long rendering time.

kintel, 2018-04-29:
@kaklik @t-paul Does the OpenSCAD process actually crash for the last example, or does it just spit out a CGAL error and keep running?

MichaelAtOz, 2018-04-29:
This rings a bell, F5 -> F6 thing, I had the issue with render(), where preview with render(), which didn't invoke CGAL (clipper??), but cached the geometry, then F6, CGAL had issues with cached geometry, but fresh F6 worked, or the other way around...I'll try to find it.

t-paul, 2018-04-30:
For me no crash, only CGAL error.

t-paul, 2018-04-30:
Oh, yes, I think there was a similar effect discussed on the forum not long ago.

dstulken, 2019-07-10:
Bringing this bug report up to date, here are some test results with OpenSCAD 2019.05 (Mac build):

- The code from @kintel in the first post causes OpenSCAD to crash, on both preview and render, even after cache flushes.
- The simplified code from @donbright has the exact same behavior.
- The code provided by @kaklik is now functional in both preview and render modes.  No cache flushing is necessary to have a successful render after previewing.

I found this bug report as I have my own model code which has similar behavior: successful preview, console error and abort on render, but successful render with no errors in the console if caches were flushed prior to render.   So while none of these original test cases show it anymore, I can confirm that issues do still exist that require the workaround that @t-paul mentioned above.

kintel, 2024-03-26:
Just for completeness: This is not an issue when using Manifold

stefano2734, 2026-09-20:
Last test here with CGAL 6.0.
Actual is CGAL 6.2.1 but here 6.1 or 6.2 system dependent.