openscad/openscad #1456: Cannot compile

boxcarmib, 2015-10-15:
make ends with:

g++ -c -m64 -pipe -fno-strict-aliasing -frounding-math -DEIGEN_DONT_ALIGN -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/freetype2 -O2 -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -DQT_WEBKIT -DOPENSCAD_VERSION=2015.10.15 -DOPENSCAD_SHORTVERSION=2015.10.15 -DOPENSCAD_YEAR=2015.0 -DOPENSCAD_MONTH=10.0 -DOPENSCAD_DAY=15.0 -DDEBUG -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/eigen3 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/qt4 -Isrc -I/usr/include/qt4 -Isrc/libtess2/Include -I/usr/X11R6/include -Iobjects -o objects/cgalutils-applyops.o src/cgalutils-applyops.cc
{standard input}: Assembler messages:
{standard input}:1610: Warning: end of file not at end of a line; newline inserted
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Makefile:2223: recipe for target 'objects/cgalutils-applyops.o' failed

donbright, 2015-10-16:
how much RAM do you have?

boxcarmib, 2015-10-16:
1024MB

kintel, 2015-10-16:
If you're doing a parallel make, it might help turning that off. gcc uses a lot of memory when compiling against CGAL.

donbright, 2015-10-16:
also, remove the "-pipe" flag in the Makefile (it looks like you removed -O2 and -g already?)... -pipe causes the compiler to pipe output to the assembler, which can use more RAM than simply storing the intermediate assembly code as a file on the disk. 

i dont think it will build with g++ in 1024MB unless you have a massive amount of Swap... now that being said.... i have built it before on a remote machine with low RAM that had SSD disk drives... i created a massive swap file (several gigabytes) using the standard linux trick of dd/mkswap/swapon/blahblahblah and was able to build in a reasonable amount of time.  with spinny-disk swap will take a very, very long time. of course you probably have to be root to mess with swap... 

clang compiler might work a lot better, it uses much less RAM. don't know if it can pull it off with only 1G though....

boxcarmib, 2015-10-16:
I was just using a slice on a hosting server... they have hourly rates... so i'll bump it up to 8GB and see how it goes.

boxcarmib, 2015-10-16:
Thanks for the help... seems to have compiled and installed fine with 2G.
I never thought memory problem... I guess I would have expected a malloc err or something.

closed 2015-10-16 by boxcarmib

donbright, 2015-10-17:
cool beans!