openscad/openscad #1457: Difference cod

durgesh5597, 2015-10-20:
I on my laptop it won't reconginize the difference code but if I uses the same file on my desktop it works. Please fix it. suppose to look like this.
[image]

But it shows like on my laptop. 
[image]

t-paul, 2015-10-20:
Please add the information listed in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs.3F including the OpenGL information.

What happens with a full render (F6) - this will drop the color information, but does it also show different results?

kintel, 2015-11-15:
@durgesh5597 Any more info from you to help us locate this?

t-paul, 2016-01-24:
Closing as there is not enough information to reproduce the issue.

closed 2016-01-24 by t-paul