openscad/openscad #1458: CGAL error in CGAL_Nef_polyhedron3()

nortcele, 2015-10-21:
On a Macbook pro, OS X 10.11 (EL Capitan), with OpenSCAD 2015.03-1.

With the following code:
```OpenScad
cylinder(r = 5, h = 10);
rotate_extrude()
translate([5, 0, 0])
scale([2, 1, 1])
circle(d = 4);
```
If I hit the render button, all is fine and the objects appear correctly.

If I change `circle(d = 4)` to  `circle(d = 5)`, when I hit the render button, only the cylinder is rendered and the log show the error:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-1/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326
```

If I hit the button again, the error doesn't appear in the log but only the cylinder is rendered. To see the error again, I have to quit an relaunch OpenSCAD.

donbright, 2015-10-24:
Bug confirmed. 

Workaround: change translate([5,0,0]) to translate(5.001,0,0])

I believe this is happening in the following location in the source code file cgalutils.cc

                bool err = CGALUtils::createPolyhedronFromPolySet(psq, P);
                 if (!err) {
                        PRINTDB("Polyhedron from psq is closed: %d", P.is_close$
                        PRINTDB("Polyhedron from psq is valid: %d", P.is_valid($
                 }
                PRINTD("CGALUtils::createPolyhedronFromPolySet success");
                if (!err) N = new CGAL_Nef_polyhedron3(P)

This assertion happens, i believe, in the process of creating the new CGAL_Nef_polyhedron3() from the "plain" CGAL Polyhedron P. 

So perhaps CGAL's Nef Polyhedron code does not like the CGAL "plain" Polyhedron being fed to it.... the 'plain' Polyhedron is created by cgalutils-polyhedron.cc 

(re errors disappearing.... I have also seen errors disappear, i believe it may be because of the CGAL cache. Sometimes clicking 'design/flush cache' will help it reappear.)

kintel, 2015-10-25:
Closing as duplicate of #1081.
If you make sure that your donut shape doesn't touch the Y axis in a single point, it will work.

closed 2015-10-25 by kintel