openscad/openscad #1459: Sheet metal design

Harvie, 2015-10-21:
Hello,
i really enjoy using OpenSCAD for designing 3D printable solid models, however recently i became in need of designing parts from bended/folded sheet metal.

Eg.: i create some box from sheets like this:

[image]

And i make some cuts using boolean operations (you can see the little windows in sides of box). And now i need to magicaly unfold whole thing into 2D plan for cuting and folding my design. Something like this "cross":

[image]

I am aware that this is not exactly simple thing to implement. And i am also aware, that you cannot simply unfold some object that wasn't designed to be unfold from the beginning. But i'd really like to see some basic sheet metal functions.

Let's say that i'd like to use OpenSCAD for designing custom sheet metal case for raspberry pis or similar PCBs. So i'd like to make basic 3D box structure folded from "cross" cut out of 2D sheet. Then i'd like to use boolean to cut model of raspberry with all it's connectors into it, so it will make holes in the 3D box. and then i'd like to unfold it to 2D pattern again. ready to cut and fold.

I like the simple and programmer-friendly interface of OpenSCAD and there are currently no CADs for sheet metal which work similary to openscad. So i think this could be really great to have.

I've been thinking about how it can be implemented. And i think that it's all about implementing some kind of foldable sheet primitive that can be rendered both folded and unfolded. it's also important to be able to set thickness of sheet and minimal radius of fold. there is obviously quite a bunch of other things that have to be done to make this work well.

I know it's long term project. But please think about it.

Neon22, 2015-10-21:
A general approach to solving this problem is seen in programs like Pepakura.

The simplification they successfully use is to think of the object as an infinitely thin sheet. This then becomes unfoldable by finding each flat surface and its boundary.

Any thickness in the walls implies either a thin sheet (which you are trying for) or for a number of folded thin sheets (as though the walls really counted and they were solid). Check out Pepakura - there is a free viewer and many examples. OpenSCAD is very much the solid kind of approach. However the end result of openSCAD - a polygonal surface - can be unfolded.

So what you probably need is to get a polygonal representation of the final object and to unfold it.

Unfolding is done by:
-  Finding all the (lets assume triangles) that are in the same planes (have same normals) and grouping them together into notional flat surfaces.
-  Then finding the boundary lines between these flat surfaces.
-  Determining the angles between each surface across the boundary line.
-  Employ a fairly sophisticated algorithm to unfold each surface by that angle and have the resulting shape not overlap any other unfolded surface.
  -  Everytime there is an unavoidable overlap - split the already folded portion off and create a new sheet to continue unfolding on.

This is quite a complex algorithm to get working to generate the minimum number of folded pieces. Starting with a brute force approach where everything is unfolded from everywhere and then just keep the solution with the minimum number of sheets is the easiest to start with (assuming not too many surfaces). There is literature from Siggraph and IEEE on these issues but Pepakura is probably the single commercial solution out there that's in everyone's price range.

In summary - I am personally not sure OpenSCAD is the right tool for your specific example because OpenSCAD makes solids and not sheets and your example is about sheets.

However the output of OpenSCAD pumped into a polygonal unfolder would generate output you could further trim to remove the wall thicknesses.
Or perhaps if there was a way to output (in polygons) only the externally facing wall but I'm not sure this is even possible in OpenSCAD's current design. 

This discussion might be better on the Forum...

laird, 2015-10-21:
There are also mechanical complexities. For example when you bend the sheet-metal it stretches in the forms in complex ways, so if you bend a 4 inch long 1/8" sheet on a quarter inch radius 90° band, the resulting shape he is more than 4 inches long because of the stretching, the amount of which varies depending on the specific material and the exact way that it is bent.

There are commercial tools that deal with this sort of thing, but the ones that I'm familiar with their very very expensive.

Paper is easier because it is close enough to true to say that paper is zero thickness and does not stretch.

Harvie, 2015-10-22:
Yes, this is very important to handle properly. I am NOT talking about manifold model and decomposing it to infinitely thin polygons. The output should be different for sheets of various thicknesses and bends of various radiuses. eg.: on following example you can see both folded and unfolded model (it's both in one picture, which can be bit misleading, but i think you'll get it):

[image]

In short: While the cuting/folding plan is 2D, whole thing also has to fold and unfold seamlessly in 3D.

On the other hand: having openscad feature to generate paper cut-out models is also great idea and i'd like to have it for fun and presentation purposes. however the idea is quite different. as in such case you decompose the manifold model to single surface.

Neon22, 2015-10-22:
Well consider another approach then - where you add a primitive (perhaps cylinder encoding radius of fold with extra properties) everywhere you want a fold.
This would explicitly define what you intend and could be interpreted correctly to unfold.

TakeItAndRun, 2015-10-22:
The Wallace line is worth a look for unfolding folding:
http://kitwallace.tumblr.com/post/115676322654/nets-of-regular-solids

Even if this is not quiet what U want, this is for thin sheets

[image: image]

Neon22, 2015-11-09:
Just found this unfolding plugin for Blender - in case anyone wants to see how they unfold meshes using Python...
http://wiki.blender.org/index.php/Extensions:2.6/Py/Scripts/Import-Export/Paper_Model

Harvie, 2026-11-05:
> In PythonSCAD there IS an Option to Export any model to a foldable Papier model. Does this aus your Idea?

yeah, something like that would be cool.