openscad/openscad #1461: Feature request: openscad-core

mateddy, 2015-10-23:
Is it possible (for linux package) to break down Openscad into:
1. A core package, does not link to qt/glib/etc, which can be run in command-line, e.g.:
openscad -o thing.stl thing.scad

2. A full package, link to qt/glib/etc, which runs the GUI

t-paul, 2015-10-23:
No, it might be possible to create a version without Qt (the test suite used to do that), but most other libraries, e.g. glib are used by the core and can't be removed without impacting language features.

Would there be any additional benefits than just saving the installation of Qt libraries? The current version does work head-less via command line.

donbright, 2015-10-24:
I did a no-QT build two or three weeks ago in the test suite (doc/testing.txt, tests/). It still works but it takes a little fiddling around with CMakeLists.txt. It starts with passing special flag to cmake, something like cmake -DNOGUI 

You can also do a NULLGL test build that will not link to OpenGL and cannot produce any pixel-image output, only vector formats like SVG, STL, etc. Again, it takes some fiddling around... some of that stuff is not tested very often.

kintel, 2015-11-15:
You can build a non-gui version of OpenSCAD at config time: ```qmake CONFIG+=nogui```.
Whether or not this should be build and packaged for various distributions is a different question which is better debated per distro. Personally, I don't see anything that warrants spending much time on this as it looks like a very limited use-case. Perhaps distros which don't do Qt at all could benefit.

closed 2015-11-15 by kintel