openscad/openscad #1465: libqt5scintilla.so.11 called by name; new version works fine....

yerazunis, 2015-10-26:
OpenSCAD Version 2015.7.02  as is currently in the Debian downloads expressly requires libqt5scintilla2.so.11  .... which is an obsolete library.

But- good news; if you "ln -s" the current libqt5scintilla2.so to (falsely)represent itself as the ...so.11 version, it all works.   

So, the problem probably is that OpenSCAD's spec file demands =...so.11, when it really needs is the much less stringent >=...so.11

Hope this helps.
     - Bill

yerazunis, 2015-10-26:
I should have noted that I tested the above with libqt5scintilla2.so.12 .   Sorry.  I'm a little new at this.

t-paul, 2015-10-26:
On which Debian version does that happen?

Is that about the OpenSCAD nightly build? There is an issue on OBS that breaks either the Debian7 or Debian8 build. It's not yet clear yet how this can be resolved.

yerazunis, 2015-10-26:
Debian version:  current (as of 2015-10-26) debian "Testing" - repository is csail.mit.edu

The OpenSCAD version is 2015.07.02 (according to "about"), as fetched from the repository.

Interestingly, apt-get was perfectly happy installing the repo version of OpenSCAD with ...so.12 installed, even though the actual binary complained when it couldn't find ...so.11 .

t-paul, 2015-10-26:
Strange, I'm running Debian testing too and I get:
```
$ apt-cache policy openscad
openscad:
  Installed: 2015.03-1+dfsg-2+b2
  Candidate: 2015.03-1+dfsg-2+b2
  Version table:
 *** 2015.03-1+dfsg-2+b2 0
        500 http://ftp.debian.org/debian/ testing/main amd64 Packages
        100 /var/lib/dpkg/status
```

t-paul, 2016-01-08:
Any updates on this? Can we close the issue?

closed 2016-02-13 by t-paul