openscad/openscad #1467: Crash on startup on Windows 8

donbright, 2015-10-29:
User spinorkit reports 2015 versions of openscad crash before start on several windows machines, where version 2014.03 works OK.

http://forum.openscad.org/Access-violation-on-starting-OpenSCAD-Win8-tt14201.html#a14204

Crash:
Windows 8.0 Intel HD 4600 graphics PC 
Windows 8.1 HP notebook with AMD Radeon HD8500M graphics
Windows 8.1 HP notebook with  NVIDIA GeForce GT630M graphics.

OK:
Windows 8.1 with a Radeon HD-5570
Windows 7 with a GeForce GT-240

Commandline vs Gui:
"Running openscad.exe --info crashes without output"

donbright, 2015-10-29:
Cannot reproduce bug on Windows 10 virtual machine x86_64 QEMU, OpenSCAD 2015-3

donbright, 2015-10-29:
A .ZIP package build using the alternative build system MSYS2 has been uploaded here. 

http://files.openscad.org/snapshots/OpenSCAD-2015.10.28-x86-64.msys2.zip

SystemsGuy, 2015-10-30:
I can confirm donbright's built version is working on my W7 64b Samsung laptop with IntelHD graphics. Silent failures as noted by spinorkit on all 2015 versions - previously 2015.03-1-x86-64 was working fine, so I'm assuming some external update caused my problems..

donbright, 2015-10-31:
Thanks for testing. It is extremely helpful!

A few notes so far.  Source code between versions is essentially identical. 

The standard build uses MXE cross-compilation system with static linking of all libraries as they come with the MXE system.

MSYS2 uses dynamic linking with a large number of .dll files copied into the distribution package root path. It is also based on Cygwin.

MSYS2 build currently requires a horrible kludge of MSYS2's windeployqt  (see https://github.com/Alexpux/MINGW-packages/issues/221 )

Why does it crash and where?

Hypothesis - it appears to be crashing before it reaches main() due to some kind of fundamental error during program load. Possible evidence - simple commands like '--info' dont work. In my experience such issues can be related to program linking, although there is no guarantee of what is causing this problem.

donbright, 2015-11-02:
A shared library version of openscad.exe under MXE cross compiler is now building and linking (running not tested yet). 

MXE CGAL and OpenCSG makefiles had to be slightly modified.
 
This is a good MXE link for guiding the modifications necessary:

https://github.com/mxe/mxe/wiki/Adding-Shared-Support-to-Packages

Testing underway, pull request to MXE to come soon (TM)

-----

I keep having one thought in the back of my head. What if the only problem is the static link of OpenGL.a under MXE? That is , if i am not mistaken, Mesa's version of OpenGL. If we static link it, but Windows was also trying to use it's own OpenGL.dll (for example from QtOpenGL.dll ) it might crash something...? This would take a few hours to test...and how to test? Can you link statically everything else without OpenGL.a? Is OpenGL.a needed only for the offscreen rendering, and what happens if we just disabled that for a test? But how to test, since I don't have a machine that will reliably crash?

SystemsGuy, 2015-11-02:
Don - your fixed version got me back up and running, but preview <F5> is not producing output.  <F6> render is working as expected.  I do have a Qt development environment installed on this laptop if that makes a difference.

kintel, 2015-11-02:
@SystemsGuy That might be an unrelated issue. Could you try to enable Preferences->Advanced->Force Goldfeather ?

SystemsGuy, 2015-11-02:
kintel - works perfectly - thanks!  I assume that means it's unrelated?

kintel, 2015-11-02:
@SystemsGuy yes, the rendering issue is related to #1409, but triggers on other Windows systems depending on GPU and drivers.

donbright, 2015-11-12:
cross posted to mailing list

Hello

There are some new test builds uploaded (listed below) using a very recent checkout of the MXE MingW cross build system.

It would be extremely helpful if anyone with a Windows machine that crashed on OpenSCAD 2015 would be able to take a few minutes and test both a shared build and a static build, just enough to see if they still crash before bootup, and if not to see whether they will do a simple preview of a cube or whatever.

http://files.openscad.org/snapshots/OpenSCAD-2015.11.11-MingW-x86-32-shared.zip

http://files.openscad.org/snapshots/OpenSCAD-2015.11.11-MingW-x86-32-static.zip

http://files.openscad.org/snapshots/OpenSCAD-2015.11.11-MingW-x86-64-shared.zip

http://files.openscad.org/snapshots/OpenSCAD-2015.11.11-MingW-x86-64-static.zip

Basically the theory being tested is whether or not the crash is caused by Static Linking vs Shared Linking... or whether it is just MXE / MingW in general that is causing the crash.

Thanks

This is a tough one... i have never been able to reproduce the crash (then again i only have access to two different windows machines). These packages were built using mods to the msys2 branch.

donbright, 2015-11-13:
OK user spinorkit and Kent Ahlberg have both tested these builds and say that they are OK. ... both Static and Shared library versions. 

What else could it be?

How different is 2015-03 than the master branch? How could it be different in some way that would cause a crash before main()? Perhaps a static variable initialization somewhere? This seems unlikely at the moment... but will check in the Future....

And was it built with some buggy form of MXE maybe? Was there a glitch in the Cosmic Radiation during upload to the server? Was there a rip in the space time continuum caused by a super intelligent pan-dimensional being?

kintel, 2015-11-13:
@donbright It sounds like a build issue rather than smth. caused by changes to the build system. Is it trivial to backport your build changes to openscad-2015.03-branch? That way we could test that it actually works against the 2015.03 codebase, plus we can release 2015.03-2 with these fixes.
The release is still imminent, but I'd happy to hold it off if this could be improved.

donbright, 2015-11-13:
as far as the MXE static build.... i technically have not changed anything in how it works. I am using debian linux 8 with mxe master branch downloaded and built a week or two ago.

You know this is kind of sounding a little bit like it might be some obscure compiler-linker-library-interaction bug that will never get tracked down or solved because its just too weird? Maybe the system 2015. was built on just had some weird combination of versions that made something not work? Somehow I doubt the Windows ABI has changed ...

I will try building 2015.03 branch as-is on a Debian 8 box with the very current MXE and maybe the users can test that?

The changes in msys2 branch are non-trivial... they enable release-package build on both MSYS2, and MXE shared library (DLL), and other stuff.

donbright, 2015-11-13:
New .zip file off the 2015.03 branch

http://files.openscad.org/snapshots/OpenSCAD-2015.11.13-x86-64.zip

t-paul, 2015-11-13:
Maybe the 2015.03 crashes are the combination of the MXE version I've used back then + possible environment issues on my system?

donbright, 2015-11-14:
User Kit Adams just tested those two 2015.03 branches above and they seem OK. So it's not shared vs static, it's not MSYS2 (cygwin) vs MXE (mingw), its... looking like you are correct.. it's a good possibility its just the system it's built on....

This reminds me of one time back in the old days there was this weird crash or misbehavior that was only triggered if you built OpenSCAD with a specific version of gcc4.8.1 or something like that. (And since OpenSCAD uses a lot of header-only libs like boost, CGAL, etc... it can also be an interaction between gcc and the headers). Oh yeah... it was  issue #514

So i guess the resolution is to just keep up diverse build machine ecosystem and if one doesn't work... try another.....

donbright, 2015-11-14:
woops forgot... should the download linkes on files.openscad.org / openscad.org be updated?

kintel, 2015-11-14:
Closing the issue is only related to the fix being committed, independent of whether the binaries are published or not.

..but if any of the stuff you committed should be backported to 2015.03, that would be nice to track somewhere.

donbright, 2015-11-14:
well... in the end all the commits i made to msys2 branch were for experimentation.... in the end i built the existing 2015.03 branch, unchanged, i just built it on a different machine with a newer MXE. so... there is no "fix" ... there is just a need for a rebuild of the binaries that are currently there on the server. i think?

kintel, 2015-11-14:
@donbright sound good to me - feel free to close this then

donbright, 2015-11-15:
OK.

I uploaded new versions to http://www.openscad.org/downloads.html by setting VERSION=2015.03-1 and running release-common.sh mingw32 and mingw64

```
don@serebryanya:~$ ls -l *exe
-rw-rw-r-- 1 don don 14361985 Nov 14 21:12 OpenSCAD-2015.03-1-x86-32-Installer.exe
-rw-rw-r-- 1 don don 14406862 Nov 14 21:13 OpenSCAD-2015.03-1-x86-64-Installer.exe
don@serebryanya:~$ ls -l *zip
-rw-rw-r-- 1 don don 14375826 Nov 14 21:12 OpenSCAD-2015.03-1-x86-32.zip
-rw-rw-r-- 1 don don 14388270 Nov 14 21:13 OpenSCAD-2015.03-1-x86-64.zip
```

i did a little bit more research. here is some info for future reference. 

I compared the versions of the tools used to build. I looked at the "library info" OS GUI report from both 32 bit and 64 bit crashy 2015.03-1 and did a diff.

32 bit

reported as crashy = <
reported as non-crashy = >

````
< Compiler: GCC "4.9.2" 32bit
> Compiler: GCC "5.2.0" 32bit
< Eigen version: 3.2.2
> Eigen version: 3.2.5
< Qt version: 5.4.1
> Qt version: 5.5.1
< GLib version: 2.42.1
> GLib version: 2.44.1
```

64 bit

````
< Compiler: GCC "4.9.2" 64bit
> Compiler: GCC "5.2.0" 64bit
< Eigen version: 3.2.2
> Eigen version: 3.2.5
< Qt version: 5.4.1
> Qt version: 5.5.1
< GLib version: 2.42.1
> GLib version: 2.44.1

------

Googling around "gcc 4.9 crash startup" reveals a lot of hits. For example. https://github.com/Beep6581/RawTherapee/issues/2741 . There are even some discussion threads about gcc 4.9.x causing failures in the linux kernel and grub and even BIND ... now i dont have scientific proof that gcc 4.9 was our problem here.... but if i had to continue this investigation, that is where i would start looking. (however, since the issue appears to be resolved by simply using newer tools, i will stop investigating for now.)

now here is a fuller library info dump (from qt gui) for future reference

-----

reported as crashy:

32 bit
OpenSCAD Version: 2015.03-1
System information: Windows 7 SP1 (v6.1.1.0) 4 CPUs WOW64 5.75 GB RAM
Compiler: GCC "4.9.2" 32bit
Boost version: 1_57
Eigen version: 3.2.2
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.4.1
QScintilla version: 2.8.4
MingW build: MingW32
GLib version: 2.42.1
Application Path: C:/Users/Valued Customer/Downloads/OpenSCAD-2015.03-1-x86-32 (1)/openscad-2015.03-1
Documents Path: C:\Users\Valued Customer\Documents
Resource Path: C:/Users/Valued Customer/Downloads/OpenSCAD-2015.03-1-x86-32 (1)/openscad-2015.03-1
[... 124 lines cut ...]
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/Valued Customer/Documents/OpenSCAD/libraries
C:/Users/Valued Customer/Downloads/OpenSCAD-2015.03-1-x86-64/openscad-2015.03-1\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 3.3.10179 Compatibility Profile Context
GL Renderer: AMD M880G with ATI Mobility Radeon HD 4250 
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

------

other notes...

great page on QT5 and MINGW potential issues here

https://wiki.qt.io/MinGW-64-bit

closed 2015-11-15 by donbright

kintel, 2015-11-15:
Thx Don!
FYI: The VERSION env variable is no longer needed on releases - the correct thing would be to check out the openscad-2015-1 tag and just rebuild. The 2015.03-branch is currently somewhere in-between releases, so any build from that branch should just be date tagged to avoid any version number confusion.

donbright, 2015-11-15:
thanks for the update, i will fix....