openscad/openscad #1468: Looks like crap on GNOME

GegoXaren, 2015-10-29:
There are a few problems with OpenSCAD on Gnome 3.18 (In order of importance):
**1) Wrong File Selector.**
    On GNOME you should use the GTK+ 3 file selector.
**2) Wrong theme on GNOME.**
  The theme is not Adwaita.
**3) Using wrong fonts.**
  It does not use the same fonts as is set by the system.
**4) Wrong Icons/symbols.**
  No, just no.

kintel, 2015-10-29:
@GegoXaren Some questions:
1) Which version of OpenSCAD are you using?
2) How did you install it?
3) Are you using Qt4 or Qt5 (See Help->Library Info)
4) Do you know of any other Qt applications that integrate properly with Gnome?

GegoXaren, 2015-10-29:
1) OpenSCAD Version: 2015.03-1

2) Install via apt-get in GNOME Ubuntu 15.10 with GNOME 3.18

3)
Qt version: 5.4.2
QScintilla version: 2.9

4)
Only OBS. And that looks like crap too...

t-paul, 2015-10-29:
**1) Wrong File Selector.**
OpenSCAD currently uses `QFileDialog::getOpenFileName()` which is documented to use the native file dialog. That does not seem to work with Qt5, setting `XDG_SESSION_DESKTOP=gnome` does not help on my system (Debian/testing/GNOME 3.18). This might be related to https://bugreports.qt.io/browse/QTBUG-36657) - there's nothing OpenSCAD can do here.

**2) Wrong theme on GNOME.**
OpenSCAD does not set a specific theme. I could not find what the official replacement for the configuration tool is for Qt5, for Qt4 there was qt4-config. Setting the environment variable `QT_STYLE_OVERRIDE=gtk+` works for me.

**3) Using wrong fonts.**
Same issue as theme selection, setting the style override should fix that too

**4) Wrong Icons/symbols.**
Feature request #1108 would make the icon set changeable, feel free to point to other options or contribute an icon set.

t-paul, 2016-01-08:
If there are no further comments, I'm going to close the issue...

closed 2016-02-13 by t-paul