openscad/openscad #1473: Optimization: Pre-calculate 2D CSG in preview mode

kintel, 2015-11-10:
When previewing 2D objects, we render each individual object (using our slow OpenGL renderer).
Since the new 2D CSG pipeline is a lot faster than it used to be, we could probably pre-calculate the CSG and render only the resulting object, which should be noticeably faster.

jordanbrown0, 2026-02-09:
Such a change could also get us "true" 2D objects, instead of presenting 2D objects as one unit thick.

However, the big downside is that the 2D renderer does not preserve color.

gsohler, 2026-02-09:
"one" is actually quite thick for some 2d applications

BTW feel free to cherry-pick  rendered 2d color here:

https://github.com/pythonscad/pythonscad/pull/200

jordanbrown0, 2026-02-09:
Mostly the one-unit-thick thing doesn't matter - it's only presentation, and if you're doing anything with 2D in the 3rd dimension then you're already on thin ice.  But it has some stupid secondary effects, like that it's affected by scaling in Z, and in some cases it can present convexity-based artifacts, Z-fighting, et cetera.  We *can* present true hairline-thickness 2D objects - build a polyhedron with a single face for an example - but they don't play nice with the 3D previewer and so difference and intersection don't work.  We could always fully render 2D before displaying (as we do for extruding), but the 2D renderer loses color.