openscad/openscad #1474: Get dependencies script missing libqt4-opengl-dev

thehans, 2015-11-10:
I'm running through the build process for the first time, on Linux Mint 17.2

Following the instructions here:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Linux/UNIX

I used the uni-get-dependencies.sh script, then qmake and make, and ran into this error:
    src/QGLView.h:4:21: fatal error: QGLWidget: No such file or directory
    #include <QGLWidget>

I found that I needed the package libqt4-opengl-dev to get it to continue.  It would be nice if the script included this one also.

SEUTec, 2015-11-10:
I have compiled OpenSCAD in LinuxMint 17.2 1 week ago and
I  have found the same problem.

Solution:

Go to synaptic and install libqglviewer-dev

Regards.

donbright, 2015-11-12:
thanks for the report, i will download a linux mint 17 iso and fix it.

thehans, 2015-11-12:
It is based on Ubuntu 14.04 LTS for what its worth.

donbright, 2015-11-14:
ok just to be clear.... you are using qt4 on mint17?

thehans, 2015-11-14:
Yes, that's what the script downloaded for me by default.  libqt4-dev.  

And so check-dependencies.sh shows qt 4.8.6

donbright, 2015-11-14:
im trying to rewrite to detect qt4 or qt5 has been selected and use whichever. this will take a while....

donbright, 2015-11-14:
oh gosh sorry... that script is using outdated qt4 on newer mint as of this writing.

donbright, 2015-11-14:
fixed and tested on ubuntu 14, ubuntu 15, mint 17, and debian 8. thanks!

donbright, 2015-11-15:
might have similar issues with other distros as qt4 transitions to qt5 but i will close this for now. (ubuntu 12 is still technically supported so will not rip out all the old code for qt4)

closed 2015-11-15 by donbright