openscad/openscad #1477: std::bad_alloc not catched

Phyks, 2015-11-11:
Hi,

When doing

```
$fn=300000
cylinder(h=1, r=1);
```

(which is kind of stupid, but nevermind)

I got a `std::bad_alloc` which is not catched.

donbright, 2015-11-15:
Using CSG peview, i am unable to reproduce this bug on my machine (6 GB RAM). It takes a long time, but it produces a very smooth looking cylinder. memory usage for OpenSCAD in 'top' on linux doesn't go above about 1.2 G ram total. 

Using STL render.... also unable to reproduce this bug.  Virt memory is 270 M and RES is 85 M steady for about 10 minutes or so, after which i get an stl file that is 193 Megabytes, and produces one of the more unusual images in Meshlab i have ever seen

[image]

     $fn=300000;
      cylinder(h=1, r=1);

that being said, i have no doubt there are "new" and/or malloc operators inside openscad being  called without exceptions being caught.

Phyks, 2015-11-16:
I do not have access to the machine on which I found this bug at the moment, and cannot reproduce it on my own machine.

The machine I was using was not a high end one at all, and I think it must have ran out of RAM, then allocation failed and it raised an `std::bad_alloc`.

After having a quick look at openscad code, I found out that no `std::bad_alloc` are ever directly catched and that no `new`/`delete` calls are wrapped inside `try`/`catch` blocks.

My quick and dirty fix for this would be to just add a global `try`/`catch` around the main code, to handle such exceptions. And, as a dirty fix, I propose to just try to save the current file to a specific filename (say e.g. `filename.backup.scad`) so that one can recover it easily. It is very stressful to have to think about saving regularly, especially if a render can break everything and make us lose one hour or more of work.

Another alternative / complementary option would be to automatically save the current file before doing a render.

Both solutions are problematic in the case the file was never saved before, and I think the best option in such cases would just be to save it to `/tmp`.

What do you think?

kintel, 2015-11-16:
Note: We already save the current file before rendering (since 2014.03). Help->Library Info should display the backup path.

Phyks, 2015-11-16:
@kintel Oh, I did not know about it. Thanks!

closed 2016-03-30 by Phyks