openscad/openscad #1483: Edges preview and render jaggedly

thehans, 2015-11-13:
Edges are rendered jaggedly with triangles of the wrong shaded color showing through presumably from the faces behind them?   

This is more apparent with a view that is somewhat zoomed in.

runing this script in the gui, in default view
`cube(0.1,center=true);`

produces these images:
preview
[image]

render
[image]

kintel, 2015-11-13:
This is z buffer tearing and is caused by z-buffer floating point inaccuracy.
We could fix this by modifying our front/back clipping planes in real-time based on the projected bounding box of the design. I think we currently don't, so very small objects are likely to tear like this.
As usual, such issues tend to be put on the back burner until we rewrite the rendering code: https://github.com/openscad/openscad/wiki/Project%3A-Improve-OpenGL-rendering

thehans, 2015-11-13:
I noticed this doesn't happen on my Windows machine, is it just down to difference in graphics card hardware or something else?

I could render a cube all the way down to cube(0.00000000000000000001, center=true);
and the edges looked perfect.

thehans, 2015-11-13:
Also, not sure if this is helpful, but I tried smaller and smaller cubes on my linux machine.  Around size 0.002 or smaller, the edges all render perfectly smoothly, its just that the entire object is inside out now.
`cube(0.001,center=true);`

[image]

thehans, 2015-11-13:
Also, whatever is used to render when -o flag is called, does not exhibit this issue.

thehans, 2015-11-13:
OK I realize now that I had been viewing in Ortho mode.  There is no difference between the windows and linux versions as I previsouly though, my  linux openscad just happened to have ortho turned on.

kintel, 2015-11-13:
@thehans oh, good catch :)

kintel, 2015-11-13:
Fixed by #1484

closed 2015-11-13 by kintel