openscad/openscad #1490: -lqtpcre link failure on OSX Mavericks (Darwin 13.4) during dependency build

donbright, 2015-11-16:
hello

i am running plain "setenv-macosx-qt5.sh" and "scripts/macosx-build-dependencies" but i get a link failure during qt5 build...

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -stdlib=libc++ -mmacosx-version-min=10.7 -single_module -dynamiclib -compatibility_version	5.5 -current_version	5.5.1 -install_name	@rpath/QtCore.framework/Versions/5/QtCore -o QtCore .obj/qlocale_mac.o .obj/qtimezoneprivate_mac.o .obj/qstring_mac.o .obj/qbytearray_mac.o .obj/qdatetime_mac.o .obj/qurl_mac.o .obj/qfilesystemwatcher_fsevents.o .obj/qstandardpaths_mac.o .obj/qcore_mac_objc.o .obj/qstring_compat.o .obj/qabstractanimation.o .obj/qvariantanimation.o .obj/qpropertyanimation.o .obj/qanimationgroup.o .obj/qsequentialanimationgroup.o .obj/qparallelanimationgroup.o .obj/qpauseanimation.o .obj/qatomic_unix.o .obj/qglobal.o .obj/qglobalstatic.o .obj/qlibraryinfo.o .obj/qmalloc.o .obj/qnumeric.o .obj/qlogging.o .obj/qhooks.o .obj/qatomic.o .obj/qexception.o .obj/qresultstore.o .obj/qfutureinterface.o .obj/qfuturewatcher.o .obj/qmutex.o .obj/qreadwritelock.o .obj/qrunnable.o .obj/qmutexpool.o .obj/qsemaphore.o .obj/qthread.o .obj/qthreadpool.o .obj/qthreadstorage.o .obj/qthread_unix.o .obj/qwaitcondition_unix.o .obj/qarraydata.o .obj/qbitarray.o .obj/qbytearray.o .obj/qbytearraylist.o .obj/qbytearraymatcher.o .obj/qcollator.o .obj/qcommandlineoption.o .obj/qcommandlineparser.o .obj/qcryptographichash.o .obj/qdatetime.o .obj/qdatetimeparser.o .obj/qeasingcurve.o .obj/qelapsedtimer.o .obj/qfreelist.o .obj/qhash.o .obj/qline.o .obj/qlinkedlist.o .obj/qlist.o .obj/qlocale.o .obj/qlocale_tools.o .obj/qpoint.o .obj/qmap.o .obj/qmargins.o .obj/qmessageauthenticationcode.o .obj/qcontiguouscache.o .obj/qrect.o .obj/qregexp.o .obj/qrefcount.o .obj/qshareddata.o .obj/qsharedpointer.o .obj/qsimd.o .obj/qsize.o .obj/qstring.o .obj/qstringbuilder.o .obj/qstringlist.o .obj/qtextboundaryfinder.o .obj/qtimeline.o .obj/qtimezone.o .obj/qtimezoneprivate.o .obj/qunicodetools.o .obj/qvector.o .obj/qvsnprintf.o .obj/qversionnumber.o .obj/qelapsedtimer_mac.o .obj/qcollator_macx.o .obj/qregularexpression.o .obj/harfbuzz-buffer.o .obj/harfbuzz-gdef.o .obj/harfbuzz-gsub.o .obj/harfbuzz-gpos.o .obj/harfbuzz-impl.o .obj/harfbuzz-open.o .obj/harfbuzz-stream.o .obj/harfbuzz-shaper-all.o .obj/qharfbuzz.o .obj/qabstractfileengine.o .obj/qbuffer.o .obj/qdatastream.o .obj/qdataurl.o .obj/qtldurl.o .obj/qdebug.o .obj/qdir.o .obj/qdiriterator.o .obj/qfile.o .obj/qfiledevice.o .obj/qfileinfo.o .obj/qipaddress.o .obj/qiodevice.o .obj/qlockfile.o .obj/qnoncontiguousbytedevice.o .obj/qprocess.o .obj/qstorageinfo.o .obj/qtextstream.o .obj/qtemporarydir.o .obj/qtemporaryfile.o .obj/qresource.o .obj/qresource_iterator.o .obj/qsavefile.o .obj/qstandardpaths.o .obj/qurl.o .obj/qurlidna.o .obj/qurlquery.o .obj/qurlrecode.o .obj/qsettings.o .obj/qfsfileengine.o .obj/qfsfileengine_iterator.o .obj/qfilesystemwatcher.o .obj/qfilesystemwatcher_polling.o .obj/qfilesystementry.o .obj/qfilesystemengine.o .obj/qfileselector.o .obj/qloggingcategory.o .obj/qloggingregistry.o .obj/qfsfileengine_unix.o .obj/qfilesystemengine_unix.o .obj/qlockfile_unix.o .obj/qprocess_unix.o .obj/qfilesystemiterator_unix.o .obj/forkfd_qt.o .obj/qsettings_mac.o .obj/qstorageinfo_mac.o .obj/qfilesystemwatcher_kqueue.o .obj/qabstractitemmodel.o .obj/qabstractproxymodel.o .obj/qitemselectionmodel.o .obj/qidentityproxymodel.o .obj/qsortfilterproxymodel.o .obj/qstringlistmodel.o .obj/qjson.o .obj/qjsondocument.o .obj/qjsonobject.o .obj/qjsonarray.o .obj/qjsonvalue.o .obj/qjsonwriter.o .obj/qjsonparser.o .obj/qpluginloader.o .obj/qfactoryloader.o .obj/quuid.o .obj/qlibrary.o .obj/qelfparser_p.o .obj/qmachparser.o .obj/qlibrary_unix.o .obj/qabstracteventdispatcher.o .obj/qabstractnativeeventfilter.o .obj/qbasictimer.o .obj/qeventloop.o .obj/qcoreapplication.o .obj/qcoreevent.o .obj/qmetaobject.o .obj/qmetatype.o .obj/qmetaobjectbuilder.o .obj/qmimedata.o .obj/qobject.o .obj/qobjectcleanuphandler.o .obj/qsignalmapper.o .obj/qsocketnotifier.o .obj/qtimer.o .obj/qtranslator.o .obj/qvariant.o .obj/qcoreglobaldata.o .obj/qsharedmemory.o .obj/qsystemsemaphore.o .obj/qpointer.o .obj/qmath.o .obj/qsystemerror.o .obj/qcoreapplication_mac.o .obj/qcore_mac.o .obj/qcore_unix.o .obj/qcrashhandler.o .obj/qeventdispatcher_unix.o .obj/qtimerinfo_unix.o .obj/qsharedmemory_posix.o .obj/qsharedmemory_systemv.o .obj/qsharedmemory_unix.o .obj/qsystemsemaphore_posix.o .obj/qsystemsemaphore_systemv.o .obj/qsystemsemaphore_unix.o .obj/qisciicodec.o .obj/qlatincodec.o .obj/qsimplecodec.o .obj/qtextcodec.o .obj/qtsciicodec.o .obj/qutfcodec.o .obj/qgb18030codec.o .obj/qjpunicode.o .obj/qeucjpcodec.o .obj/qjiscodec.o .obj/qsjiscodec.o .obj/qeuckrcodec.o .obj/qbig5codec.o .obj/qstatemachine.o .obj/qabstractstate.o .obj/qstate.o .obj/qfinalstate.o .obj/qhistorystate.o .obj/qabstracttransition.o .obj/qsignaltransition.o .obj/qeventtransition.o .obj/qmimedatabase.o .obj/qmimetype.o .obj/qmimemagicrulematcher.o .obj/qmimetypeparser.o .obj/qmimemagicrule.o .obj/qmimeglobpattern.o .obj/qmimeprovider.o .obj/qxmlstream.o .obj/qxmlutils.o .obj/qrc_mimetypes.o .obj/moc_qabstractanimation_p.o .obj/moc_qnamespace.o .obj/moc_qthread.o .obj/moc_qthreadpool.o .obj/moc_qfuturewatcher.o .obj/moc_qeasingcurve.o .obj/moc_qlocale.o .obj/moc_qtimeline.o .obj/moc_qfile.o .obj/moc_qfiledevice.o .obj/moc_qiodevice.o .obj/moc_qnoncontiguousbytedevice_p.o .obj/moc_qtextstream_p.o .obj/moc_qtemporaryfile.o .obj/moc_qsavefile.o .obj/moc_qsettings.o .obj/moc_qfilesystemwatcher_p.o .obj/moc_qfilesystemwatcher_polling_p.o .obj/moc_qfilesystemwatcher_fsevents_p.o .obj/moc_qfilesystemwatcher_kqueue_p.o .obj/moc_qabstractitemmodel.o .obj/moc_qstringlistmodel.o .obj/moc_qpluginloader.o .obj/moc_qlibrary.o .obj/moc_qfactoryloader_p.o .obj/moc_qabstracteventdispatcher.o .obj/moc_qeventloop.o .obj/moc_qcoreapplication.o .obj/moc_qcoreevent.o .obj/moc_qmimedata.o .obj/moc_qsocketnotifier.o .obj/moc_qtimer.o .obj/moc_qtranslator.o .obj/moc_qobjectcleanuphandler.o .obj/moc_qsharedmemory.o .obj/moc_qeventdispatcher_unix_p.o .obj/moc_qabstractstate.o .obj/moc_qstate.o .obj/moc_qfinalstate.o .obj/moc_qhistorystate.o .obj/moc_qabstracttransition.o .obj/moc_qsignaltransition.o .obj/moc_qeventtransition.o  -framework DiskArbitration -framework IOKit -lz -L/Volumes/Macintosh\ HD/Users/user900743/libraries/src/qt-everywhere-opensource-src-5.5.1/qtbase/lib -lqtpcre -lm -framework Foundation -framework CoreServices -framework ApplicationServices -framework CoreFoundation 
ld: library not found for -lqtpcre
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [../../lib/QtCore.framework/QtCore] Error 1
make[2]: *** [sub-corelib-install_subtargets] Error 2
make[1]: *** [sub-src-install_subtargets] Error 2
make: *** [module-qtbase-install_subtargets] Error 2
user900743@darwin:~/openscad$ uname -a
Darwin 74-80-237-108.ic2net.net 13.4.0 Darwin Kernel Version 13.4.0: Wed Mar 18 16:20:14 PDT 2015; root:xnu-2422.115.14~1/RELEASE_X86_64 x86_64
user900743@darwin:~/openscad$ echo $OSTYPE
darwin13
user900743@darwin:~/openscad$ clang --version
Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin13.4.0
Thread model: posix

```

kintel, 2015-11-16:
I think this is a bug in Qt's build system when using parallel make. Try this:
```
cd ../libraries/src/qt-everywhere-opensource-src-5.5.1
make install
```

donbright, 2015-11-17:
thanks!!!! now it has decided that 

    /Users 

is actually 

    /Volume/Macintosh\\ HD/Users

which is confusing make and killing the build

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-dead_strip -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -stdlib=libc++ -mmacosx-version-min=10.7 -Wl,-rpath,@loader_path/../lib -o ../../../bin/qlalr .obj/compress.o .obj/cppgenerator.o .obj/dotgraph.o .obj/lalr.o .obj/main.o .obj/parsetable.o .obj/recognizer.o .obj/grammar.o   -F/Volumes/Macintosh\\ HD/Users/user900743/libraries/src/qt-everywhere-opensource-src-5.5.1/qtbase/lib -framework QtCore 
clang: error: no such file or directory: 'HD/Users/user900743/libraries/src/qt-everywhere-opensource-src-5.5.1/qtbase/lib'
```

kintel, 2015-11-17:
@donbright How did you get a '/Volumes/Macintosh HD' in the first place? Are you mounting multiple hard drives with software spread across them?

donbright, 2015-11-17:
im on a MacInCloud shell account. Everything looks like "/Users/userblah/blah" except when you run qmake it spits all these prepends of "Volumes/Macintosh HD" to Makefiles. 

someone with cmake is seeing prepending like this as well

https://cmake.org/Bug/view.php?id=15488#bugnotes

feels like it might be some version of "return canonical path" deciding to prepend volume names.

donbright, 2015-11-17:
considering closing this as it is definitely upstream bug, and apparently has no easy fix?

donbright, 2015-11-17:
ok this is .... ... really difficult without root access. going to go crawl into a fetal position.

closed 2015-11-17 by donbright

clothbot, 2015-11-21:
I've been running into this as well.  Seeing if I can get farther along...

donbright, 2015-11-21:
during "configure" of qt it generates a "qmake" binary
which calls

   ./qtbase/qmake/generators/makefile.cpp
    MakefileGenerator::fileFixify

which in turn calls
    http://doc.qt.io/qt-4.8/qfileinfo.html#canonicalFilePath

This gives the "/Volumes/Macintosh HD" with no escape for space

At some point, somewhere, "Macintosh HD" changes to  "Macintosh\\ HD"

it will work if the escape is "Macintosh\ HD" instead of "Macintosh\\ HD"  but not sure where to change this

clothbot, 2015-11-21:
Of course, in my case my directory names are /Volumes/MasterHD and /Volumes/CaseSensitive...

I wonder if that's actually my problem, that I'm doing the building in my case-sensitive path.  Wouldn't be the first time.

kintel, 2015-11-21:
On a "normal" OS X installation, "/Volumes/Macintosh HD" is just a symlink to "/".
I still don't understand how Qt decides to pick up "/Volumes/Macintosh HD" at all, since that wouldn't be visible unless you're specifically looking for it.
The MacInCloud thing might of course have a different way of mounting volumes, which triggers this behavior.

donbright, 2015-11-21:
its setup with /Volumes/OSX linked to /

```
user900743@darwin:~/libraries/src/qt-everywhere-opensource-src-5.5.1$ ls -l /Volumes/
total 8
drwxr-xr-x  31 root  wheel  1122 Apr 12  2015 Macintosh HD
lrwxr-xr-x   1 root  admin     1 Nov 21 02:31 OSX -> /
user900743@darwin:~/libraries/src/qt-everywhere-opensource-src-5.5.1$ df
Filesystem    512-blocks      Used  Available Capacity  iused     ifree %iused  Mounted on
/dev/disk0s2   486717952 422102848   64103104    87% 52826854   8012888   87%   /
devfs                371       371          0   100%      642         0  100%   /dev
/dev/disk1s2  1951845952 384553416 1567292536    20% 48069175 195911567   20%   /Volumes/Macintosh HD
map -hosts             0         0          0   100%        0         0  100%   /net
map auto_home          0         0          0   100%        0         0  100%   /home

```

clothbot, 2015-11-21:
Cautiously optimistic. Running manual configure ... make ... make install with an extra "-skip qtwebchannel" configure option ran to completion where it error'd out previously.

Trying with the extra option inserted into the build script now...

clothbot, 2015-11-22:
With the extra option I was able to run "./scripts/macosx-build-dependencies.sh qt5" to completion.

Now to try the full build-deps from scratch, after removing ../libraries.

clothbot, 2015-11-22:
That seems to have done the trick!  Successful build of OpenSCAD.app!

I'm going to back out all the other changes I made and confirm that only adding "-skip qtwebchannel" is sufficient.

kintel, 2015-11-22:
@donbright ok, so I assume that /Users is symlinked into /Volumes/Macintosh HD ?
..which is probably what confuses Qt. My theory would be that Qt hasn't been tested on such configurations and that its a weakness in their build system.

clothbot, 2015-11-22:
@donbright give it a try now.  Splitting the single "make install" into 2-stage "make; make install" addressed the first -lqtpcre linking issue for me, then adding the extra configure "-skip qtwebchannel" option resolved the second QJSValue not found error.

donbright, 2015-11-22:
THANKS GUYS

yup it was linked as you say, 

      user900743@darwin:~$ ls -l /Users
lrwxr-xr-x  1 root  wheel  28 Mar  9  2015 /Users -> /Volumes/Macintosh HD/Users/

i combined clothbot instructions and a awful nasty hack to 

     qtbase/qmake/generators/unix/unixmake.cpp  ::escapeFilePath()
      if (ret.startsWith(QString("/Volumes/Macintosh\\ HD"))) {
      ret.replace(QString("/Volumes/Macintosh\\ HD"),QString("")); }

to manually remove any ```"/Volumes/Macintosh\ HD"``` prepend. not exactly something i could pull request to QT (there are dozens of places where they modify filepaths, couldnt figure out the key appropriate point) but it worked for this system.

kintel, 2015-11-22:
@donbright One idea: Could it be that they simply don't support spaces in path names?

donbright, 2015-11-22:
freetype
----
 has a failure "install-sh no such file or directory". work around is to edit builds/unix/unix-def.mk and replace install-sh -c d with quotes, "./builds/unix/install-sh -c d"   MacPorts had the same issue and they are working on reporting to upstream (here is link:)
 
     https://svn.macports.org/repository/macports/trunk/dports/print/freetype/Portfile

ragel
---------
c++11 problems and clang

```
user900743@darwin:~/libraries/src/ragel-6.9$ make VERBOSE=1
Making all in ragel
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-am
g++ -DHAVE_CONFIG_H -I. -I../aapl   -Wall -std=c++11 -stdlib=libc++ -MT ragel-common.o -MD -MP -MF .deps/ragel-common.Tpo -c -o ragel-common.o `test -f 'common.cpp' || echo './'`common.cpp
common.cpp:37:70: error: constant expression evaluates to 18446744073709551615
      which cannot be narrowed to type 'long long' [-Wc++11-narrowing]
  ..."long",  "ulong",   false,  true,  false,  0,         ULONG_MAX,  sizeof...
                                                           ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/6.0/include/limits.h:74:19: note: 
      expanded from macro 'ULONG_MAX'
#define ULONG_MAX (__LONG_MAX__ *2UL+1UL)
                  ^~~~~~~~~~~~~~~~~~~~~~~
common.cpp:37:70: note: override this message by inserting an explicit cast
  ...false,  true,  false,  0,         ULONG_MAX,  sizeof(unsigned long) }
                                       ^~~~~~~~~
                                       static_cast<long long>( )

```

even if you fix this there are other errors related to c++11

the workaround is to compile ragel not in c+11 mode by removing -std=c++11 from the CXXFLAGS while configuring/building. 

-----
also qmake kludge above doesn't completely work, still had to edit openscad makefile to remove "/Volumes/Macintosh HD" because the space was unescaped.

i have llimited time on this machine so cant really do patches. 

in the positive news... i have now built openscad on macosx for my first time :) fun feeling. thanks.

now i just have to google "What is a .app" :)

------

interesting --info dump, OpenSCAD is also picking up the "/Volumes/Macintosh HD/" thing  and it cannot do PNG dumps from commandline over SSH, cant get a offscreen GL context.

```
 ./OpenSCAD.app/Contents/MacOS/OpenSCAD --info
Could not initialize localization.
OpenSCAD Version: 2015.11.22
System information: Mac OS X Version 10.9.5 (Build 13F1096) Macmini6,1 2 CPUs 10.00 GB RAM 
Compiler: Clang "6.0 (clang-600.0.57)" 64bit
Boost version: 1_59
Eigen version: 3.2.6
CGAL version, kernels: 4.6.3, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9
MingW build: No
GLib version: 2.46.1
Application Path: /Volumes/Macintosh HD/Users/user900743/openscad/bin/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/user900743/Documents
Resource Path: /Volumes/Macintosh HD/Users/user900743/openscad/bin/OpenSCAD.app/Contents/MacOS
User Library Path: /Volumes/Macintosh HD/Users/user900743/Documents/OpenSCAD/libraries
User Config Path: /Users/user900743/Library/Application Support/OpenSCAD
Backup Path: /Volumes/Macintosh HD/Users/user900743/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Volumes/Macintosh HD/Users/user900743/Documents/OpenSCAD/libraries
  /Volumes/Macintosh HD/Users/user900743/openscad/bin

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

2015-11-22 08:43:46.282 OpenSCAD[55323:507] invalid CoreGraphics connection
2015-11-22 08:43:46.283 OpenSCAD[55323:507] invalid pixel format
2015-11-22 08:43:46.284 OpenSCAD[55323:507] invalid context
Unable to create NSOpenGLContext
Can't create OpenGL OffscreenView. Code: -1. Exiting.

```

donbright, 2015-11-22:
@kintel they do support on Windows b/c they have to, spaces are all over the place ("C:\program files, user dirs, etc etc) ... on unix (their mac port has a lot of unix calls in it apparently) it looks like they do something with spaces in pathnames because they are using both double-backslash and single-backslash inconsistently to escape them. it might even be that clang just doesnt know what double-backslash means, but maybe that is the shell? giving up for now as i have a sort-of working build.

kintel, 2015-11-22:
Some notes:
* We haven't fully moved to C++11 on Mac yet (2015.03-* is still on C++03), so I'm not on top of the build issues.
* I don't have access to a headless Mac, so I haven't tested offscreen rendering. Perhaps I should give macincloud a spin..
* Travis CI offers Mac builds, but it looks like they've maxed out capacity right now.