openscad/openscad #1491: notes from the debian/copyright file

chrysn, 2015-11-16:
as the flattr icon went away in 2015.03-2, i reviewed the debian copyright file on openscad, and noted that there are minor points that, if resolved, would allow debian to use (and thus mirror) the original shipped tar ball:

* the rainbow.min.js file is shipped without source and any hint on origin or copyright. if the file needs to be shipped (i don't see exactly what examples-html does), please consider shipping its source file (which presumably contains a copyright/license header in its comments) next to it.
* SparkleAutoUpdater: "BSD license" can mean different things. could you clarify what it means here (preferably not "original 4-clause BSD", as that is GPL incompatible), and add that information to the package?

t-paul, 2015-11-16:
Yeah, I saw that *.min.js issues on OBS too. Shipping the non-minified java script files in addition seems a bit strange, is that the default solution in other packages too? We don't ship most other library source code. Is there some packaging program to do the minification in the build process maybe?

The examples-html stuff generates some static web pages from the example scad script. That's a bit work in progress, but hopefully might integrate with the main website later. See http://files.openscad.org/examples/Basics/CSG-modules.html

The Sparkle license should be https://github.com/sparkle-project/Sparkle/blob/master/LICENSE. Is that the 2-clause BSD?

kintel, 2015-11-16:
@t-paul I guess we shouldn't ship any javascript sources, but rather pull them using a package manager like bower or npm. ..or at least have those package managers configured, so that we can reproduce the shipped files from their original location.

chrysn, 2015-11-16:
On Mon, Nov 16, 2015 at 02:08:12AM -0800, Torsten Paul wrote:
> Shipping the non-minified java script files in addition seems a bit
> strange, is that the default solution in other packages too? We don't
> ship most other library source code. Is there some packaging program
> to do the minification in the build process maybe?

the debian stance on this is "don't ship dependencies", but that really
works well only with a stable ecosystem, which not all of javascript is
yet. there are minifiers that can be integrated into build processes,
but there have been situations where even minification depends on a
particular minifier. (it's a bit of a hassle inside debian, simple
packages start depending on huge and new/changing build systems).

the practical suggestion i'd give if the .min.js needs to be in the
distribution is to provide the original javascript next to it.
(depending on the license, that might be even a requirement.)

a bit like "here's the .c source, but you'll need the newest and
fanciest compiler to build it, so here's the elf file for your
convenience". i still wouldn't check the elf into my repositories, but
projects' needs differ.

> The Sparkle license should be
> https://github.com/sparkle-project/Sparkle/blob/master/LICENSE. Is
> that the 2-clause BSD?

it looks like MIT/X11 license (aka BSD two-clause).

back when i first tried to drill down the license, it looked like not
directly from Sparkle, but copied from some website that said "BSD
licensed" (i think it was [1]). if you have enough context to infer that
the abovementioned license applies, please consider clarifying that in
SparkleAutoUpdater.{h,mm}.

[1] https://el-tramo.be/blog/mixing-cocoa-and-qt/

kintel, 2015-11-16:
@chrysn Would this work for debian?
```
/*
 * Copyright (C) 2008 Remko Troncon.
 *   See https://github.com/remko/mixing-cocoa-and-qt
 *       https://el-tramo.be/blog/mixing-cocoa-and-qt
 *   License: BSD. Assuming Sparke's variant of BSD based on context:
 *       https://github.com/sparkle-project/Sparkle/blob/master/LICENSE
 * Copyright (C) 2013 Marius Kintel.
 *   License: 2-clause BSD License (http://opensource.org/licenses/BSD-2-Clause)
 */
```

chrysn, 2015-11-16:
i think it would. (things are easier if you pick
http://opensource.org/licenses/MIT, because that matches precisely their
license.)

t-paul, 2015-11-30:
@chrysn are that all open issues related to packaging?

Note: I came across some issues with cgal+cmake (as we do for the test suite). The cgal package in testing does need libcgal-qt5-dev (CGAL/cgal#471) even if that's not actually used.

chrysn, 2015-11-30:
On Mon, Nov 30, 2015 at 01:50:49AM -0800, Torsten Paul wrote:
> Note: I came across some issues with cgal+cmake (as we do for the test
> suite). The cgal package in testing does need libcgal-qt5-dev
> (CGAL/cgal#471) even if that's not actually used.

that does affect the openscad debian package, but afaict its effect is
limited to the debian package having an essentially needless dependency
that will result in no more harm than another library package installed;
once they've worked out the build system fixes there, openscad will
automatically drop the dependency in debian.

> @chrysn are that all open issues related to packaging?

most of the patches i have around are debian-specific (eg. [1] to print
the installed debian package's version in test_pretty_print instead of
trying to read git versions which is bound to fail, or [2] because mcad
is built from a separate source package) or already forwarded.

the only hackery that remains is the path twiddling for ctest ([3], [4],
[5] and [6]). all of that is done in order to have the ctest runnable
out-of-tree at user-defined locations, typically by the openscad-testrun
script [7] (which, it seems, i haven't pushed again in quite a while
since the "no-testprograms branch" thread, so thanks for asking).

i'm not sure of how to apply that cleanly to upstream, as it currenlty
depends on the perl lines in [3] bending the absolute paths to relative
ones in the *.cmake files. if there is a way to have cmake produce
relative paths, chances are the rest of it could be applied, and we
could extend openscad-testrun to be generally shippable with openscad as
a tool for running the graphics-card dependent part of the test suite on
the very hardware openscad is used (and not necessarily built) on.

[1] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/patches/Report-Debian-versions-in-test_pretty_print.patch?h=debian
[2] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/patches/patch-out-test-for-mcad.patch?h=debian
[3] the perl lines in http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/rules?h=debian
[4] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/patches/don-t-override-OPENSCADPATH.patch?h=debian
[5] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/patches/test_pretty_print-drop-file-name-munging.patch?h=debian
[6] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/patches/test_pretty_print-remove-a-pathname-normalization-st.patch?h=debian
[7] http://anonscm.debian.org/cgit/collab-maint/openscad.git/tree/debian/openscad-testrun?h=debian

kintel, 2019-01-16:
I believe this was all addressed in Nov 2015 : /

closed 2019-01-16 by kintel