openscad/openscad #1492: Test suite ignores alpha, threshold is wrong, etc.

thehans, 2015-11-16:
This issue if to discuss possible changes to the tests to include the alpha channel as part of the checks.  
It is related to Pull request #1485 which would add support for transparent backgrounds in PNG exports.  Nophead's comment [here](https://github.com/openscad/openscad/pull/1485#issuecomment-156729928) brought up an issue, where including the alpha channel in PNG causes semi-transparent objects to leave semi-transparent holes in the PNG, even when the background is supposed to be fully opaque.  I am working on a solution to the PNG export with alpha, but what I'm writing about right now is more about the tests. The problem is that the tests do not detect any of these alpha channel inconsistencies.

First, there is the default imagemagick comparison 
Line 154 in tests/test_cmdline_tool.py shows the arguments passed to imagemagick
```args = [expectedfilename, resultfilename, "-alpha", "Off", "-compose", "difference", "-composite", "-threshold", "10%", "-morphology", "Erode", "Square", "-format", "%[fx:w*h*mean]", "info:"]```

OK, alpha is off.  Setting alpha to `"On"` makes the tests start detecting some of the problematic images.  But I noticed that strangely still not all of them are even noticed.  

For example the throwntogether/background-modifier test image passes on my computer
Here is the expected image:
[image]
Here is the generated one, with alpha export enabled, (maybe hard to see with no checkerboard on github, but the semi-transparent objects here are more white since the webpage background is showing through.)  This should not pass tests but it does.
[image]
So looking at how the imagemagick comparison works, it seems it performs a difference operation on the pixels, and then threshold on the difference, then erodes away small edge aliasing differences, then sums up the count of different pixels.
Breaking down the process by running partial command manually, I got this for the difference which looks roughly how I would expect:
[image]
Now applying the threshold of 10% gives this result, which is completely wrong.  The differences are just thin lines that will disappear during the erode step and allow it to pass.
[image]
I think it should look more like this (I had to use the options `-black-threshold 10% -white-threshold 10%` instead of `-threshold 10%` to get this)
[image]

This could be a fairly big issue if the test server is affected by the same threshold bug as mine.  It 
lets all kinds of false things pass, and doesn't seem to be even tied to turning on alpha.

For reference I have ImageMagick Version: 8:6.7.7.10-6ubuntu3
on Linux Mint 17.2(based on ubuntu 14.04 LTS)

So using this line on test_cmdline_tool.py:154 makes the default imagemagick comparison function correctly as far as I can tell.
```args = [expectedfilename, resultfilename, "-alpha", "On", "-compose", "difference", "-composite", "-black-threshold", "10%", "-white-threshold", "10%", "-morphology", "Erode", "Square", "-format", "%[fx:w*h*mean]", "info:"]```
I still have no idea why the -threshold 10% produces such blatantly wrong output.

So the default testing config is covered above, but there are three other types of possible comparisons used by the tests
* "ncc" comparison, which produces even crazier output with alpha enabled which I cannot even comprehend.  Comparing an image with itself wouldn't even pass, I am at a loss if it is even possible to support alpha channel here
* the "old" imagemagick setting, I'm not sure what to use to test this so I didn't even try.  Don't know if older versions have the same -threshold issues also or if there is anything else needed to make an "old" version work.
* Finally the "diffpng" comparison program, which appears to also ignore alpha since I ran that and all the images with incorrect transparencies passed.  I looked at the code a bit and it seemed fairly complex, not sure how feasible it would be for it to check alpha differences also.

thehans, 2015-11-16:
For testing the "old" version, I just realized that's what Travis CI uses.  Turning alpha on works to detect image with erroneous alpha.  It did not require altering the `-threshold` option.

donbright, 2015-11-16:
if you can get both "old" and "default" to work that is probably good enough for most linux systems. (can't speak for OSX...).  

"ncc" and "diffpng" are for more unusual systems that are not encountered so often. the reason for the different comparisons (ncc, diffpng) is because imagemagick is extremely unstable across platforms and versions and there was a need to run the tests on lots of different systems. for example there was a system where imagemagick could not tell the difference between green and yellow, which is why there is now a verification during the "cmake" run of imagemagick's comparison (magicktest1.png vs magicktest2.png).  If you want to fall down a rabbit hole, diffpng can be tuned inside of the setup in the "CompareArgs" section of diffpng.cpp (although i am unsure if it will be able to do alpha comparison). but like i said diffpng is for rare systems where imagemagick is completely broken.

thehans, 2017-12-09:
With the merge of #2210, there is some basic transparency checking now active.  I ended up turning `-alpha` `On`, but did not mess with the other `-threshold` changes I mentioned above.

I haven't had a chance to go through manually checking every step of the process in my original report, to verify that particular test-case, but I'm not too concerned for a couple reasons:
1) I opened this a couple years ago, and travis has changed an ubuntu version in that time, presumably with an updated imagemagick package that I suspect may have addressed some of the problems I was running into here.
2) If I remember correctly, the faulty images in the above test case were due to some very messed up alpha blending I had from an test build (from working on #1485 ), that I don't remember how to reproduce,  and I feel would be too much effort to try to retrace those steps.

I think I'm fine with closing for now, and can re-open if I encounter any more issues during further work towards #1485

closed 2017-12-09 by kintel