openscad/openscad #1493: Entire text line blinks in solid black on each character typed (Mac OSX)

zapta, 2015-11-17:
Using Mac OSX 10.10.5 and upgraded recently to OpenScad version 2015.03-2  (got the upgrade notification so I assume I had the latest version previously). 

Whenever typing text in the editing window the entire line blink with solid black. That is, on each character that I type, the entire line (including the background) blinks in solid black. 

This is non usable. I will try to downgrade to the previous version.

zapta, 2015-11-17:
I downgraded to 2015.03-1 and the problem disappeared.  Seems to be a regression in 2015.03-2

kintel, 2015-11-17:
I can reproduce on Mac - haven't tried other platforms.

donbright, 2015-11-18:
cannot reproduce on linux amd64 

OpenSCAD Version: 2015.03-2
System information: Linux 3.13.0-68-generic #111-Ubuntu SMP Fri Nov 6 18:17:06 UTC 2015 x86_64 Ubuntu 14.04.3 LTS 4 CPUs 5.58 GB RAM
Compiler: GCC "4.8.4" 64bit
Boost version: 1_54
Eigen version: 3.2.0
CGAL version, kernels: 4.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 5.2.1
QScintilla version: 2.8.1
MingW build: No
GLib version: 2.40.2
Application Path: /tmp/openscad-2015.03-2
Documents Path: /home/don/.local/share
Resource Path: /tmp/openscad-2015.03-2
User Library Path: /home/don/.local/share/OpenSCAD/libraries
User Config Path: /home/don/.config/OpenSCAD
Backup Path: /home/don/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/don/.local/share/OpenSCAD/libraries
  /tmp/openscad-2015.03-2/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

GLEW version: 1.10.0
OpenGL Version: 3.0 Mesa 10.1.3
GL Renderer: Gallium 0.4 on AMD RS880
GL Vendor: X.Org
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 3.13.0-68-generic #111-Ubuntu SMP Fri Nov 6 18:17:06 UTC 2015
Machine: x86_64
don@serebryanya:/tmp/openscad

kintel, 2015-11-18:
OpenSCAD-2015.03-1 compiled with the latest dependencies still exhibit this bug, so it's caused by one of the dependencies. Likely candidates: QScintilla 2.84->2.9 or Qt 5.4.1->5.5.1

kintel, 2015-11-20:
Update: The issue was introduced with QScintilla 2.9. It looks like they updated a bunch of lexer stuff, which we use for syntax highlighting.

kintel, 2015-11-20:
@donbright Could you give it a spin with QScintilla 2.9 if that's convenient for you?

donbright, 2015-11-21:
cannot reproduce, QScintilla 2.9.1

kintel, 2015-11-30:
I've rebuilt the Mac binaries for 2015.03-2 with QScintilla 2.8.4.
Let's keep this open to track the QScintilla regression in 2.9

kintel, 2016-10-24:
Fixed in QScintilla 2.9.3

closed 2016-10-24 by kintel